FormerLurker/Octolapse #495: Python Crash when using Snap-To-Print

FormerLurker, 2020-04-11:
This problem has been reported for Octolapse 0.4.0rc1.  Basically it crashes while pre-processing some gcode outputted by Cura (perhaps others, it is unknown at the moment).

Keichide, 2020-04-11:
So i got a bit further into the Problem and found out, that the Gcode actually works without crashing Python, the only thing i did, was to change the Slicer Type from "Above Cura 4.2" to "Automatic Detection" and it worked.. And of course i also changed the setting back to "Snap-to-print High Quality" with "Disabled" Stabilization. 

So only strange thing is the number of Snapshots before and after i changed the Slicer Type.. I think that are now a few hundred less. At least acording to the Logs...

With Cura 4.2 as Slicer Type: 
`2020-04-11 18:54:02,231 - octolapse.snapshot_plan - INFO - Completed file processing
	Bytes Processed      : 16893118
	Lines Processed      : 503002
	Snapshots Found      : 121967
	Total Seconds        : 25.2621`

With Automatic Detection: 
`2020-04-11 22:40:51,707 - octolapse.snapshot_plan - INFO - Completed file processing
	Bytes Processed      : 16629320
	Lines Processed      : 495126
	Snapshots Found      : 775
	Total Seconds        : 7.67065`
[settings.zip](https://github.com/FormerLurker/Octolapse/files/4465385/settings.zip)

FormerLurker, 2020-04-11:
There is probably a value that is killing the C++ routine somehow, and I'd LOVE to be able to recreate it and fix before I release, so thank you for posting your settings.  I'll edit your post a bit later and add the gcode link as well, to keep everything clean.  Also, can you real quick change your slicer type back to cura 4.2 and make sure none of the values changed (automatic slicer settings detection will replace them depending on other settings, it's a long story...).  I just want to make sure the settings I'm testing are the same ones that are causing the crash.

Also, if you are running vase mode, definitely do NOT use the high quality snap to print (it refuses to take snapshots unless your extruder is retracted).  That's probably why you found fewer snapshots.  There are likely layers with no retraction at all (surfaces).  The new automatic feature detection will take care of most retraction issues by only taking snapshots over infill (preferred) or interior perimeters (if no infill is available).  Vases are a different story (you will get a seam most likely, but it will work better than Octolapse 0.3.4).

NO IDEA why it took over three times as long to preprocess when using the cura 4.2 slicer type...  I'm sure it has something to do with why it crashed.

Keichide, 2020-04-12:
Okayy.... I just changed it back to Cura tried to print the same gcode that i gave you and it worked... So maybe it could be related to the "Internal Server Error" Massage that i got in the other Ticket, where it could'nt save the Settings to a file? I think i saved the Printer Settings before i applied your Patch, because i did'nt get any Error Message on the other Settings Tabs. But it could also be that i has something todo with the Preview Windows, since Python Crashed while it was rendering it. 

And after a sucessful print last night, i took a look inside the Logfiles again and found that in the Octoprint.log 
`2020-04-11 22:52:05,898 - tornado.application - ERROR - Uncaught exception GET /downloads/logs/plugin_octolapse.log (::ffff:192.168.2.190)
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/web.py", line 1512, in _execute
    result = yield result
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/gen.py", line 1055, in run
    value = future.result()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/concurrent.py", line 238, in result
    raise_exc_info(self._exc_info)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/gen.py", line 1069, in run
    yielded = self.gen.send(value)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/web.py", line 2422, in get
    yield self.flush()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/web.py", line 953, in flush
    return self.request.connection.write(chunk, callback=callback)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/http1connection.py", line 437, in write
    self._pending_write = self.stream.write(self._format_chunk(chunk))
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/http1connection.py", line 412, in _format_chunk
    "Tried to write more data than Content-Length")
HTTPOutputError: Tried to write more data than Content-Length
2020-04-11 22:52:06,936 - tornado.general - ERROR - Cannot send error response after headers written
2020-04-11 22:52:08,916 - tornado.application - ERROR - Uncaught exception GET /downloads/logs/plugin_octolapse.log (::ffff:192.168.2.190)
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/web.py", line 1512, in _execute
    result = yield result
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/gen.py", line 1055, in run
    value = future.result()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/concurrent.py", line 238, in result
    raise_exc_info(self._exc_info)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/gen.py", line 1069, in run
    yielded = self.gen.send(value)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/web.py", line 2422, in get
    yield self.flush()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/web.py", line 953, in flush
    return self.request.connection.write(chunk, callback=callback)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/http1connection.py", line 437, in write
    self._pending_write = self.stream.write(self._format_chunk(chunk))
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/http1connection.py", line 412, in _format_chunk
    "Tried to write more data than Content-Length")
HTTPOutputError: Tried to write more data than Content-Length
2020-04-11 22:52:09,359 - tornado.general - ERROR - Cannot send error response after headers written`

> Also, if you are running vase mode, definitely do NOT use the high quality snap to print (it refuses to take snapshots unless your extruder is retracted). That's probably why you found fewer [...]

You should change the hightlight text than... I mean you explicitly stated there that it is suitable for Vase Mode. 

> NO IDEA why it took over three times as long to preprocess when using the cura 4.2 slicer type... I'm sure it has something to do with why it crashed.

Still the Same... Cura 10.6229 Seconds Automatic Detection 4.67426 Seconds and in both cases 775 Snapshots, which is equal to all 3 Snap-to-print Profiles.

But since i was already on it.. I also tested a bit further with another Gcode that i sliced (which has a much higher Speed Setting, because i just found a bug in Cura where it limits the speed to the "Minimum speed" set for the whole print) 
[CE3_Twisted_Vase_Basic.zip](https://github.com/FormerLurker/Octolapse/files/4465981/CE3_Twisted_Vase_Basic.zip)

With Cura selected it takes 0.681785 Seconds and 1249 Snapshots for the one without "normal" Surface Mode and 0.680807 Seconds and 1250 Snapshots with "surface" mode. 

In "Auto Detection" it takes 0.939549 Seconds with 3740 Snapshots in the "normal" mode and 0.684374 seconds with 1250 Snapshots in the "surface" mode. 

So it has something todo with the File Size of the Gcode

FormerLurker, 2020-05-06:
Sorry this took so long for me to get around to.  Just been getting slammed recently :)

> You should change the hightlight text than... I mean you explicitly stated there that it is suitable for Vase Mode.

Yes, I will do that.  I definitely intended to (the description was obviously copy pasted from another profile), but missed it in the settings repository.  I will remedy that.  If you notice any other documentation issues please let me know.  There is just a ton of documentation, and I'm the only person to have reviewed the vast bulk of it :(

> Still the Same... Cura 10.6229 Seconds Automatic Detection 4.67426 Seconds and in both cases 775 Snapshots, which is equal to all 3 Snap-to-print Profiles.

I have some ideas as to what is happening here, which I will investigate over the weekend.

Lastly, and this is a bit embarrassing, but I can no longer locate the gcode file that you were using when python crashed!?  I believe this is the most serious issue, and the one I definitely need to solve before the next RC is released.  If you have the time, and can find the file, would you be so kind as to re-post it in this thread?  Sorry about that...

FormerLurker, 2020-05-06:
Hey, I just found the gcode (posted in message RIGHT IN THIS THREAD... I'm losing it... lol), so ignore that request :)

FormerLurker, 2020-05-09:
Ok, so I still can't replicate the crash.   If you can replicate the crash, could you export your printer profile settings and send it to me?  The settings you posted earlier are either the post-crash settings, or they got changed by the automatic slicer settings detection.  You may want to upgrade to the latest devel branch code, since I fixed several issues, including the 'surface mode' issue you reported (this setting should just be gone from all cura profile settings).