FormerLurker/Octolapse #496: [0.4.x] Vase Surface Mode should be independent

Keichide, 2020-04-12:
I'm opening a new Ticket for this, since i did'nt want to spam Ticket [#495](https://github.com/FormerLurker/Octolapse/issues/495) with it.

Now the thing that i really don't like right now is the fact, that Cura does'nt add any Bottom Layers to the print in "Surface" mode, which makes the whole "Vase Mode" thing quite useless for anyone. I mean you can't even print something like this [Cloth Vase](https://www.thingiverse.com/thing:2203039) which is only for Pencils and stuff, because you need the Bottom for it. Don't get me wrong here, i love the way you extened the Plugin, but that restriction in the Slicing Settings is crap. 

I think it would be better to just check the Gcode for the Length of a Layer and the Type, since the Gcode from Cura is pretty straight forward in Vase Mode after he printed the Bottom Layers. I think a simple comparsion between the total Layers and the counts of ";TYPE:WALL-OUTER" would be enough to detect the Vase Mode. Or just check the end of the Gcode where it says "\\nmagic_spiralize = Tr
;SETTING_3 ue\\"

FormerLurker, 2020-04-14:
This is probably a lapse in my  understanding of Cura.  If magic_spiralize is the only setting that triggers vase mode, than that would be good enough.  However, the output you mention is not good enough to detect vase mode, since that is only included in the gcode if it differs from the default profile values.  However, I believe this command is included in the settings script linked to in the wiki, so I just need to adjust my vase mode detection in python to account for that setting.

Can you verify that the 'surface' mode doesn't trigger vase mode?  Basically I need to detect any possible configuration that could trigger a vase mode style print.  Also, those settings aren't a 'restriction' really at all, as it only affects octolapse layer change detection.  I could have easily just included a 'vase mode' check box, but I wanted to match the cura settings so that users don't need to guess about what triggers vase mode style prints, where z continually increases.

Does any of that make sense?

FormerLurker, 2020-05-06:
@Keichide, any word on what settings exactly trigger vase mode in cura?  Is it ONLY the 'Spiralise Outer Contour' setting (What a strange name for vase mode...) or are there other combinations?   Anyway, I have found conflicting info about what triggers vase mode, and it seems to vary by version.   
 I hear people talking about 'Classic Vase Mode' and discussing combinations of settings, but haven't found any simple description of what triggers smooth Z increments.  If you have any information I'd appreciate it.

Also, I want to make clear what Octolapse actually does with this information.  It sets an internal flag called 'Vase Mode' and that is all (no gcode changes whatsoever are triggered).  If it is enabled AND the layer height is available in the gcode file it will take at most one photo for the supplied layer height.  If vase mode is NOT detected OR no layer height can be extracted from the gcode, it falls back to the 'minimum layer height' setting:

[image]

If you set that to your actual layer height, vase mode will always work as expected.  Does that all make sense?

Anyway, there has been a battle going on about how to handle vase mode within Octolapse for a long time.  I've been chewed out on more than one occasion for ruining someone's day, in fact, because Octolapse was taking a snapshot on every movement (though a height trigger exists in the old version to fix this issue). The 'minimum layer height' was added as a compromise to prevent Octolapse from totally destroying the print if the slicer settings are entered or detected incorrectly.  My personal opinion is that Octolapse should probably be turned off for these prints, but sometimes you want a timelapse, right?  The 'Snap to Print' mode is somewhat less destructive to vases if it is properly configured and tuned, but even in the best case there will be a slight seam (but smooth mode is pretty awesome if you haven't tried it).  Even the stock octoprint timelapse can cause issues with vase mode if you select 'at layer change', but it's not as bad as what octolapse 0.3.4 (old version) will do.

Anyway, I will be spending some time on this over the weekend, so if you can give me some information I'll try to work this in.

Tamaniha, 2021-03-18:
@FormerLurker 
I just setup octolapse and sliced a normal print (not vase mode) with Cura 4.8 using the auto configuration and found that vase mode was turned on in the gcode

```
; Script based on an original created by tjjfvi (https://github.com/tjjfvi)
; An up-to-date version of the tjjfvi's original script can be found
; here:  https://csi.t6.fyi/
; Note - This script will only work in Cura V4.2 and above!
; --- Global Settings
; layer_height = 0.12
; smooth_spiralized_contours = True
; magic_mesh_surface_mode = normal
; machine_extruder_count = 1
; --- Single Extruder Settings
; speed_z_hop = 5
; retraction_amount = 6.5
; retraction_hop = 0.2
; retraction_hop_enabled = False
; retraction_enable = True
; retraction_speed = 25
; retraction_retract_speed = 25
; retraction_prime_speed = 25
; speed_travel = 150.0

```
So I looked into it and found that the token {smooth_spiralized_contours} that's used to activate vase mode is a sub-setting of the vase mode that is ON by default
[image]

I believe that the correct token to activate vase mode is {magic_spiralize}  which I found here: http://files.fieldofview.com/cura/Replacement_Patterns.html
[image]