FormerLurker/Octolapse #498: Octolapse not compatible with fast-math gcc-avr compile option

vajonam, 2020-04-13:
When using Octolapse with marlin 2.0.5.x and using some custom comiple options mentiond in here, be sure to avoid 

https://thborges.github.io/blog/marlin/2019/01/07/reducing-marlin-binary-size.html

```-ffast-math```

Working compile options
`-fno-tree-scev-cprop -fno-split-wide-types -Wl,--relax -mcall-prologues -finline-limit=3
`
The result of using ` -ffast-math` is really choppy moves to snapshot position and back. 

Might be related to #16

vajonam, 2020-04-13:
Please include in the docs/faq warning. i spent many hours debugging this!

FormerLurker, 2020-04-13:
Hmm, I'm not sure how that's possible since the pre-processor in Octolapse uses doubles, and rounds to 0.000005 for output and comparisons (equal to, greater than, etc...).  My understanding of that flag is that number of significant digits shouldn't be affected.

Can you show me a before and after of the stabilization movements (a video would do)?

FormerLurker, 2020-04-13:
Hmm, I didn't notice that your compiler doesn't really support doubles (it converts them into floats).  That isn't good.  Why are you using that compiler for Octolapse?  Also, is this a Marlin issue?  If so, it seems like it would affect all printing, and not just Octolapse.  Additionally, octolapse 0.3.4 does not use the c++ compiler at all (0.4.0+ does).  Are you using the pre-release version?  If so, I haven't pushed out the documentation for that.  

Sorry if I'm being dense.  I'm just confused.

vajonam, 2020-04-13:
Sorry I should have been more clear.

It happens when compiling marlin with those optimizations for space mainly. (running on original ender3 board for now). Normal printing seems unaffected. I did a print with octolapse disabled and octolapse (0.4.0x) enabled and it was appertenly only when it was enabled. I am not doing any optimizations for octolapse itself. 

So i did a build of the firmware without the fast-math option and this worked in both cases. 

The reason I went to the devel branch is that I couldn't get the exturder to work after the first snapshot, and setting G90/91 affects exturder to true to have any effect, it was still in the wrong mode. 

I am in the middle of a print, I will update it with video with and without the flag for comparison.

FormerLurker, 2020-04-13:
Excellent, that clears things up.  I was confused because Octolapse also uses the fast-math compile flag!  

The g90/91 problem is a known issue, and has been fixed in 0.4.x, so you were right to upgrade.

Looking forward to seeing your comparison video!  I know that will be time consuming to create.

vajonam, 2020-04-13:
interesting, given that fast-math on marlin may be causing some issues. but the video will tell!

vajonam, 2020-04-13:
Here is the video with `-ffast-math` enabled for marlin compile

https://www.dropbox.com/s/cneb1fvcij66z15/Video%202020-04-13%2C%207%2022%2028%20PM.mov?dl=0

Here is a video with `"ffast-math` disabled for marlin compile

https://www.dropbox.com/s/0xu3o7x7ks3lbcb/Video%202020-04-
13%2C%206%2038%2024%20PM.mov?dl=0

I am next uploading a print with fast-math enabled but octolapse plugin disabled. The stuttering only happens when moving to and from the snapshot spot (Back left in my case)

vajonam, 2020-04-13:
Well you are right, I think with all the version flip flopping (G90/91 issue) I was doing last night and various build combinations I was mistaken! 

Here is a print with octolapse disabled. It still happens, but is less pronounced because of the lack of travel by the nozzle.  I might have missed the start of the print and came to see it was printing fine, its more pronounced at the start of the print. 

[](https://www.dropbox.com/s/pmwxilmwhb0nzlz/Video%202020-04-13%2C%207%2040%2017%20PM.mov?dl=0)

Call off the dogs.. not sure how I raise this withe Marlin folks google doesn't resolve it.

closed 2020-04-13 by vajonam

FormerLurker, 2020-04-14:
So, take a look at [this issue](https://github.com/FormerLurker/Octolapse/issues/451).  The same behavior is noted, and it was tracked back to marlin.  It only happens when the buffer is empty, and then not every time.  It appears to be a planner issue that nobody has bothered to fix since few ppl are using m400 in the middle of their prints.