FormerLurker/Octolapse #499: insecure content loaded by pnotify

vajonam, 2020-04-14:
Looks like pnotify is loading some content like icons from 

http://sciactive.com/pnotify/includes/desktop/error.png

I am running octorprint behind a https reverse proxy. But loading this resource causes it to become an unsecure page as its loading mixed conent.

to reproduce

Insect the tab with chrome, and cancel a print half way and you see a pnotify toast, and in the network dialog you can see  the inscure load of the icon.

vajonam, 2020-04-14:
[image]

FormerLurker, 2020-04-14:
Wow, nice catch!  The pnotify js code is loaded by OctoPrint, so do you see the same thing without Octolapse enabled?  If not, perhaps I'm using some functionality OctoPrint does not.  In any case, I'll look into this and see if I can spot the part that loads this image.

vajonam, 2020-04-14:
I have not noticed the pnotify errors issues prior to 0.4.0 octolapse. 0.3.4 did not have this issue. It happens when there a toast notification to be shown.

FormerLurker, 2020-04-14:
I've tracked this down to OctoPrint/src/octoprint/static/js/lib/pnotify/pnotify.desktop.min.js.  It looks like that icon is a fallback if one is not specified.  I will look for a suitable icon for replacement.

vajonam, 2020-04-14:
Just changing it to https works. If you want to use that and include in
octolapse.

FormerLurker, 2020-05-06:
Quick update:  still trying to deal with this, but haven't had a lot of time.  Will be spending time on it over the weekend.  Thanks for your patience.

FormerLurker, 2020-05-08:
This should be solved in the devel branch.  I basically turned all of the desktop notifications off.  They are unnecessary anyway.  Let me know if you are able to try it out.

vajonam, 2020-05-15:
I pulled and installed devel.zip from github, looks good. I don't see issues so far. I will keep an eye out for it. you can close this bug for now. I will re-open if if its not resolved. or something else "pops" up.

closed 2020-05-15 by vajonam