FormerLurker/Octolapse #500: Position Error

dgarrison1408, 2020-04-14:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

Version of Octolapse
 
Octolapse Version: OCTOLAPSE_VERSION_GOES_HERE
Octolapse 0.3.4

OctoPrint 1.4.0 running on OctoPi 0.16.0 Version of OctoPrint

OctoPrint Version: OCTOPRINT_VERSION_GOES_HERE
OctoPrint 1.4.0 running on OctoPi 0.16.0

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES_OR_NO
NO

#### What were you doing when the problem occurred

1. STEP_ONE_GOES_HERE Began a new print
2. STEP_TWO_GOES_HERE began getting the attached error
3. STEP_...

#### What should have happened?
PUT_YOUR_DESCRIPTION_HERE
I use CURA to slice my prints. I have mirrored the settings and followed the video for setting it up. 

#### What happened instead?
PUT_YOUR_DESCRIPTION_HERE
I am getting the attached error message

#### Operating System running OctoPrint and Octolapse

OS Name: OS_NAME_GOES_HERE OctoPi 
Os Version: OS_VERSION_GOES_HERE unknown version

#### Printer model & used firmware incl. version

Printer Model: PRINTER_MODEL_GOES_HERE Ender 3 Pro with EZABL bed leveling 
Printer Firmware Version: PRINTER_FIRMWARE_VERSION_GOES_HERE Marlin 2.0

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE Version 80.0.3987.163 (Official Build) (64-bit)
Browser OS: BROWSER_OS_GOES_HERE Google Chrome

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

[Settings.txt](https://github.com/FormerLurker/Octolapse/files/4477045/Settings.txt)
[image]

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2020-04-14:
It looks like your min and max x,y and z is set to the same thing.  Try setting min x,and z to 0 and min y to -3 (for purging and priming on the MK2) and see if that clears it up.  It's definitely a printer volume configuration issue though.

dgarrison1408, 2020-04-14:
Thank you. I did get an error when it tried to take a snapshot following the purge line but after that it seems to be working fine.

Danny

FormerLurker, 2020-04-14:
Don't worry too much about it.  I've eliminated that error from the next version (v0.4.0, rc1 has been released and rc2 is coming soon), and it's a rather benign error (though irritating).  If the printer volume is incorrect, however, it may not stabilize in the expected position.  You'll be happy to know that the next version has a library of printers by make/model, so in most cases there is no configuration beyond making the proper selection.

dgarrison1408, 2020-04-14:
Thank you very much for your help with this. I have been racking my brain with this issue and could not figure it out.

FormerLurker, 2020-04-14:
Anytime!  That's what I'm here for after all.  Can I consider this closed?

Feel free to post again if you have more problems.

FormerLurker, 2020-05-06:
Going to go ahead and close this since I believe things are working for you now.  Thanks for posting!

closed 2020-05-06 by FormerLurker