FormerLurker/Octolapse #501: After disconnecting and reconnecting a USB camera the plugin keeps crashing, preventing any printing.

poikilos, 2020-04-17:
#### Version of Octolapse
Octolapse Version: 0.3.4

#### Version of OctoPrint
OctoPrint Version:  1.4.0 (Python 2)

#### When you ran into the problem, did you have diagnostic logging enabled?
Diagnostic Logging was Enabled: NO

#### What were you doing when the problem occurred
1. Disconnect the webcam from the USB port.
2. Reconnect the webcam to the USB port.
3. (re-choosing the webcam still fails. Restarting octoprint, it still fails)

#### What should have happened?
The plugin should detect the camera.

#### What happened instead?
- "Octolapse Startup Failed Unable to start the timelapse. Cancelling print. Error: An unknown exception occurred while testing the 'Default Webcam' camera"
- clicking Print again shows the error again. The print job does not start. The following steps do not work:
  1. disable "Default Webcam"
  2. enable "Default Webcam"
  3. Re-choose the webcam from the webcam settings of octolapse
  4. disable the octoprint plugin
  5. `systemctl restart octoprint`
  6. enable the octoprint plugin
  7. systemctl restart octoprint
- The only way that I was able to fix the issue is by restarting the OS.

#### Operating System running OctoPrint and Octolapse
OS Name: Debian
Os Version: 10

#### Printer model & used firmware incl. version
Printer Model: N/A
Printer Firmware Version: N/A

#### Browser and version of browser, operating system running browser
Browser: Brave 1.4.96
Browser OS: Fedora 31

#### Link to the gcode file you were printing when the problem occurred
Link to Gcode File: N/A

#### Link to settings.json
Link to settings.json with all passwords removed: N/A

#### Link to plugin_octolapse.log
Link to plugin_octolapse.log:  https://pastebin.com/wer42scB

#### Link to octoprint.log
Link to octoprint.log:  https://pastebin.com/aHL8z6fH (includes starting a print successfully after restarting the OS)

#### Link to contents of Javascript console in the browser
Link to javascript console output: N/A

#### Screenshots and/or videos of the problem:
Screenshot/Video Links: N/A

FormerLurker, 2020-04-18:
Does your camera continue to work after you plug it in again?  If so, perhaps you'd be willing to try the latest release candidate version.  There are so many bugfixes in there.  Let me know.

poikilos, 2020-04-18:
mjpeg-streamer will not start, and seems to be the root cause of the issue. I think the bug is invalid and a fault of MJPG-streamer or perhaps even V4L2. You can close this issue as invalid. I have posted the issue at <https://github.com/jacksonliam/mjpg-streamer/issues/246>.

More information about previous test:

I failed to ensure at what stages the "control" tab could even see the webcam. In the test above, at some point it showed an image, but it may have been a stale image. 

Retest information:

The following does not work:
- Verify Octolapse (from pip) is working.
- Disconnect the webcam from the USB port.
- Reconnect the webcam from the USB port.
- (Octolapse (from pip) is not working at this point.)
- Uninstall Octolapse
- `systemctl restart octoprint`
- Install `https://github.com/FormerLurker/Octolapse/archive/v0.4.0rc1.tar.gz` from the Octoprint web UI.
-  `systemctl restart octoprint`
- (Octolapse (0.4.0rc1) is not working at this point.)

FormerLurker, 2020-04-18:
OK, that makes a lot more sense.  My camera usually starts working if I unplug it and plug it back in.  Maybe you could try updating mjpg-streamer?  I will follow that issue you linked to.  Thanks for letting me know!

closed 2020-04-18 by FormerLurker