FormerLurker/Octolapse #502: Use ffmpeg for rendering videos, avconv no longer a supported with a Raspberry Pi 4 and Buster

TheBennyJ, 2020-04-18:
#### If this is a feature request describe it here

It looks like installing ffmpeg no longer includes avconv and moving to only use ffmpeg. I've been unable to find a way to work around and install avconv from any package. Please change over to use ffmpeg for rendering the timelaspses

Here's an article on the change over https://wiki.debian.org/Debate/libav-provider/ffmpeg

the relevant entry in the plugin.octolapse.log:
RenderError: return-code: Could not render movie, got return code -100: 
2020-04-16 18:43:58,307 - octolapse.script - ERROR - The script at path '/usr/bin/avconv' could not be found for  for Unknown.  Please check your script path and try again.
2020-04-16 18:43:58,336 - octolapse.render - ERROR - Rendering Error
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_octolapse/render.py", line 1578, in _render
    return_code, stderr_text))

#### Version of Octolapse
 
Octolapse Version: 0.4.0rc1

#### Version of OctoPrint

OctoPrint Version: 1.4.0

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES_OR_NO

#### What were you doing when the problem occurred

1. STEP_ONE_GOES_HERE
2. STEP_TWO_GOES_HERE
3. STEP_...

#### What should have happened?
PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: OS_NAME_GOES_HERE
Os Version: OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: PRINTER_MODEL_GOES_HERE
Printer Firmware Version: PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2020-04-18:
Just change the path to ffmpeg in the octoprint webcam settings.

TheBennyJ, 2020-04-18:
I've set that value already, it and it is successfully gathering all the screenshots, but the final rendering is failing because it is looking for avconv (is there another place it needs changed?)

[image]

FormerLurker, 2020-04-18:
Hmmm.  Try rebooting, maybe it is cached.  I will take a look a bit later to confirm (making fajitas now, yum).

FormerLurker, 2020-04-18:
Out of curiosity, is the failed rendering showing up on the tab so you can retry?  I haven't heard much feedback about that feature and am curious if it is working.

TheBennyJ, 2020-04-18:
will do in a couple hours, mid print right now

FormerLurker, 2020-04-18:
Of course, thanks very much for the report!  It will get solved today or tomorrow for sure

TheBennyJ, 2020-04-18:
Yes sir, I do see them and can retry (though same results). If the reboot fails I will also try reinstalling the plugin just to make sure it isn't hold the original value somewhere else. I updated the path after installing and configuring the plugin the first time. 

[image]

[image]

FormerLurker, 2020-04-18:
Sweet!  At least you didn't lose the timelapses!  In the meantime feel free to download the images and render in your favorite software.  You can even edit and upload the images back into octolapse!

TheBennyJ, 2020-04-18:
The reboot did it, must have cached the old value. Thanks for the support, love the timelapses!

closed 2020-04-18 by TheBennyJ