FormerLurker/Octolapse #503: An error popup appears when you click buttons on the Main Settings of 0.4.0rc1

poikilos, 2020-04-18:
#### Version of Octolapse
Octolapse Version: 0.4.0rc1

#### Version of OctoPrint
OctoPrint Version: 1.4.0 (Python 2)

#### When you ran into the problem, did you have diagnostic logging enabled?
Diagnostic Logging was Enabled: NO

#### What were you doing when the problem occurred
1. Install version 0.3.4 from pip. Uninstall it. Do not clear settings.
2. Install 0.4.0rc1
3. Go to "Main Settings" of Octolapse (via OctoPrint web UI, Plugins, Octolapse, Edit Main Settings)
4.a. Click any of the "Test" buttons in the "Folders" section, while the field beside the test button is blank.
4.b. (alternate step) Click "Save"

#### What should have happened?
A (usually blue) info message stating that the default directory derived from OctoPrint is valid and writable. 

#### What happened instead?
- If you click any of the 3 "Test" buttons, a red popup message appears:
> **Enable/Disable Error**
> Unable to test the directory. Status: error. Error: INTERNAL SERVER ERROR

- If you click the "Save" button, the following error appears:
> Main Settings Save Error
> Unable to save the main settings. Status: error. Error: INTERNAL SERVER ERROR

#### Operating System running OctoPrint and Octolapse
OS Name: Debian
Os Version: 10

#### Printer model & used firmware incl. version
Printer Model: N/A
Printer Firmware Version: N/A

#### Browser and version of browser, operating system running browser
Browser: Brave 1.4.96
Browser OS: Fedora 31

#### Link to the gcode file you were printing when the problem occurred
Link to Gcode File: N/A

#### Link to settings.json
Link to settings.json with all passwords removed: N/A

#### Link to plugin_octolapse.log
Link to plugin_octolapse.log:  It doesn't have any errors from the time period of the issue.

#### Link to octoprint.log
Link to octoprint.log:   It doesn't have any errors from the time period of the issue.

#### Link to contents of Javascript console in the browser
Link to javascript console output: N/A

#### Screenshots and/or videos of the problem:
Screenshot/Video Links: N/A

FormerLurker, 2020-04-18:
This has already been fixed, but thanks for the report!  It has been applied to the devel branch.  Working on a few things, and will then drop rc2.  If you try the devel branch code, let me know if it is fixed for you.

FormerLurker, 2020-05-06:
So, I've been delayed in releasing RC2 by a few reported issues I haven't been able to figure out.  Have you had a chance to verify that the code on the devel branch (can be installed in the plugin manager with this url: ```https://github.com/FormerLurker/Octolapse/archive/devel.zip```) fixes your issue?

FormerLurker, 2020-06-13:
This is fixed.

closed 2020-06-13 by FormerLurker