FormerLurker/Octolapse #504: Manual focus range on camera setting page doesn't match camera

ragbagger, 2020-04-22:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.3.4

#### Version of OctoPrint

OctoPrint Version: 1.4.0

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: NO

#### What were you doing when the problem occurred

1. Taking time lapse with autofocus disabled

#### What should have happened?
I should be able to put a manual focus value in that gives correct focus on the camera page in Octolapse.

#### What happened instead?
The range is limited to between 0 and 255 in Octolapse. I find if I go to control.htm in MJPEGStreamer the available range is 0 to 1023 for my camera and that about 670 gives good focus. The only work around I have found is to turn off the settings in Octolapse and use control.htm to make settings. It would be more convenient to be able to do it in Octolapse. It appears that control.htm checks the camera and starts with the maximum value for the camera in the manual setting. 

#### Operating System running OctoPrint and Octolapse

OS Name: Octopi
Os Version: 0.17.0

#### Printer model & used firmware incl. version

Printer Model: PRINTER_MODEL_GOES_HERE
Printer Firmware Version: PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2020-04-22:
Question:  What kind of camera specifically?

Can you go to http://{ip_of_pi}/webcam/input.json and put the contents in gist.github.com for me to review?  The sliders are supposed to be controlled by querying the camera's capabilities (which come from that json file).  The controls are dynamically built based on the output of that file.  I did add the capability to add custom control pages (they exist already for a few cameras), but I've not seen incorrect values like this before.  Could be a bug in the drivers or mjpegstreamer (or octolapse potentially).

ragbagger, 2020-04-23:
Hi, 

The camera is a generic one off of Amazon:

https://www.amazon.com/gp/product/B07PMJY6Q4/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8 <https://www.amazon.com/gp/product/B07PMJY6Q4/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1> &psc=1

 

I haven’t quite figured out how to upload the file contents to Gist yet so I attached the file as a txt file. In reviewing the file I see in the absolute focus section it has min= 0, max=1023, and default=600 but that is not what I am getting in the Octolapse interface in Octoprint. Weird.

 

"controls": [
{
"name": "Brightness",
"id": "9963776",
"type": "1",
"min": "-64",
"max": "64",
"step": "1",
"default": "0",
"value": "35",
"dest": "0",
"flags": "0",
"group": "1"
},
{
"name": "Contrast",
"id": "9963777",
"type": "1",
"min": "0",
"max": "64",
"step": "1",
"default": "32",
"value": "32",
"dest": "0",
"flags": "0",
"group": "1"
},
{
"name": "Saturation",
"id": "9963778",
"type": "1",
"min": "0",
"max": "128",
"step": "1",
"default": "60",
"value": "128",
"dest": "0",
"flags": "0",
"group": "1"
},
{
"name": "Hue",
"id": "9963779",
"type": "1",
"min": "-40",
"max": "40",
"step": "1",
"default": "0",
"value": "0",
"dest": "0",
"flags": "0",
"group": "1"
},
{
"name": "White Balance Temperature, Auto",
"id": "9963788",
"type": "2",
"min": "0",
"max": "1",
"step": "1",
"default": "1",
"value": "0",
"dest": "0",
"flags": "0",
"group": "1"
},
{
"name": "Gamma",
"id": "9963792",
"type": "1",
"min": "72",
"max": "500",
"step": "1",
"default": "100",
"value": "100",
"dest": "0",
"flags": "0",
"group": "1"
},
{
"name": "Gain",
"id": "9963795",
"type": "1",
"min": "0",
"max": "100",
"step": "1",
"default": "0",
"value": "100",
"dest": "0",
"flags": "0",
"group": "1"
},
{
"name": "Power Line Frequency",
"id": "9963800",
"type": "3",
"min": "0",
"max": "2",
"step": "1",
"default": "1",
"value": "2",
"dest": "0",
"flags": "0",
"group": "1",
"menu": {"0": "Disabled", "1": "50 Hz", "2": "60 Hz"}
},
{
"name": "White Balance Temperature",
"id": "9963802",
"type": "1",
"min": "2800",
"max": "6500",
"step": "1",
"default": "4600",
"value": "5000",
"dest": "0",
"flags": "16",
"group": "1"
},
{
"name": "Sharpness",
"id": "9963803",
"type": "1",
"min": "0",
"max": "6",
"step": "1",
"default": "2",
"value": "6",
"dest": "0",
"flags": "0",
"group": "1"
},
{
"name": "Backlight Compensation",
"id": "9963804",
"type": "1",
"min": "0",
"max": "2",
"step": "1",
"default": "1",
"value": "0",
"dest": "0",
"flags": "0",
"group": "1"
},
{
"name": "Exposure, Auto",
"id": "10094849",
"type": "3",
"min": "0",
"max": "3",
"step": "1",
"default": "3",
"value": "3",
"dest": "0",
"flags": "0",
"group": "1",
"menu": {"0": "  ?   ? ss", "1": "Manual Mode", "2": "rast", "3": "Aperture Priority Mode"}
},
{
"name": "Exposure (Absolute)",
"id": "10094850",
"type": "1",
"min": "1",
"max": "5000",
"step": "1",
"default": "157",
"value": "157",
"dest": "0",
"flags": "16",
"group": "1"
},
{
"name": "Exposure, Auto Priority",
"id": "10094851",
"type": "2",
"min": "0",
"max": "1",
"step": "1",
"default": "0",
"value": "1",
"dest": "0",
"flags": "0",
"group": "1"
},
{
"name": "Focus (absolute)",
"id": "10094858",
"type": "1",
"min": "1",
"max": "1023",
"step": "1",
"default": "600",
"value": "1023",
"dest": "0",
"flags": "16",
"group": "1"
},
{
"name": "Focus, Auto",
"id": "10094860",
"type": "2",
"min": "0",
"max": "1",
"step": "1",
"default": "1",
"value": "1",
"dest": "0",
"flags": "0",
"group": "1"
},
{
"name": "JPEG quality",
"id": "1",
"type": "1",
"min": "0",
"max": "100",
"step": "1",
"default": "50",
"value": "0",
"dest": "0",
"flags": "0",
"group": "3"
}
],
"formats": [
{
"id": "0",
"name": "Motion-JPEG",
"compressed": "true",
"emulated": "false",
"current": "true",
"resolutions": {"0": "1920x1080", "1": "640x480", "2": "800x600", "3": "1280x720", "4": "1280x1024", "5": "320x240"}
,
"currentResolution": "5"
},
{
"id": "1",
"name": "YUYV 4:2:2",
"compressed": "false",
"emulated": "false",
"current": "true",
"resolutions": {"0": "1920x1080", "1": "640x480", "2": "800x600", "3": "1280x720", "4": "1280x1024", "5": "320x240"}
,
"currentResolution": "255"
},
{
"id": "2",
"name": "RGB3",
"compressed": "false",
"emulated": "true",
"current": "true",
"resolutions": {"0": "1920x1080", "1": "640x480", "2": "800x600", "3": "1280x720", "4": "1280x1024", "5": "320x240"}
,
"currentResolution": "255"
},
{
"id": "3",
"name": "BGR3",
"compressed": "false",
"emulated": "true",
"current": "true",
"resolutions": {"0": "1920x1080", "1": "640x480", "2": "800x600", "3": "1280x720", "4": "1280x1024", "5": "320x240"}
,
"currentResolution": "255"
},
{
"id": "4",
"name": "YU12",
"compressed": "false",
"emulated": "true",
"current": "true",
"resolutions": {"0": "1920x1080", "1": "640x480", "2": "800x600", "3": "1280x720", "4": "1280x1024", "5": "320x240"}
,
"currentResolution": "255"
},
{
"id": "5",
"name": "YV12",
"compressed": "false",
"emulated": "true",
"current": "true",
"resolutions": {"0": "1920x1080", "1": "640x480", "2": "800x600", "3": "1280x720", "4": "1280x1024", "5": "320x240"}
,
"currentResolution": "255"
}

]
}

FormerLurker, 2020-04-23:
Ok, I'm sure I messed something up then.  Will take a look in the next few days and will send you a link with the fix.  Shouldn't be too hard to fix

FormerLurker, 2020-04-23:
Oh, try clicking the custom/default page toggle button in the camera settings page.  Maybe I messed up the camera detector for custom pages

ragbagger, 2020-04-23:
It is currently set to default. Should it have been custom?

 

In the middle of printing Covid respirators. I will give it a try when print is done.

 

Thanks

Brad

FormerLurker, 2020-04-23:
> It is currently set to default. Should it have been custom?

No, it just means my guess about what might be wrong was incorrect. 

> In the middle of printing Covid respirators. I will give it a try when print is done.

Carry on!

ragbagger, 2020-04-25:
Hi, maybe I’m blind or something but I can’t find the custom/default page toggle button in the camera settings page.

 

Brad

FormerLurker, 2020-04-25:
Here is the location:

[image]

ragbagger, 2020-04-26:
Sorry to be a pain. I was looking at the camera setting page within the Octoprint Octolapse tab and it doesn’t look like this. How do you get to this camera setting page?

 

 <https://user-images.githubusercontent.com/33363128/80291512-a9bbb180-8713-11ea-9790-99c379b2e265.png>

FormerLurker, 2020-04-26:
Omg... I didn't notice that you were still on 0.3.4.  i had (have) an explosion of issues due to quarantine.  Apparently everyone is printing non-stop, lol.  It has made me a bit less thorough in order to be able to respond to everyone....  Sorry about that.

0.3.4 does NOT read input.json, so that is definitely the problem.  Try upgrading to 0.4.0rc1 by [reading this link](https://github.com/FormerLurker/Octolapse/wiki/V0.4-Getting-Started-with-Octolapse) and follow option 2 under the installation header.

ragbagger, 2020-04-26:
Brilliant! It is working great now. Issue closed.

 

Thanks much.

Brad

FormerLurker, 2020-05-06:
Just closing this issue out since it has been solved.  Thanks for your post!

closed 2020-05-06 by FormerLurker