FormerLurker/Octolapse #505: Position Error

tonychan818, 2020-04-24:
#### If this is a feature request describe it here
FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
Octolapse Version: 0.3.4

#### Version of OctoPrint
OctoPrint Version: 1.4.0

#### When you ran into the problem, did you have diagnostic logging enabled?
Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred
My printer is AnyCubic Mega i3 Mega S, I did select my Printer as Anycubic i3 Mega. Because I can't see the one with S. when I finish setup the Octolapse and start to print. I got an error message.

#### What should have happened?
Octolapse run without error

#### What happened instead?
When printing start
Octolapse - Position Error
Position - Coordinates (X:-5.00000,Y:20.00000,Z:0.27500,E:246.45000) are out of the printer area! Cannot resume position tracking until the axis is homed, or until absolute coordinates are received.

When printing finished
Octolapse - Position Error
Position - Coordinates (X:-5.00000,Y:288.43000,Z:10.50000,E:252.45000) are out of the printer area! Cannot resume position tracking until the axis is homed, or until absolute coordinates are received.

#### Operating System running OctoPrint and Octolapse
OS Name: OctoPi
Os Version: 0.17.0

#### Printer model & used firmware incl. version
Printer Model: Mega i3 Mega S
Printer Firmware Version: V1.1.2

#### Browser and version of browser, operating system running browser
Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred
Link to Gcode File: https://drive.google.com/file/d/1PeU5KJfVy3HwiBiyfDoRnGMBANbukVK8/view?usp=sharing

#### Link to settings.json
Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://gist.github.com/tonychan818/f417ba36a86d99e4c16d8f2655ca734c

#### Link to octoprint.log
Link to octoprint.log:  https://gist.github.com/tonychan818/43a18facc90e6f4f4ba236c5c9c8c5fb

#### Link to contents of Javascript console in the browser
Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:
Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

tonychan818, 2020-04-25:
After changing the custom printer volume to this.

X: -5, 195
Y: 0 195
Z: 0 195

everything is fine and able to do the octolapse.
But when the printing is done.
Still getting this error

Octolapse - Position Error
Position - Coordinates (X:-5.00000,Y:288.43000,Z:10.50000,E:239.68000) are out of the printer area! Cannot resume position tracking until the axis is homed, or until absolute coordinates are received.

it didnt affect my octolapse, but just wanna clean this error as well =D

FormerLurker, 2020-04-25:
Looks like your end gcode is trying to go to y=288.43.  is your bed that large?  Could there be an error in the end gcode?

tonychan818, 2020-04-26:
my bed is 210 x 210 
and this is the printer setting in Cura
[image]

FormerLurker, 2020-04-26:
What is your end gcode?  Could be a problem there

tonychan818, 2020-04-26:
M104 S0 ; turn off extruder
M140 S0 ; turn off bed
M84 ; disable motors
M107
G91 ;relative positioning
G1 E-1 F300 ;retract the filament a bit before lifting the nozzle, to release some of the pressure
G1 Z+0.5 E-5 ;X-20 Y-20 F{speed_travel} ;move Z up a bit and retract filament even more
G28 X0 ;Y0 ;move X/Y to min endstops, so the head is out of the way
G1 Y180 F2000
M84 ;steppers off
G90
M300 P300 S4000

Thanks :)

FormerLurker, 2020-04-26:
There is a mistake there, but your printer must enforce endstops, which is why it didn't run into one.  After this line:

G1 Z+0.5 E-5 ;X-20 Y-20 F{speed_travel} ;move Z up a bit and retract filament even more

Add this line:

G90

Your printer is in relative mode before this, causing this command:

G1 Y180 F2000

To move in a relative direction.  The G28 with X0 and Y0 is not clearly defined in relative mode, so the implementation is firmware specific.

Anyway, adding g90 there should stop the error, even if your printer works fine without it.

tonychan818, 2020-04-26:
Thanks!
is this the similar problem on the Start G-code? 
Can I solve the error by added G90 after "G1 Z15.0 F{speed_travel} ;move the platform down 15mm"?

The error at the beginning:
Position - Coordinates (X:-5.00000,Y:20.00000,Z:0.27500,E:246.45000) are out of the printer area! Cannot resume position tracking until the axis is homed, or until absolute coordinates are received.

Start G-code:

G21 ;metric values
G90 ;absolute positioning
M82 ;set extruder to absolute mode
M107 ;start with the fan off
G28 X0 Y0 ;move X/Y to min endstops
G28 Z0 ;move Z to min endstops
G1 Z15.0 F{speed_travel} ;move the platform down 15mm
G92 E0 ;zero the extruded length
G1 F200 E3 ;extrude 3mm of feed stock
G92 E0 ;zero the extruded length again
G1 F{speed_travel}
G0 Y20 F{speed_travel}
M117 Printing...
G5

FormerLurker, 2020-04-26:
No that all looks correct.  Not sure how you're getting an error from that code since there are no X movements in that gcode at all, and -5 is within the X axis boundary.  Search for a gcode towards the beginning that contains X-5, and maybe I can gain some insight.

However, I wouldn't sweat that error much.  I've removed those notifications from the next version of Octolapse (V0.4.0rc1 is the latest).  I'm working today to try to push out rc2 (though I am swamped with issues, lol!  Everybody is printing overtime during quarantine apparently.).

If you want to try the new release, let me know and I will attach some instructions.

tonychan818, 2020-04-26:
Ok~ thanks so much

closed 2020-04-26 by tonychan818