FormerLurker/Octolapse #506: Z lift problem after top layer

Goryaschenko, 2020-04-25:
Good day. I have interesting problem.
When I turn on octolapse, the printer skips one layer. This happens mainly after the top layer of the model. When octolapse is turned off, the printer prints all layers normally.

I'm using Flsun Q5 printer with marlin 2.0. / motherboard MKS Robin Nano 
[Google drive with .gcode and .log files ](https://drive.google.com/drive/folders/1rgTQH53e87kmRzLKKinZTCQNw-CGAQkj?usp=sharing)

[Video1](https://youtu.be/YPoL57YQqto)
The problem occurs when the printer prints the last top layer. And then on top of it prints something else. On the video, it is clearly visible that the cabin prints without errors, the stern perimeter is printed with an error. The nozzle rises, but falls a layer higher than it should.

[Video2](https://youtu.be/4GtmoWIft28)
In the second video, on the perimeter of the stern, the nozzle, on the contrary, drops one layer lower than necessary and scratches the printed surface. And the cabin is printed one layer higher than necessary.

Cura settings.
[image]
Octolapse settings.
[image]

I noticed the same problem on my item. there is a cone in it, which was also printed with a skip layer. more precisely, with a nozzle lift error. I solved this problem for myself by modeling the cut walls.
[image]

[Base print video](https://youtu.be/LJDLh0dUOeU)

FormerLurker, 2020-04-25:
Would you be willing to see if this problem still exists in the latest development rc?  I've been working on it for nearly 2 years, so a whole lot has changed.  It's quite possible that the issue has vanished in that time.  [Here is a link](https://github.com/FormerLurker/Octolapse/wiki/V0.4-Getting-Started-with-Octolapse) for getting started.  Follow option 2 under the installation heading (change plugin manager settings), and be sure to read and follow the entire guide.  I think you'll like the changes!

Thanks for reporting your issue in such a detailed way.  I REALLY appreciate that.

Goryaschenko, 2020-04-26:
Unfortunately, the problem persists, but now appears in other places.

Now the problem is visible on the boasts of trays, when printing the cabin the first layer is after the top layer and the bow of the boat is after the top layer.

[Vidoe](https://youtu.be/uOMkM2fvI30l)

FormerLurker, 2020-05-06:
Sorry about the delay, I've been getting crushed lately, lol!

Can you verify that an entire layer is skipped, or just parts of one?  If it's only parts of a layer and you are using Marlin 2.0, it could be the G90/91 influences extruder setting.  If that's wrong you could be some really odd results, like most of a layer (or more than one) missing.

The other thing you could try (after flipping the G90/91 influences extruder setting) is to turn of axis compatibility mode in your printer profile (or on if it is currently off).

If neither of those work, could you generate a log for me with the new version?  Just edit your logging profile first and click 'Clear Log' so we can get a nice clean example.  Print your benchy (or better yet something smaller that has the same problem) until it misses the layer, and let it go for one more layer before stopping.  If you can record the layer number that would be excellent.  If you end up using different gcode than what you posted earlier, post that too.

I've been saving up some of the more complicated issues for this weekend, so if you can have it by then odds are I can solve it before next week.  THANK YOU very much for your patience!

FormerLurker, 2020-05-09:
So, I reviewed the firmware for your printer, and it looks like you should set G90/91 influences extruder to FALSE here:

[image]

This doesn't appear to be a z-lift issues, as I suspected, but rather an absolute/relative extrusion issue.  When your printer receives G90/G91 commands, the E axis mode remains the same (this behavior changes in Marlin 2.0).  See the code for handling G90/G91 in your firmware [here](https://github.com/LVD-AC/Marlin-AC/blob/f18238fedbb5901b3c50e31f34b4566d80d00160/Marlin/Marlin_main.cpp#L12241-L12242). and compare that with M82/M83 [here](https://github.com/LVD-AC/Marlin-AC/blob/f18238fedbb5901b3c50e31f34b4566d80d00160/Marlin/Marlin_main.cpp#L8565-L8573).

Please let me know ASAP when if this works, or if you still have issues since I'm trying like mad to release RC2 this weekend still :)

FormerLurker, 2020-05-30:
Any update?  Did my most recent suggestion help at all?