FormerLurker/Octolapse #507: Problem Delay issue during snapshot Octolapse_Gh4_gphoto2

AlainSop00, 2020-04-25:
Hello,
I have a problem during the snapshot, even at 5000ms delay, the image is taken too early. Is there a solution for the snapshot to be done later when the extruder is in final position ?

FormerLurker, 2020-04-25:
It could be that your printer does not support M400.  I've seen a few that just ignore this command.  It's not easy to test for either.  What kind of firmware are you running?  At least one other person has reported a similar issue where their printer wanted G4 P0 instead of M400.  I plan on adding an option to support this as soon as I get the current release out the door.

Does it work with a webcam?  DSLRs are so complicated compared to a basic webcam timelapse.  It's usually a good idea to make sure things work properly with a webcam before adding complexity.

AlainSop00, 2020-05-05:
Hello,
I'm back on the project! Forgive me but work called me ..
I have an Alphawise U20 with the original software updated.
I used a raspberry pi camera in addition to the DSLR, the two photos are triggered too early. So it would come from the printer software?
Thank you

FormerLurker, 2020-05-06:
It may in fact be your firmware.  For octolapse to work the M400 command must work as advertised.  I just looked at the github page for the Alfawise U20 [here](https://github.com/tpruvot/Marlin), and it should be good.  Perhaps try upgrading to the latest firmware?

Also, only 2 photos triggered early?  The rest looked good?  Can you post a link to the timelapse so I can see what's going on?  It is possible that the snapshots were actually too late maybe, since the timelapse would look the same either way (that's actually easier to understand if that's the case).

AlainSop00, 2020-05-07:
On the Github link that you sent me, the software used is "Marlin", should
I install it instead of the original firmware?
No it's all the photos that are taken earlier, The "snapshot - camera" logo
on the octolapse preview launches just before the nozzle leaves in the
corner of the tray.
I make an impression and send this to you ,

FormerLurker, 2020-05-07:
> The "snapshot - camera" logo on the octolapse preview launches just before the nozzle leaves in the corner of the tray.

It is normal for the camera icon to appear before the snapshot.  It appears right before the triggering gcode is sent to the printer, but Octolapse has to wait for the buffer to run out before taking a snapshot.  Depending on what gcodes were sent, and how long your buffer is, this can be a while.

Now if your camera icon only shows up for a very brief time (under a second) then I would say it's probably your firmware.  

Also, you are using 5000MS delay for your DSLR.  I recommend using a 0MS delay for DSLRs.

You left some important info out of your ticket.  What version of Octolapse and Octoprint are you using?  Also, if you could create a log for me I could verify all of this.  Use one of the Debug profiles, run a short print (maybe 4 or 5 layers of a print), then paste the log into gist.github.com and paste a link to the gist in here.

FormerLurker, 2020-05-07:
Oh, and 

> On the Github link that you sent me, the software used is "Marlin", should I install it instead of the original firmware?

I'm not sure if it is actually safe to use the firmware I linked to you.  It is a fork of Marlin 2 made by a user, so it may have some issues.  I think it would actually be better to verify the problem first.

AlainSop00, 2020-05-10:
Hello,

I made my impressions at 0ms, here are three example videos :
https://we.tl/t-1lZY2tdMbf
I can't download the logs on octoprint, is that what you need to do?

FormerLurker, 2020-05-10:
Thank you for the videos.  They pretty much confirmed the issue for me.  It definitely looks to be a problem with the way your printer responds to M400.  Perhaps you could send a question to the manufacturer and ask them if M400 or G4 P0 is the 'Wait for all moves to complete' command?  Also, it is important that the printer NOT respond to requests such as M114 until AFTER all print moves are finished.  This is the behavior of the vast majority of firmware.

Also, please note that the ```G4 P0``` command is NOT yet supported by Octolapse as an alternative to M400.  There is an open issue about this, and I plan to add support as soon as V0.4.0 stable drops.

Perhaps you could also ask them if there is an official github page for their firmware?  I would be willing to take a look at it and suggest changes to make it compatible with Octolapse, or for any other situation where a properly behaved M400 command is used (there are several uses for this).

AlainSop00, 2020-05-10:
> Here are the downloaded logs, there are too many characters to be able to
> copy them to github.
> So I send them to you as an attachment.
> Thank you for everything!

Lien Wetransfer Octolapse.log : https://we.tl/t-8QBV1ExMLK

Daska737, 2020-06-08:
Hello,

I post to follow this, after much research, I come to the same conclusion.
With an Alfawise U20 V2.16A

I'm using a Canon 1100D EOS, and a webcam. Both trigger too early regardless of the setting.
I could provide you, if you guide me, the elements which could advance on the problem.

FormerLurker - Many thanks for all this work.

FormerLurker, 2020-06-08:
@Daska737 and @AlainSop00,

I just dropped Octolapse V0.4.0, which consumed all of my free time over the last several weeks.  I'm planning to start fixing some known issues, including adding support for G4 P0 as the 'wait for print moves to finish' command.  However, I still have to do some research on the Alfawise printer to make sure this is actually the problem.  I'll post some instructions later to see if we can get to the bottom of this.

Thank you for your patience!

AlainSop00, 2020-06-08:
I installed Marlin on the Alphawise, the problem is solved now !
Thank you so much for all the work you do for the community !

FormerLurker, 2020-06-08:
@AlainSop00, Awesome!  I'm so glad it's working for you now.

Perhaps this issue should be reported to the Alphawise folks?

yufanyufan, 2020-07-22:
I use marlin 2.0.x latest code, and still have this issue. The photo is taken while hothead is still over the print.

yufanyufan, 2020-07-22:
My log shows that printer waited on M400, and reported correct position on M114

Send: N66005 G1 E-6.50000 F1500.000*52
Recv: ok
Send: N66006 G0 X175.000 Y346.500 F12000.000*98
Recv: ok
Send: N66007 M400*32
[...]
Recv: ok
Send: N66008 M114*47
Recv: X:175.00 Y:346.50 Z:6.80 E:881.83 Count X:14000 Y:27720 Z:5360
Recv: ok
[...]
Send: N66009 G0 X184.683 Y200.769*43
Recv: ok