FormerLurker/Octolapse #508: Last bit of video is not shown

mhfleming, 2020-04-27:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.3.4

#### Version of OctoPrint

OctoPrint Version: 1.4.0

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred

1. Slice model in Prusa's slicer.  The model I used was https://www.myminifactory.com/object/3d-print-cute-dinosaur-toy-115883
2. Upload the gcode to OctoPrint
3. Print

The settings I am using are:
Printer: Prusa I3 MK3
Stabilization: Fixed - Extruder at Center Right
Snapshot: Layer change - high quality
Rendering: MP4 - 30 FPS
   I added 1 second at the end of the video in here.  But I have had this issue even with this set to zero.
Camera - Logitech C270
Debug: Full Diagnostics - Live Print

I have tried both the fixed length and the 30 FPS, I was hoping to get a view of the model all the way through the print.

#### What should have happened?
I expected that the video would run all the way to the end of the print and pause there for 1 second.

#### What happened instead?
I video got almost to the end then paused for 1 second and ended.  So the video did not show the real end of the print.

I put a video of the timelapse here:  https://youtu.be/oBhikRsEe1M

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: 0.17.0

#### Printer model & used firmware incl. version

Printer Model: Prusa I3 Mk3S
Printer Firmware Version: 3.8.1

#### Browser and version of browser, operating system running browser

Browser: Chrome
Browser OS: Windows 10

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE
https://drive.google.com/open?id=15StrYJDXZvfkgnDReCbCGoPlh6uiuB0e

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE
https://drive.google.com/open?id=1OgwLwX9AfXamMFkLh-5aXAtJJo_K9i46

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE
https://drive.google.com/open?id=1VDjWS-faXwllQ1U16Pouz8K5yKh6UnGG

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE
https://drive.google.com/open?id=129zd8Q1y3tIIRcKBUNeuGSC2iTF8_7Fc

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: https://youtu.be/oBhikRsEe1M

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

mhfleming, 2020-04-27:
I have reverted back to the defaults and just completed a 4 hour print and it worked just fine.  So I am not sure if I did something different in these settings

FormerLurker, 2020-04-27:
This is an issue with the 'Layer Change - High Quality' snapshot profile.  This feature has been radically altered in the next version to avoid issues, but your slicer probably has a minimum layer time, which reduces the feedrate on those layers.  Since the current master version of octolapse uses speeds for feature detection, the slower feedrates fool octolapse, which results in missing snapshots.

Solutions: 

1.  Use the regular layer change snapshot profile.
2.  Upgrade to Octolapse 0.4.0rc1 and use the smart triggers, which use comment based feature detection, and will take snapshots even if no suitable features are detected (in smart - compatibility mode).

mhfleming, 2020-04-27:
Great news thank you!

closed 2020-04-27 by mhfleming