FormerLurker/Octolapse #509: Octolapse uninstall does not remove Octolapse tab and settings navigation.

uberkeeper, 2020-04-28:
**If this is a feature request describe it here**
This is not a feature request

**Version of Octolapse**
Octolapse Version: 0.3.4

**Version of OctoPrint**
OctoPrint Version: 1.4.0

**When you ran into the problem, did you have diagnostic logging enabled?**
Diagnostic Logging was **NOT** enabled.

**What were you doing when the problem occurred**
Having tried Octolapse I concluded it was far more sophisticated than I required and decided to uninstall the plugin.  Having previously disabled the plugin in the settings I followed the steps to uninstall a plugin, including a full cleanup. 

**What should have happened?**
The plugin should have been fully removed from my installation.

**What happened instead?**
After completing the uninstall of the plugin "successfully", the Octolapse tab remains in the gui, but has no content, and the Octolapse item remains in the left hand navigation of the Octoprint settings.

**Operating System running OctoPrint and Octolapse**
OS Name: Ubuntu Server
Os Version: 20.04 LTS

**Printer model & used firmware incl. version**
Not relevant

**Browser and version of browser, operating system running browser**
Browser: Safari 13.1 / Chrome 81.0.4044.122 (Official Build) (64-bit)
Browser OS: Mac OS High Sierra 10.13.6

Note: this is not a cache/history issue - issue has been confirmed from clean browser.

FormerLurker, 2020-05-01:
Did you reboot?

FormerLurker, 2020-05-01:
The server i mean.

uberkeeper, 2020-05-02:
Hi - Yes, thanks for the response.  I did reboot the server, made no difference.  I'm going to close the issue as I've since done a full reinstall and can no longer provide details of what happened/happens.  But again thanks for taking the time to read and respond.

closed 2020-05-02 by uberkeeper