FormerLurker/Octolapse #512: Extrusion leading into print after each picture

caughron, 2020-05-03:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.3.4

#### Version of OctoPrint

OctoPrint Version: 1.4.0

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred

1. Created gcode from Cura, uploaded print file and began print with Octolapse enabled
2. Print initiates and begins as normal.
3. Following every layer, bed travels to expected location, delays as expected and takes picture
4. Print head then moves to a location about 10mm outside of print and begins an extrusion back into the print - this causes a blob where the extrusion begins thinning to a thin line at the point the nozzle rejoins the print. (This feature is not in the model).
5. This only occurs when Octolapse is enabled. I can print clean, near perfect prints with Octolapse disabled and have manually set Cura to simulate the same movements through postscript add on. When simulated, the nozzle leaves the print area, pauses and returns to resume the print without this anomaly.
6. I have ensured at least a dozen times that all the settings match what I have in Cura.
7. I have attempted to no avail to find a setting that might be causing this, but have not been succesful.

#### What should have happened?
The print should have moved to the defined location for the picture, then returned to the print and resumed without the leading extrusion back into the model

#### What happened instead?
Following every layer, bed travels to expected location, delays as expected and takes picture.
Print head then moves to a location about 10mm outside of print and begins an extrusion back into the print causing a blob where the extrusion begins thinning to a thin line at the point the nozzle rejoins the print. ***I have attached links to pictures and a video showing the problem.

#### Operating System running OctoPrint and Octolapse

OS Name: Windows 10 64-bit
Os Version: 1909
OctoPi 0.17.0

#### Printer model & used firmware incl. version

Printer Model: Ender 3 Pro
Printer Firmware Version: 1.1.6.2

#### Browser and version of browser, operating system running browser

Browser: Version 81.0.4044.129 (Official Build) (64-bit)
Browser OS: Google Chrome

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: https://drive.google.com/open?id=1rKJ13gT2mSkN9eeImIxR8ho286_hid2y

#### Link to settings.json

Link to settings.json with all passwords removed: https://drive.google.com/open?id=19ry9N3-pXBF-C8fmzWiGirrKcdLyb3hH

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://drive.google.com/open?id=1j-ca8RIQbLdZzyFyTWZiyFuIVY9Yfglp

#### Link to octoprint.log

Link to octoprint.log:  https://drive.google.com/open?id=1Lhc3BWPERpxDVNtECHPg0UzwyBfxhzQa

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: https://drive.google.com/file/d/1cQdf6C9Jrh2TCqXJMPudRaWa96YK6VF1/view?usp=sharing
https://drive.google.com/file/d/1izeafF7ii3q4gfwq_A6qgxIMGue41ueK/view?usp=sharing
https://drive.google.com/file/d/16j1k-qjL24AvYdsZnGy2s2KALSBAkR0z/view?usp=sharing
https://drive.google.com/open?id=17PX9ZsW3o9ag6IN4_U4X7UmU1xWs7JcF

[image]
[image]
[image]

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2020-05-03:
Maybe you would try this in v0.4.0rc1?  It is probably an issue that has been fixed, namely problems with switching axis modes for printers that change the e axis mode when g90/91 is used.  Let me know if you are willing to try this

caughron, 2020-05-03:
Yes u would be willing to try that. Thank you for the quick reply.

FormerLurker, 2020-05-03:
Check out [this guide](https://github.com/FormerLurker/Octolapse/wiki/V0.4-Getting-Started-with-Octolapse), and make sure to read the whole thing.  I recommend adjusting your software update settings (option 2 under installation), that way you'll be notified when rc2 drops.  There are a few known issues in rc1, the biggest being the main settings don't update.  If you want, you could also install from the devel branch using this link in the plugin manager, which has a fix for the main settings and a few other small items: ```https://github.com/FormerLurker/Octolapse/archive/devel.zip```

There is a profile for your printer that is already configured.  Edit your printer settings after installation and search for your make and model in the drop down.  Then try reprinting that item.  If it doesn't work, it's possible some adjustments need to be made to your printer profile (I don't have most of the printers in the list, so I sort of winged the settings).

Good luck!

caughron, 2020-05-03:
Thanks, appreciate that. Were there Any other reports of that problem with
the current version. I tried adjusting every setting I thought could be
causing it to no avail. Thanks for the software btw, very cool plugin.

FormerLurker, 2020-05-03:
There isn't any good work around in 0.3.4 unfortunately.  Yes, this issue has been reported in 0.3.4, and even in some of the alpha versions of 0.4.0, but it was eventually tracked down and fixed.  Basically Octolapse switches XYZ axis modes to relative to make the snapshot gcode a bit easier to create and interpret.  That was affecting the extruder, so retraction gets messed up, probably leading to what you are seeing.  More machines now seem to use this odd setting (G90/G91 sets the E axis mode too), so it's becoming more obvious.  It's been around for a while, but was only recently tracked down.

There is also an experimental option in 0.4.0 where Octolapse will not send any g90/91 or M82/83s during the snapshot process.  This leads to fewer gcodes being sent, but it much trickier to pull off.  If you still have problems after upgrading, you may investigate that option (it's in the printer profile).

That being said, 0.4.0 is a MUCH better piece of software and has a ton of new features.  In particular, you may want to investigate the automatic slicer settings extraction, which makes Octolapse much more convenient to use (no more copying settings, PrusaSlicer/Slic3r, cura and simplify 3d currently supported).  There are also new pre-processing triggers (smart triggers) that speed up prints when using octolapse to some extent, and are much better at picking good places to take a snapshot that should minimize artifacts.  Pay close attention to the slicer specific settings changes mentioned in the guide.  If you are using Cura a bit of code will need to go into your start gcode for automatic settings detection to work.  For Slic3r and PrusaSlicer you will need to turn on verbose gcode output to get a big quality boost when using smart triggers.  Simplify works out of the box with no modifications.

caughron, 2020-05-03:
Sounds good. Thanks for the explanation. I’ll start to work on switching
over. Appreciate your help

roomme13, 2020-05-06:
Sorry Jumpin, 
I tried latest devel ( Octolapse v0.4.0rc1.dev5+3ffd305 )
it work for 3 bottom layer, then it start to crush after that.
i even disable Snapshot "Gcode Axis Compatibility Mode" , maybe better, but still only for afew first layers.

FormerLurker, 2020-05-06:
@roomme13, first, upgrade to 0.4.0rc1 (or install from the devel branch, a few bugs were fixed in there).  Next, check the G90/91 influences extruder setting in your printer profile.  If you are using marlin 2, that needs to be true.

FormerLurker, 2020-05-06:
@roomme13, also, when you said, "It start to crush after that."  Can you be a bit more clear? If this isn't related to over extrusion when traveling to take a snapshot, you may want to create a new issue.  If it is related to the issue, then this is the right place to post :)

FormerLurker, 2020-05-06:
@caughron, did the upgrade fix your issue?  If you haven't had time to get to it that's fine :)

roomme13, 2020-05-06:
@FormerLurker 
its not an over extrusion, when using Octolapse its kind of mess with the z lift or z position, somehow next layer put in wrong z position make distance to the bed more closer and squeeze the filament. 
i tried your suggestion to Change G90/91 influences extruder, and it works!
i don't know this never work in 0.3.4 or maybe I'm wrong.
Thank you for the suggestion.

By the way, new automatic configuration will match configuration from slicer right? kind of worry for the retraction length.

FormerLurker, 2020-05-06:
If that settings change fixed it, it probably was over extrusion.  Regarding the automatic settings extraction, octolapse will complain and refuse to print if it cant find all of the settings it needs.  It is safer than manual settings, since it eliminates typos :)

roomme13, 2020-05-06:
why you thought so? 
i cannot get down anymore for the e-step, its already under extrution.
it not only crash down, i got the nozzle higher than it should be too, its make a spaghetti.

Noted for that,  do i still need to make the speed unique? (its already unique in my settings)

FormerLurker, 2020-05-06:
> i don't know this never work in 0.3.4 or maybe I'm wrong.

It did NOT work in all cases.  There was an issue even in some earlier dev versions of Octolapse 0.4!  A very knowledgeable user tracked down the exact problem for me, which made solving it quite easy.  I wish I could remember the username, because it was a fantastic issue report!

> why you thought so?

So, the G90/G91 influences extruder setting has a HUGE impact on extrusion.  If the axis modes are off very bad things can happen.  To make matters worse, slicers like to use G92 a lot to reset the E axis to 0, which makes it look like there are only issues with certain parts, when in reality it's just random what the impact will be.  This has been a common issue as people are upgrading to Marlin 2.0.  In fact, the Prusa fork almost changed this behavior for MK2 and MK3 firmware, but it would have caused these kinds of issues which are very difficult to debug.  To my knowledge there is no particularly easy way to even test your printer to verify the behavior (I've had mixed results with a few scripts, but they all involve running to your printer and seeing if the extruder gear is turning).  The worst part is, now there are printers where some versions of the firmware use false for this setting, and others use true, so even the default profiles can no longer be trusted.  The ender 3 comes to mind, especially with the upgraded board.  Also, there is no way to ask the firmware how it behaves, so it is currently a pretty difficult issue to deal with, and there is no good solution.

> i cannot get down anymore for the e-step, its already under extrution.

I'm not sure what you mean here, but the G90/G91 setting is there so that octolapse can retract properly and send the correct retraction/deretraction command before and after the snapshot.

> it not only crash down, i got the nozzle higher than it should be too, its make a spaghetti.

Again, are you 100% sure it was too high, or did it just stop extruding for most or all of a layer?  The G90/G91 setting can cause both over extrusion (huge blobbs) and under extrusion (no printing on one or several layers in a row).

> Noted for that, do i still need to make the speed unique? (its already unique in my settings)

Nope.  Just make sure if you are using Prusa Slicer that you should turn on Verbose gcode output (it's in the guides, and in the help under 'Automatic' slicer settings option in your printer profile) so that Octolapse can detect the print features.  It works with Cura and Simplify 3D out of the box.  The smart layer triggers use this to prevent snapshots from occurring over exterior perimeters, while bridging, etc.  Octolapse prefers to take snapshots while printing wipe towers, over infill, or interior perimeters (in that order), to reduce the print quality impact.

Does that all make sense?

roomme13, 2020-05-06:
> It did NOT work in all cases. There was an issue even in some earlier dev versions of Octolapse 0.4! A very knowledgeable user tracked down the exact problem for me, which made solving it quite easy. I wish I could remember the username, because it was a fantastic issue report!

That awesome!, in 0.3.4 i tried both options for that setting, but its doesn't work at all, but in 0.4 WOW!

> So, the G90/G91 influences extruder setting has a HUGE impact on extrusion. If the axis modes are off very bad things can happen. To make matters worse, slicers like to use G92 a lot to reset the E [...]

This issue only happen on Marlin2.0 ? infact i used G92 in my layer gcode, dont know it would make an impact too.

> I'm not sure what you mean here, but the G90/G91 setting is there so that octolapse can retract properly and send the correct retraction/deretraction command before and after the snapshot.

my e step its already bellow calibrated value, that's why i`m confused you said over extrusion

> Again, are you 100% sure it was too high, or did it just stop extruding for most or all of a layer? The G90/G91 setting can cause both over extrusion (huge blobbs) and under extrusion (no printing on one or several layers in a row).

Both, I print vase mode and it make the spaghetti, i believe that time i use Octoprint preference (false) for G90/G91 setting, then i tested with true value for benchy, then no extrusion or the over extrusion (not smooth layers) happened.  thats before i install 0.4 devel.

> Nope. Just make sure if you are using Prusa Slicer that you should turn on Verbose gcode output (it's in the guides, and in the help under 'Automatic' slicer settings option in your printer profile) [...]

Noted, that's already setup in my slicer. i got first print with octolapse after upgrade 0.4!

Thanks Man!

caughron, 2020-05-06:
Haven't had a chance yet, just got my printer up and running again last
night after some hardware upgrades. Will work on it this week and report
back. Thanks for checking.

FormerLurker, 2020-05-06:
> This issue only happen on Marlin2.0 ? infact i used G92 in my layer gcode, dont know it would make an impact too.

No, originally it was a smoothieware thing I believe.  I'm not sure what the history of which firmwares use this setting and which do not and when they changed, but it was uncommon enough that I almost never heard about the kinds of issues from this thread.  Recently there has been a HUGE uptick of confused users (and developers).  Honestly, I HATE that behavior and think it just adds pointless confusion.  If ALL firmwares behaved that way it would be fine, or if there was a command to set/retrieve the behavior it would also be fine.  As it is now it just confuses absolutely everyone.

> my e step its already bellow calibrated value, that's why i`m confused you said over extrusion

Ah yes, it's over extrusion in the sense that the wrong value for E in G1 (G1 E{some number here}) was being sent.  If Octolapse thinks you are in relative E mode it might send G1 E-0.6 to retract, but in absolute units that would rewind your filament, and probably cause it to skip and grind.  The correct command *might* have been G1 E93923.3 or something :)
> Noted, that's already setup in my slicer. i got first print with octolapse after upgrade 0.4!

Fantastic!  I love it when people follow the guides :)

> Thanks Man!

You are welcome!

caughron, 2020-05-07:
Quick update. Part of my upgrade was to Marlin 2 and after reading other
posts in this thread, felt I might have other issues moving forward.
Working off your G90/G91 info, I decided to play a little more with v0.3.4.
Long story short, I set the setting in X/Y/Z Status mode from require
explicit G90/G91 to default to absolute. Voila - problem fixed. The app is
now printing as expected. moving to position to take picture, then returns
right to print and resumes without any artifacts. Not presuming to have any
educated idea why, it seems that changing this setting has removed whatever
code injection that was causing the issue.

I will plan to still move up to v0.4 as I like some of the abilities of it.
My question is if it is too buggy, is it simply a matter of removing the
0.4 version and reverting back to the 0.3.4 version which is now working?

Thanks for the help and the hard work on this app - very awesome
functionality. I'll be jumping in the Patreon group to show support.

FormerLurker, 2020-05-07:
The absolute axis thing is a relic of when I knew much less about gcode than I know now.  All of the current  profiles now default to absolute.

0.4 is definitely not more buggy that 0.3.4.  the reason it hasn't been released yet is that it has so many features they are difficult to test, and that tye installation is extremely complicated (dynamic compilation of c++ code for python is involved, plus python 3 compatibility).  I'm trying to make sure I don't get drowned in issues.  My number 1 suggestion for fixing octolapse issues is to try the new RC since most people who post on GitHub have above average understanding, and it generally works better than 0.3.4.  that before said, it is not problem free.  I don't think you will want to downgrade after trying 0.4.

FormerLurker, 2020-05-08:
@caughron, it sounds like your issue was solved, so I'm going to go ahead and close this.  Thank you for posting!

closed 2020-05-08 by FormerLurker