FormerLurker/Octolapse #513: Pi camera V2.1 Custom image preferences.

sam01508, 2020-05-03:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: OCTOLAPSE_VERSION_GOES_HERE

#### Version of OctoPrint

OctoPrint Version: OCTOPRINT_VERSION_GOES_HERE

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES_OR_NO

#### What were you doing when the problem occurred

Setting up Octolapse
I am using a Pi Cam
Setting up camera profile
Attempting to enable 'Custom Image Preferences'

#### What should have happened?

I should be able to adjust my camera through the interface.

#### What happened instead?
I get this message: unknown server type: an unknown webcam server type ‘Werkzeug/0.16.1
Python/2.7.16’ is being used for the ‘Default Webcam’ camera profile. Currently only MJPEGStreamer is supported. Unable to apply custom image preferences.

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPrint 1.4.0 
Os Version: OS_VERSION_GOES_HERE OctoPrint OctoPi 0.17.

#### Printer model & used firmware incl. version

Printer Model: Ender 3
Printer Firmware TH3D unified

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2020-05-03:
You left out some important info.  What version of octolapse are you using?  0.3.4 doesn't support the picam's settings, but 0.4.0rc1 does.

sam01508, 2020-05-03:
Hi, Thanks for your reply, sorry I’m new to this still. I’ve updated to that version (0.4.0rc1.dev5) between posting and your reply and it still says that same message about werkzeug server.

FormerLurker, 2020-05-03:
Check out this guide: https://github.com/FormerLurker/Octolapse/wiki/Configuring-a-Raspberry-Pi-Camera

Hope it helps!

sam01508, 2020-05-04:
I followed the guide and watched your YouTube video on it and it’s still the same.

FormerLurker, 2020-05-04:
What other plugins are you using?  A few creat their own servers which do not support camera settings control.  I believe the spaghetti detective and octoprint anywhere do this, and maybe some others.  Try disabling all plugins but octolapse, reboot, then see if it works.

sam01508, 2020-05-04:
I have spaghetti detective, octoprint anywhere, firmware updater, full screen webcam view? I’m off to work now so I’ll do that when I return and let you know the results. All the mjpeg files are on the sd card when I view through WinSCP. Do you think it’s my raspberry pi? A setting that just needs changing maybe. I’m sorry to create you a problem it’s just my luck I’m afraid, I’ve subbed to you channel and have been watching your videos, thank you for your help.

FormerLurker, 2020-05-04:
No problem, whenever you get around to it.  However, I'm pretty sure it is those two plugins I mentioned.  Ive been meaning to find a workaround for that, but so far have not found a good solution.  Have a good day at work!

sam01508, 2020-05-04:
Hi, ok so I’ve disabled those two plugins as you suggested, camera tests ok. I get a different message now, “an unexpected error was raised while testing custom image preferences for Raspi-camera profile. 
I can also log on to the mjpeg streamer page using IP address of the pi and view the picture. I couldn’t do this before. I feel like progress has been made!

FormerLurker, 2020-05-04:
It definitely has been!  Did you add the driver to the modules file and change the camera type to USB as was explained in the guide?  I know you said you followed it, but just double checking since each step is important.

sam01508, 2020-05-04:
I did yes. In the section webcam set up, do you have to change the 127.0.0.1:8080? Also the line below does that need changing? Apart from those two lines I’ve followed the guide as is. Just this error now.

FormerLurker, 2020-05-04:
Try using 127.0.0.1/webcam

sam01508, 2020-05-05:
Hi, using the 127.0.0.1/webcam it can’t connect to the camera. Change it back it’s fine, just the other error.

sam01508, 2020-05-05:
Also mine looks like this and when you start a print nothing happens with octolapse in the 0.4.0rc1.dev5 set up it has different buttons At the top etc. I’ll try and attach a photo to show you, it my be correct but I thought I would mention it. I was trying to use test mode but a print just starts as normal. 
[image]

sam01508, 2020-05-05:
Just one or two more things I’ve noticed trying to save main settings I get an error to say unable to save internal server error. In version 0.3.4 today I couldn’t enable octolapse due to the same error.
It’s my luck as I said before. Sorry about this.

FormerLurker, 2020-05-05:
Try installing octolapse from the devel branch.  There was a bug in the main settings :(

sam01508, 2020-05-05:
Ok I’ll try whatever! Is that the second link on the page “getting started with octolapse v0.4?
I appreciate you taking the time to reply.

FormerLurker, 2020-05-05:
Yes, this one:  https://github.com/FormerLurker/Octolapse/archive/devel.zip

And no problem!  We will get it working, it just may take some time.

sam01508, 2020-05-05:
Ok, downloaded. Save settings error has gone. By the way it was just main settings that wouldn’t save. Octolapse tab is still the same as my previous picture no icons at the top with no preview. Camera test is ok, MJPEGStreamer test page is ok. Still have Unexpected error when clicking custom image preferences.

FormerLurker, 2020-05-05:
The tab is correct.  That's what it looks like now.  Also, that preview isn't going to show you the camera stream, it will show you the latest snapshot taken with your camera and possibly a preview of the timelapse depending on configuration and a few other things.  If you have multiple camera profiles, make sure the right camera is selected in the drop down box under directly under the timelapse preview.

Try this:  Edit your logging profile settings in the octolapse tab and click 'Clear Log...'.  Close the popup.  Now change your logging profile to 'Debug':

[image]

Now try enabling custom image preferences.  Once you get the error, open up your logging profile and click 'download log':

[image]

Now export your settings for your camera so I can take a look.  Click the 'Cameras' link on the tab here:

[image]

And export the correct profile here (whichever one you are using, if you are in doubt, export them all):

[image]

Then create a new gist at gist.github.com and paste in the contents of the log and the camera profile(s) (separate each please), then paste a link to them in here.

It wouldn't hurt for you to paste in your octopi.txt file as well as your modules file too (the ones you changed in the guide).

sam01508, 2020-05-05:
Ok I’ll give that ago and get it sent to you. Never done it before so bare with me. I was going by the pictures in the guide sorry about that. Did I mention earlier I had test made on but the print just started as normal. Probably something I haven’t done.

FormerLurker, 2020-05-05:
When you say started as normal, what do you mean exactly?  Did your printer warm up and extrude plastic?

sam01508, 2020-05-05:
Yes that’s what happened. I’ve downloaded to logs do I copy and paste into the gist or create a file I’ve never done one before!

FormerLurker, 2020-05-05:
I've not heard of test mode failing for anyone.  Are you 100% sure Octolapse was enabled?? If so I am stumped and would need to see logs.  However, one issue at a time.

[image]

1.  Name the Gist.  Call it whatever you want
2.  Name the file (just use the file name or something close enough so I know what it is).
3.  Paste in the contents of the file
4.  Click 'Add File' to add a new file (you already have one at this point), and repeat steps 1-3 for the new file.
5.  After all files have been added click create secret gist (no need for it to be public).
6.  Copy the link in your browser address bar and paste it in here.  Please check the link after you post to make sure it works.  If it doesn't edit your post and try again (you may need to hit F5 to refresh your browser within the gist page, but probably not).

sam01508, 2020-05-05:
Yes it was probably me, I’ll do it now.

sam01508, 2020-05-05:
Hopefully it is all you need just let me know if I've done it correctly. Link as follows:
https://gist.github.com/sam01508/afef1c5ad5d1b20c53a5633b0ede541f

FormerLurker, 2020-05-05:
Yes, just fine.  The first thing that jumps out at me is this:

```CameraError: mjpg_streamer_control_error: The server denied access to the mjpg-streamer control.htm```

I was going to look at your settings (the exported profile), but it looks like something went wrong with the copy paste.  Can you double check that and make sure the entire json file gets uploaded to that gist?  You shouldn't have any trouble editing it, but if you do just create a new one.

Also, while you are at it, is there any way you can get your Octopi.txt file in the gist as well?  My guess is the bottom part is incorrect.  It should look like this:

```
camera_http_webroot="./www"
camera_http_options=""
```
NOT like this:

```
#camera_http_webroot="./www"
#camera_http_options=""
```

Notice the # marks in front.  That indicates it's commented out and will not be used.  I could be wrong, but that's usually why the control.htm page is denied.  You should be able to access this page via a web browser after substituting in your PI's IP address:

```http://{IP_OF_PI}:8080/control.htm```

or maybe

```http://{IP_OF_PI}/webcam/control.htm```

If you can't get to that page, Octolapse won't be able to control your settings.  Anyway, let me know how it goes.

FormerLurker, 2020-05-05:
Oh, if you change octopi.txt, make sure you reboot the pi completely (not just octoprint restart) else the changes wont be applied.

sam01508, 2020-05-05:
Ok I’ll look in to that and get the logs over to you again.

sam01508, 2020-05-05:
The hashtags were still there, I did a complete install the other day when things went wrong! Now when you click (address of pi)8080/control.htm the page comes up. It says unknown control type but if you press ok that goes away. Files to follow.

FormerLurker, 2020-05-05:
Unknown control type!?  Check the camer guide again..  go through it all very very carefully.  You are probably missing the module file change, and probably tye usb change in your octopi.txt file.  Again, every step is crucial.  Go slow and double check everything, then it should work

FormerLurker, 2020-05-05:
Hold on, i may be wrong about the modules file and the USB setting!!  If the error message you saw went away AND many controls appeared, then it should be working!!  Sorry to have maybe jumped the gun on that before I thought it through...

sam01508, 2020-05-05:
Yep it does work though, you can control the camera brightness etc. Custom image preferences brings up 4 new lines now with tick boxes, no errors. I’ll get you the octopi.txt file. The other logs are the same as what I sent. What else should there be?
[image]
[image]
[image]

FormerLurker, 2020-05-05:
That means it worked!!! Now go to the octolapse tab and there should be a new edit pen under the timelapse preview to the right of the camera drop down box (check the help links in the camera profile next to the image preferences if you need any additional explanation to find the settings.  Let me know if the documentation needs improvement).  If everything is working you should see a camera stream and lots of settings.  Look everything over and don't be afraid of clicking the help links.  Let me know how it goes!

sam01508, 2020-05-05:
Yes that’s all there lots settings. I’ll go through everything again tomorrow and get back to you. So when I start a print now it’ll just kick in and start taking snap shots or do I have to do anything I did add G4 P1 I think to the Gcode. Any other changes now?

FormerLurker, 2020-05-06:
No gcode changes required.  I suggest starting with the smart layer trigger (the compatibility one).  Make sure to check out the automatic slicer settings in your printer profile (read the help!), It is a great feature 👍.  If you are using prusa slicer, enable verbose gcode (this is all in yye getting started with octolapse 0.4 in the wiki).  Once that is done it should be smooth sailing!

sam01508, 2020-05-06:
Been through most of it now, all looks good and the same as the guide. How silly was I, just goes to show when you look at something a lot and you’re sure it’s right it can still be wrong. Got to sort out the focus but I’m getting there.
When I start to print now it warms up and print starts, pluggin is enabled and so is test mode. Is there something I’m not doing correctly. What should happen? Sorry I’m a pain aren’t I.

sam01508, 2020-05-06:
Using Prusa slicer and enabled verbose setting. I’ve taken out the G4 P1. I’ll try it again after I go through  octolapse 0.4 Wiki. I tell you this sort of stuff only happens to me. I was following a YouTube guide before that’s how I ended up with the G4 P1 in the gcode.

sam01508, 2020-05-06:
I've updated the gist with the log from last night and this morning. Could you outline what should happen in test mode. Then what happens when test mode is off. Sounds funny doesn't it but sometimes if I don't know exactly what happens I think its wrong or over think it. 
I'll add my start gcode now too.

FormerLurker, 2020-05-06:
@sam01508, I'm very glad the camera is working, but it sounds like something is up with test mode.  it's supposed to print without warming up or extruding plastic.  Your printer should just go through the motions, but certain gcode changes should happen.  Temperature and fan control gcodes should be suppressed.  E axis parameters should be stripped from G0, G1, G2 and G3 commands, and a few other things should happen.  I will review the log, but perhaps you could create a new issue called something like 'Printer warms up and extrudes in test mode'?

Also, fyi, if test mode isn't working I have a feeling a few other things might not work.  Test mode has been a reliable feature for years, but there have been massive changes to the core gcode handling routines, which is why 0.4.0 is in the release candidate phase.  One step at a time :)

FormerLurker, 2020-05-06:
Oh, when you create the new issue, perhaps you can provide a new log?  Set logging to 'debug', clear the current log like you did last time, then start a print in test mode.  After it warms up, homes, then prints for maybe two minutes cancel the print, download the log, change your logging profile back to 'Log All Errors'.  Then post the new log to that gist and link to it in the new issue.  Finally, get on with your day :)  I should be able to see what's going on with that log.

Can I close this issue since the camera problems have been solved, or do you still have concerns about that?

sam01508, 2020-05-06:
I think its safe to say the camera issue is resolved. I'm going to do as you've suggested and i will create a new issue with that. Mind you I went ahead and took test mode off and started a print and nothing happened with octolapse. I will get the new log to you asap.

FormerLurker, 2020-05-06:
Resolved, closed!  Thanks for creating the new issue.  Will get to it ASAP.

closed 2020-05-06 by FormerLurker