FormerLurker/Octolapse #514: Octolapse cuts off the last few minutes of the print

hughdaniel, 2020-05-06:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: OCTOLAPSE_VERSION_GOES_HERE

#### Version of OctoPrint

OctoPrint Version: OCTOPRINT_VERSION_GOES_HERE

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES_OR_NO

#### What were you doing when the problem occurred

1. STEP_ONE_GOES_HERE
2. STEP_TWO_GOES_HERE
3. STEP_...

#### What should have happened?
PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: OS_NAME_GOES_HERE
Os Version: OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: PRINTER_MODEL_GOES_HERE
Printer Firmware Version: PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2020-05-06:
Not much info to go on here, but if I had to guess i would say you are using octolapse 0.3.4 using the high quality layer trigger.  Try the compatibility one.  Feature detection doest work all that well due to feedrate changes in small layers (filament cooling settings).  That has been replaced with a new trigger that doesn't have this problem in v0.4.0rc1

hughdaniel, 2020-05-08:
Thank you for your response.

I tried the compatibility setting as you suggested and it appears to have solve the problem.

I’m a 74 year old who has recently got into 3D printing, and people like yourself enable me to move forward, be it at a slow pace.

Take care of yourself and others.

Hugh Daniel.

FormerLurker, 2020-05-08:
Absolutely my pleasure Hugh!  These interactions are exactly why I do what I do.  You keep learning and progressing and I promise I will do the same.  Don't hesitate to create another issue if you have problems, or email me at formerlurker@protonmail.com if you have any questions, thoughts, suggestions, or if you just feel like saying hi!

FormerLurker, 2020-05-08:
Forgot to close the issue, thanks!

closed 2020-05-08 by FormerLurker