FormerLurker/Octolapse #515: Printer warming up in test mode.

sam01508, 2020-05-06:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: OCTOLAPSE_VERSION_GOES_HERE

#### Version of OctoPrint

OctoPrint Version: OCTOPRINT_VERSION_GOES_HERE

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES_OR_NO
Yes

#### What were you doing when the problem occurred

1. Set printer for new print in test mode.

#### What should have happened?
Printer go through the motions but not actually warm up or extrude plastic.

#### What happened instead?
Printer started to heat immediately and start the print as it would without test mode on. Even with test mode off octolapse did not take any snapshots. 

#### Operating System running OctoPrint and Octolapse

OS Name: Octoprint 1.4.0
Os Version: V0.4.0rc1+u. 163be9f

#### Printer model & used firmware incl. version

Printer Model: Ender 3
Printer Firmware Version: TH3D

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

sam01508, 2020-05-06:
Gist updated. Might not be right it just seems to be about the camera again.

FormerLurker, 2020-05-06:
Once you get it straightened out, please post a link to the gist in this thread, thanks!

sam01508, 2020-05-06:
Ok I’ve added to the previous one did you want a new gist as well.

FormerLurker, 2020-05-06:
Hmm, I don't see any logging in there from a print.  It should be a pretty huge file containing all the gcodes sent and received.  After looking at it, yes please create a new gist so we can cleanly separate the issues.  Also, please update the ticket with the octoprint version and add links to your new gist for the settings.json and gcode file.  You can export all settings from here:

[image]

[image]

Thanks!

sam01508, 2020-05-06:
https://gist.github.com/sam01508/e9699f490848de3d06ecdbda209b8727 try this.

FormerLurker, 2020-05-06:
Settings are there.  Can you try creating a log as I described earlier?  

I quick ran a print with your settings and test mode worked :(  I have a feeling this will be tricky to solve.  what version of OctoPi are you using (update the ticket with that)?   I may have to flash a fresh SD card with your exact instance.  Also, are any other plugins enabled?  Perhaps there is a conflict somewhere.

sam01508, 2020-05-06:
Octopi version is 0.17.0 
Let me try the log again, it only seems to be the what relates to the camera.

sam01508, 2020-05-06:
Other plugins are action command prompt, and smart heat.

FormerLurker, 2020-05-06:
So, using the default 'debug' logging profile and starting a print in safe mode results in no logging of the print commands?  Sounds like something is preventing Octolapse from seeing the gcodes.  Try disabling all other plugins and try again.

sam01508, 2020-05-06:
I've updated the gist again, less than the other times, I might flash another card myself too see what happens. I'll try disabling the other plugins first, some it says not advisable to shall I just do it anyway to test it.

FormerLurker, 2020-05-06:
Just disable the two you installed, else you may run into issues

sam01508, 2020-05-06:
Ok it is just smart pre heat the action command prompt is this....
You are about to disable "Action Command Prompt Support".

Without this plugin your printer will no longer be able to trigger confirmation or selection prompts in OctoPrint

Do you still want to disable it?
Shall I disable it anyway?

FormerLurker, 2020-05-06:
Leave that one alone.  Just 3rd party plugins, so smart preheat.  You can re-enable it after testing the effect of disabling.

sam01508, 2020-05-06:
I've added two more to the gist.

FormerLurker, 2020-05-06:
My last comment didn't show up, sorry (if you were notified, I apologize for the duplicate message).  There are only a couple lines in the log and there should be 1000s.  Recheck your logging profile and make sure it is set to 'debug' and that all loggers are enabled and set to debug level.  Then try a print in test mode again and see if you get a longer log (it should contain every gcode sent from the printer, all of the printer's responses, all of the gcode Octolapse created to take snapshots, all preprocessing and slicer settings messages.. basically it will be MASSIVE).

Also, I did a fresh install with the latest version of OctoPi, upgraded OctoPrint to 1.4.0, installed octolapse from the devel branch, and everything worked :(  I would really like to figure out what is going on here since you will most definitely not be the only one to have these issues (if 1 person reports a bug, 100s of others have had it too).

sam01508, 2020-05-06:
I’ve also done a fresh install exactly the same. When you say all loggers what do you mean? Mine is set as in the pictures. 
I get the same log from this install too, I’ll add it to the gist.
[image]
[image]

FormerLurker, 2020-05-06:
Well, if you did a fresh install of OctoPi AND Octolapse from that devel branch link I don't see how it could be wrong.  Here is what I am talking about:

[image]

That is in the login profile if you click 'Customize Profile' and scroll down.

I am just baffled how two fresh installs of the same versions could behave so differently....  Maybe you could record your screen while you try the test mode print with logging?  Perhaps I can spot some differences.  

The logger AND test mode is failing, both errors I have never seen before and both at the same time.  At this point I have no idea what is going on or how to deal with it.  We will get there, but expect this to be a LONG LONG thread :(

sam01508, 2020-05-06:
I didn’t have addional logging checked. Now I have this list.
[image]

FormerLurker, 2020-05-06:
Holy cow!  And you never changed the profile at all?  I wonder how that could be.  I checked the profile repository, and 'Enabled' is set to true for that profile (additional logging check box).  It is also true in the default settings files (there are two for various reasons, settings_default_0.4.0rc1.json and settings_default_current.json).

This just keeps getting stranger and stranger..  I'm starting to question this whole build.

FormerLurker, 2020-05-06:
Hey, after we get this figured out (it can be after this ticket is closed), would you be willing to re-flash and re-install, then immediately when you open the octolapse tab export all of your settings.  I absolutely need to figure out how that 'additional logging' setting was unchecked in the debug profile.  That will cause a lot of trouble down the road for everyone if it isn't figured out.

sam01508, 2020-05-06:
Yes of course at this stage i'll do whatever. Now I have done a test again and downloaded the log and its only two lines again.

FormerLurker, 2020-05-06:
So, maybe time for an online meeting?  Can I send an invite via email?

sam01508, 2020-05-06:
I've got an octoprint log would that help? It contains  ip addresses  is it secure on the gist?

sam01508, 2020-05-06:
I've updated gist.
Yes send it over no problem

FormerLurker, 2020-05-06:
Great!  Send me an email to formerlurker@protonmail.com and I will reply with a link.

FormerLurker, 2020-05-06:
Update for anyone reading this:  The issue was @sam01508 was printing from SD, but received NO warnings at all.  Octolapse is supposed to report that it will not work while printing from SD, but it did Not do that.  See issue #516.

Thanks @sam01508!  Closed!

closed 2020-05-06 by FormerLurker

sam01508, 2020-05-06:
No problem it was a pleasure.