FormerLurker/Octolapse #519: Feedback for v0.4.0rc2

FormerLurker, 2020-05-10:
Please see the release notes [here](https://github.com/FormerLurker/Octolapse/releases/tag/v0.4.0rc1.zip). This includes a list of enhancements, bug fixes, and issues known at the time of release.

All feedback is welcome here!  Let me know what you like, what you hate, what works, what is broken, etc. 

I would like to verify that nothing obvious is broken with OctoPrint v1.4.0, and if possible *Python 3*.  If anyone adventurous has *Python 3* installed and wants to help test I would be most appreciative.

You can install this version of Octolapse using this URL in the OctoPrint plugin manager:
```
https://github.com/FormerLurker/Octolapse/archive/v0.4.0rc2.zip
```
Please [read this guide completely](https://github.com/FormerLurker/Octolapse/wiki/V0.4-Getting-Started-with-Octolapse) before installation.  

Thanks!

nebbywan, 2020-05-11:
Failed on build, here is the output.

[Octolapse.txt](https://github.com/FormerLurker/Octolapse/files/4606980/Octolapse.txt)

Running on an Ubuntu 20.04 LTS box if it matters.

FormerLurker, 2020-05-11:
You need to install the dev python package.  Let me know if you need help with the command.

nebbywan, 2020-05-11:
That was it. Thank you sir!

steve220, 2020-05-11:
Installed on 3 PI's this morning. All went well and all showing the correct version.
Will try a time lapse this morning.

steve220, 2020-05-11:
Time lapse worked great!

FormerLurker, 2020-05-11:
@steve220, thank you so much for giving this new rc a go!  I love getting some, "It worked!" feedback :)  I'm going to give it a week or two, and hopefully this can be promoted to v0.4.0!

biga888, 2020-05-11:
same here, no issues :) thank you!

Archer36, 2020-05-12:
Upgraded with no issues. Fixed the "Main Settings Save Error" when trying to save updated main settings.

Ant8490, 2020-05-17:
Everything worked for me running on Debian 10.  Fresh install and imported saved settings from rc1. 
 I did a time lapse test with a Nikon D90:  https://www.youtube.com/watch?v=WYzKNPHY2iQ
This was my first full run of a time lapse, so I will continue to play around with the settings.

rolinge, 2020-05-18:
Installed on a Raspberry Pi 2B with a PiCamera V1 and klipper on Tevo Tornado printer.  Installation successful, but as you can guess it took a long time (because of Pi2).  

I would highly recommend adding the statement about installing python3-dev to the installation instructions when installing from a specific branch in the git repo.  My first install failed, but I saw the posting above.

Doing a print now.  Will update when done.

nyanroberts, 2020-05-19:
Just installed the latest update, and had been using the smart features in the previous version with great success. I found that in vase mode my "seam" wasn't perfectly straight as it went up the model, which I'm only noting because it causes the resulting timelapse to not appear perfectly stable, still it's an amazing feature. Please check out my posts (link in profile), and if there is a preferred way to credit you, I would love to tag you in all my octolapse posts. Otherwise I'll just use #Octolapse Thanks!

TattooedHead, 2020-05-19:
Just installed it, and seems to be working great. No more issues with Position error I was getting and the Interface is slick! I'm loving the preview it gives before it starts. That's pretty awesome. It gives me a chance to make sure it's where I want everything to be.

Thanks for the new version!

FormerLurker, 2020-05-21:
@TattooedHead, I believe you probably hit the timeout value for the stabilization.  It's a couple minutes I believe.  It only happens if you just happen to pause Octolapse while it is waiting to take a snapshot, or in the middle of a snapshot.  In general things work out right, but it's just possible that a critical command stalls out, messing up the next gcode.  I wanted to add a warning so that the user knows something potentially bad happened.

I've got some ideas for dealing with this issue, but since I've been working on this release for 2.5 years now, I figure it's best to get it out the door and deal with these types of issues later.

Thanks for your compliments, and for your report!  I'm glad it's working for you for the most part!

FormerLurker, 2020-05-21:
@rolinge, regarding this comment:

> Hopefully, however, this creation is run with the nice command, so as not to interfere with the printer I/O.

The printer IO is not affected very much by the timelapse preview.  This is because only 1 image is downloaded to your client per snapshot.  The timelapse is a 'fake' video that is made from images cached by your browser from previous snapshots.  You will notice that if you refresh your browser or switch browsers, the timelapse preview video isn't available.  In this case it loads the most recent snapshot, with is stored on the server for each camera.  Octolapse basically stores a stack of cached images, and shows/hides them in the proper order to simulate a video.  You can actually configure the maximum number of frames Octolapse will store in the main settings.  The default value is pretty conservative, so I usually crank mine up a bit.

FormerLurker, 2020-05-21:
@nyanroberts, @FormerLurker1 on twitter.  #Octolapse works fine too :)  I'm not on twitter much.

bitshiftr, 2020-05-30:
First of all, I want to thank you for all the effort and time you have put into this plugin - it's phenomenal!

I just upgrade from the main branch build to this and printed a baby groot. I saw a lot of excess extrusions on the back side of the groot (https://www.thingiverse.com/thing:2014307) and on the top of the head. I don't mind experimenting various triggers and stabilization options - what settings would you recommend to try out? 

I have now moved to automatic slicer (and imported the server Ender 5 profile). Which trigger option would you recommend? I had used a custom snapshot trigger of G4 P1 (based on TeachingTech video), but now that I am using the server based printer profile for Ender 5, I am not sure if this custom snapshot code is still being used.

FormerLurker, 2020-05-30:
@bitshiftr,

> First of all, I want to thank you for all the effort and time you have put into this plugin - it's phenomenal!

Thank you very much for these kind words :)

>  I saw a lot of excess extrusions on the back side of the groot

I've spent the last two weeks (and all day today) rewriting the wiki, and have created a page for [troubleshooting print quality problems](https://github.com/FormerLurker/Octolapse/wiki/V0.4---Troubleshooting-Print-Quality-Issues).  Perhaps take a look at that?  It would be nice to get some feedback on the new documentation, though it might be changing pretty rapidly today.

> Which trigger option would you recommend

Definitely start out with the *Smart - Compatibility* trigger.  Check out the [getting started](https://github.com/FormerLurker/Octolapse/wiki/V0.4---Getting-Started) guide for a complete walk through of a first print.  I have recently **massively** overhauled this page.  Even though you probably already know most of what is explained in the guide, I bet you will learn some new things that you will find useful!

> I had used a custom snapshot trigger of G4 P1 (based on TeachingTech video), but now that I am using the server based printer profile for Ender 5, I am not sure if this custom snapshot code is still being used.

That is a **very** popular video, and is a great guide, but things have changed a lot in Ocolapse since it was made, and I don't really recommend that the ```G4 P1``` method.  Not because it's a bad method, but because I think the smart triggers will perform much better for most users.  There is a new **Smart Gcode** trigger that will also give you a nice preview and snapshot count *before* your print starts, so if you must use the gcode trigger, use that one instead of the *classic* version.

Also, I just detected and corrected an issue with the ```G4 P1``` method.  A previous guide used ```G4 P0```, and I had entered this in as the default snapshot command for all server profiles.  However, that command means *Wait for All Print Moves to Finish* in some firmware (quite strange, but that's the way it is), so it was changed at some point.  I updated the server profiles, but there was a bug in the update code that prevented this field from being automatically updated.  I have corrected that and it will be included in the next release.

bitshiftr, 2020-05-31:
Appreciation is the least I can do for everything that you are doing for the community out of your own time.

> I've spent the last two weeks (and all day today) rewriting the wiki, and have created a page for [troubleshooting print quality [...]

Yes, I did notice a bunch of edits today. I actually went through the wiki while I was setting up profile with the latest RC release. The most recent update to the wiki is quite comprehensive, so this will be extremely helpful. 

For those of us who have followed some of the popular guides out there, if you can put in a small section that describes the major changes in Octolapse since then and why some of these settings might not be appropriate any longer will be quite helpful.
> Definitely start out with the _Smart - Compatibility_ trigger. Check out the [getting started](https://github.com/FormerLurker/Octolapse/wiki/V0.4---Getting-Started) guide for a complete walk [...]

So, here is the weird part - I saw the recommendation on the wiki regarding "smart compatibility", but I was not able to find it in my drop down menu in the "triggers" section. I thought maybe it was renamed to something else, but now that you mentioned it again, I went back and still don't see it in the drop down menu, but if I go to the main settings page of Octoprint and I go into the settings for Octolapse and then to "Stabilization" -> add profile. I see "Smart - Compatibility" as one of the profiles that I can import. I will go ahead and create a new profile and import this trigger for now. I am seeing the same issue with Stabilization too - "centered" option doesn't show up in the  Stabilization drop-down unless I create a new profile and import this.

> That is a **very** popular video, and is a great guide, but things have changed a lot in Ocolapse since it was made, and I don't really recommend that the `G4 P1` method. Not because it's a bad [...]

Got it - I will remove this custom code from my slicer "layer change script".
> Also, I just detected and corrected an issue with the `G4 P1` method. A previous guide used `G4 P0`, and I had entered this in as the default snapshot command for all server profiles. However, that [...]

cool - can't wait to test out the new version when it's out. It looks like you are getting really close to a GA release of 0.4 :)

FormerLurker, 2020-05-31:
@bitshiftr, I've added some information to the [getting started guide](https://github.com/FormerLurker/Octolapse/wiki/V0.4---Getting-Started) and to the [upgrade instructions](https://github.com/FormerLurker/Octolapse/wiki/V0.4---Installing-and-Upgrading-Octolapse#upgrading-from-a-previous-version) regarding missing profiles when upgrading from a previous version.  Let me know if this would have helped you.

I'm going to be testing upgrades from V0.3.4 today, so it's possible there are some settings migration errors.  From your comments it looks like either you are confusing the *stabilization* and *trigger* profiles, or perhaps there is some kind of bug in the settings migration.  Either way, with a little luck I'll figure this out today.  I'm really hoping to push this upgrade out finally!

bitshiftr, 2020-06-01:
> @bitshiftr, I've added some information to the [getting started guide](https://github.com/FormerLurker/Octolapse/wiki/V0.4---Getting-Started) and to the [upgrade [...]
cool - will take a look and let you know
> I'm going to be testing upgrades from V0.3.4 today, so it's possible there are some settings migration errors. From your comments it looks like either you are confusing the _stabilization_ and [...]

I don't think it's confusion on my part, but after you have done a bunch of testing for various upgrade scenarios, we will have a more comprehensive view. The issue I am describing is quite simple - after I customized my instance of Octolapse 0.3.4 (based on TeachingTech video), when I upgraded to the latest 0.4.0 rc2, I was missing certain stabilization and trigger profiles (centered, smart - compatibility) from the drop down menu in the Octolapse tab. I had to go to the main settings page and create new profiles and import the ones mentioned above. After that, I am able to see these profiles and choose them from the drop down menu. Hope I was able to elaborate better this time around.

FormerLurker, 2020-06-01:
Closing this issue now, THANK YOU everyone who commented here.  I am so grateful!

I scrubbed my planned V0.4.0 release this weekend because of two pretty major bugs I fixed.  I vacillated between just releasing it anyway and adding a new RC.  In the end I decided to drop another RC since I wouldn't be deploying until the weekend anyway.  If anyone is interested in testing any of the bug fixes, please see the new [feedback issue here](https://github.com/FormerLurker/Octolapse/issues/530).

Once again, **thank you all so much**!

closed 2020-06-01 by FormerLurker