FormerLurker/Octolapse #520: Octolapse appears to ignore settings in gcode

kbuley, 2020-05-13:
#### If this is a feature request describe it here

.

#### Version of Octolapse
 
Octolapse Version: 0.4.0rc2

#### Version of OctoPrint

OctoPrint Version: 1.4.0

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: No

#### What were you doing when the problem occurred

1. Slice a model with Cura 4.6
2. Send to Octoprint
3. Attempt to print the model

#### What should have happened?
The model prints

#### What happened instead?
`octolapse.__init__ - ERROR - Some slicer settings were missing from your gcode file.  Missing Settings: Retraction Length,Retraction Speed,Deretraction Speed,Z Lift Height,X/Y Travel Speed,Z Travel Speed`

Enabled verbose logs, Cura inserted appropriate values:

```Process type: forward, line: 31, gcode: ;speed_z_hop = 5
Process type: forward, line: 32, gcode: ;retraction_amount = 5
Process type: forward, line: 33, gcode: ;retraction_hop = 0.2
Process type: forward, line: 34, gcode: ;retraction_hop_enabled = False
Process type: forward, line: 35, gcode: ;retraction_enable = True
Process type: forward, line: 36, gcode: ;retraction_speed = 45
Process type: forward, line: 37, gcode: ;retraction_retract_speed = 45
Process type: forward, line: 38, gcode: ;retraction_prime_speed = 45
Process type: forward, line: 39, gcode: ;speed_travel = 150.0
Process type: forward, line: 40, gcode: M201 X1000.00 Y1000.00 Z1000.00 E5000.00 ;Setup machine max acceleration
Process type: forward, line: 41, gcode: M203 X500.00 Y500.00 Z12.00 E120.00 ;Setup machine max feedrate
Process type: forward, line: 42, gcode: M204 P2000.00 R1500.00 T200.00 ;Setup Print/Retract/Travel acceleration
Process type: forward, line: 43, gcode: M205 X8.00 Y8.00 Z0.40 E5.00 ;Setup Jerk
Process type: forward, line: 44, gcode: M205 S0 T0 ; sets the minimum extruding and travel feed rate, mm/sec
Process type: forward, line: 45, gcode: M220 S100 ;Reset Feedrate
Process type: forward, line: 46, gcode: M221 S100 ;Reset Flowrate
Process type: forward, line: 47, gcode: M117 Homing...
Process type: forward, line: 48, gcode: G21 ;units to mm
Process type: forward, line: 49, gcode: G90 ;absolute coordinates
Process type: forward, line: 50, gcode: G28 ;Home
Process type: forward, line: 51, gcode: M117 Setting bed temp
Process type: forward, line: 52, gcode: M190 S50 ; Set bed temp.
Process type: forward, line: 53, gcode: M140 S50 ; Wait for bed temp.
Process type: forward, line: 54, gcode: M117 Setting nozzle temp...
Process type: forward, line: 55, gcode: M109 S200 ; Wait for all used extruders to reach temperature.
Process type: forward, line: 56, gcode: M117 Wiping...
Process type: forward, line: 57, gcode: G1 Z0.2 F1200 ; raise nozzle 0.2mm
Process type: forward, line: 58, gcode: G1 Y10 ; move to prep for purge
Process type: forward, line: 59, gcode: G92 E-30; reset filament position
Process type: forward, line: 60, gcode: G1 F200 E0 ; prime
Process type: forward, line: 61, gcode: G1 X100 E12 F600 ; move X-carriage 100mm while purging 12mm of filament
Process type: forward, line: 62, gcode: G92 E0 ; reset extrusion distance
Process type: forward, line: 63, gcode: M117 Making...
Process type: forward, line: 64, gcode: G92 E0
Process type: forward, line: 65, gcode: G92 E0
Process type: forward, line: 66, gcode: G1 F2700 E-5
Process type: forward, line: 67, gcode: ;LAYER_COUNT:15
Process type: forward, line: 68, gcode: ;LAYER:0
Process type: forward, line: 69, gcode: M107
Process type: forward, line: 70, gcode: G0 F6000 X178.585 Y122.498 Z0.2
Process type: forward, line: 71, gcode: ;TYPE:SKIRT```

#### Operating System running OctoPrint and Octolapse
<!--
OctoPi, Linux, Windows, MacOS, something else?
Include the version number if possible.
OctoPi's version can be found in /etc/octopi_version or in the lower left
corner of the web interface.
DO NOT OMIT
-->
OS Name: OS_NAME_GOES_HERE
Os Version: OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version
<!--
OMIT ONLY IF IT IS DEFINITELY NOT NEEDED.
-->
Printer Model: PRINTER_MODEL_GOES_HERE
Printer Firmware Version: PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser
<!--
OMIT ONLY IF IT IS DEFINITELY NOT NEEDED
-->
Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred
<!--
Many problems I've seen are due to specific gcode files.
Sometimes it's a bug, sometimes it can be slicer settings,
other times it can be Octolapse or OctoPrint setting related.

In any case, the gcode file is often very useful.
On gist.github.com or pastebin.com.
OMIT ONLY IF IT IS DEFINITELY NOT NEEDED
!-->
Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json
<!--
Lots of problems are settings related, and having them available
makes debugging a lot easier.  You can export the settings.json
file from within the Octoprint settings.

BE AWARE - IF YOU HAVE ANY PASSWORDS IN YOUR CAMERA PROFILE,
THEY WILL BE EXPORTED ALONG WITH THE REST OF YOUR SETTINGS!
IF YOU ARE USING CAMERA AUTHENTICATION IN ANY OF YOUR PROFILES,
EITHER REMOVE THE PASSWORDS BEFORE EXPORTING, DELETE THEM FROM THE FILE
AFTER EXPORTING, OR DON'T POST SETTINGS.JSON!  You can search for

"password":

to find all of the entries containing passwords.
OMIT ONLY IF YOU HAVE ENTERED A PASSWORD TO ACCESS YOUR CAMERA
WITHIN OCTOLAPSE
!-->
Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log
<!--
On gist.github.com or pastebin.com.

DO NOT OMIT IF YOU ARE USING OCTOLAPSE V0.2.2 OR LATER.

IF YOU ARE USING V0.2.1 OR EARLIER DO NOT POST YOUR LOGFILE
SINCE IT MAY CONTAIN CAMERA AUTHENTICATION PASSWORDS IF THEY WERE
ENTERED.
-->
Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log
<!--
On gist.github.com or pastebin.com.
DO NOT OMIT
-->
Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser
<!--
If you are having UI problems, like buttons not working, or problems saving
settings, please inlude the javascript console output
on gist.github.com or pastebin.com.
OMIT ONLY IF IT IS DEFINITELY NOT NEEDED
-->
Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:
<!--
This is really nice to include, especially if you're having trouble
putting your issue into words.  A picture speaks 1000 of them, afterall!
-->
Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2020-05-13:
Can you export your settings (octolapse tab->gear icon->export settings)?  From the look of it it is not detecting any settings.  Could it be that your printer profile is set to cura 4.2+ and not to automatic?

kbuley, 2020-05-13:
```
{
    "main_settings": {
        "show_navbar_icon": true,
        "show_navbar_when_not_printing": true,
        "is_octolapse_enabled": false,
        "auto_reload_latest_snapshot": true,
        "auto_reload_frames": 30,
        "show_printer_state_changes": true,
        "show_position_changes": true,
        "show_extruder_state_changes": true,
        "show_trigger_state_changes": true,
        "show_snapshot_plan_information": true,
        "cancel_print_on_startup_error": true,
        "platform": "unknown",
        "version": "0.4.0rc2",
        "git_version": "3b59d445f283ed7228959f9fcf67666edf9d305c",
        "preview_snapshot_plans": false,
        "preview_snapshot_plan_autoclose": false,
        "preview_snapshot_plan_seconds": 30,
        "automatic_updates_enabled": true,
        "automatic_update_interval_days": 30,
        "snapshot_archive_directory": "",
        "timelapse_directory": "",
        "temporary_directory": "",
        "test_mode_enabled": false
    },
    "profiles": {
        "options": null,
        "defaults": null,
        "printers": {
            "4cad9937-6e92-4bfd-9eda-6186ed16045b": {
                "name": "Sunlu S8",
                "description": "",
                "guid": "4cad9937-6e92-4bfd-9eda-6186ed16045b",
                "automatic_configuration": {
                    "key_values": [
                        {
                            "name": "Creality3d",
                            "value": "creality3d"
                        },
                        {
                            "name": "CR-10",
                            "value": "cr_10"
                        }
                    ],
                    "version": "1.0",
                    "suppress_update_notification_version": false,
                    "is_custom": true
                },
                "has_been_saved_by_user": true,
                "slicer_type": "automatic",
                "gcode_generation_settings": {
                    "vase_mode": null,
                    "layer_height": null,
                    "extruders": []
                },
                "slicers": {
                    "cura": {
                        "slicer_type": "cura",
                        "version": "4.6.1",
                        "extruders": [
                            {
                                "version": "4.6.1",
                                "speed_z_hop": 5.0,
                                "max_feedrate_z_override": null,
                                "retraction_amount": 5.0,
                                "retraction_hop": 0.2,
                                "retraction_hop_enabled": false,
                                "retraction_enable": true,
[... 1443 lines cut ...]
                        "log_level": 5
                    },
                    {
                        "name": "octolapse.gcode_parser",
                        "log_level": 5
                    },
                    {
                        "name": "octolapse.script",
                        "log_level": 5
                    },
                    {
                        "name": "octolapse.migration",
                        "log_level": 5
                    }
                ]
            }
        }
    },
    "global_options": null,
    "upgrade_info": {
        "was_upgraded": false,
        "previous_version": null
    }
}
```

FormerLurker, 2020-05-13:
Well that looks right.  Can you post your gcode too?  This time use gist.github.com or pastebin so I can download easily, and so as not to increase the thread length so much.

Thanks!

kbuley, 2020-05-13:
https://gist.github.com/kbuley/d46a32e417972b57e5baa6ae88311e1a

FormerLurker, 2020-05-13:
Thanks!  Will run this through my debugging using your settings.  Will get to the bottom of this.

kbuley, 2020-05-13:
Additionally... this started yesterday when I changed from absolute to relative extruder settings (I _thought_ that was all I changed anyway). I upgraded to rc2 just to see if it made any difference, but alas no... and after looking and the gcode and log I wasn't sure where to go next.

FormerLurker, 2020-05-13:
That should not have had any impact.  Let's see what I come up with.  I have a few deployments to finish for work tonight, but I may be able to squeeze this in between reboots.  If not I will tackle this tomorrow.

kbuley, 2020-05-13:
No hurry, I've just disabled the plugin for now... I honestly didn't even expect a first reply for a few days.

FormerLurker, 2020-05-13:
So it looks like the spaces after the ; have been removed.  I could probably adapt the script to handle this, but if you use this template within your start gcdoe it should work fine:

```
; Script based on an original created by tjjfvi (https://github.com/tjjfvi)
; An up-to-date version of the tjjfvi's original script can be found
; here:  https://csi.t6.fyi/
; Note - This script will not work in Cura V4.2 and above!
; max_feedrate_z_override = {max_feedrate_z_override}
; retraction_amount = {retraction_amount}
; retraction_hop = {retraction_hop}
; retraction_hop_enabled = {retraction_hop_enabled}
; retraction_prime_speed = {retraction_prime_speed}
; retraction_retract_speed = {retraction_retract_speed}
; retraction_speed = {retraction_speed}
; speed_travel = {speed_travel}
; retraction_enable = {retraction_enable}
; layer_height = {layer_height}
; smooth_spiralized_contours = {smooth_spiralized_contours}
```

kbuley, 2020-05-14:
Sorry, had to wait for the current print to finish up. The spaces appear to be the magic sauce, I sliced a new model, re-enabled the plugin and everything appears to be right with the world again.

FormerLurker, 2020-05-14:
Excellent!  I will see if I can make it space agnostic.  Thanks for posting!

FormerLurker, 2020-05-14:
Going to close this and add a new issue for enhancing the cura settings detection to ignore spaces.

CLOSED!!!!

closed 2020-05-14 by FormerLurker