FormerLurker/Octolapse #522: [Request] Add 60 second interval (or custom interval) for timed shots

rickyh777, 2020-05-14:
I'm trying to use a gopro with octolapse but there's no good way to have octolapse and a gopro communicate. I had the idea that if I set the snapshot time to 1-2 seconds and time it right I can do a time lapse on the gopro set to 60 seconds and start it at the same time octolapse expects to capture the first picture. Assuming the clocks are somewhat similar I figured over a couple hour print the images should line up pretty good. I was tinkering with octolapse however and noticed the shortest time I can set for time interval captures is 2 minutes 30 seconds. Unfortunately my gopro doesn't let you exceed 60 seconds for time lapse. Is there any way to edit the time interval for octolapse be it from GUI or inside the code? Is a custom time interval something that can be added in a future release? Thank you so much for all your work!

FormerLurker, 2020-05-14:
Sorry, but that's not going to work unfortunately :(  You might get one or two shots to work, but there is drift in the timing for many reasons.  However, you'll be happy to know that you can set the timer trigger to capture as often as every 1 second.  The reason this won't work for the gopro is that the 1s interval is actually 1s of processing time and not print time.  Also, the time it takes to capture a snapshot is not included.  You printer buffers many commands, so that when Octolapse triggers it has to wait for these commands to execute, and it has no idea how long that will take.  The bigger your buffer the more time it will have to wait.  Even if Octolapse took this into account, it would only be guessing at how long it would take to clear the current buffer.  In short, it would be incredibly difficult to make it work in your scenario.

However, some people have gotten gopros to work using a few different methods.  One of my favorite is illustrated in this video:  https://www.youtube.com/watch?v=Bg5qwvyKCuw

That user rigged a mechanical switch that triggered voice playback that in turn activated the gopro via voice control.  Ingenious.

Next, and this is more complicated but also more powerful since it requires no switches and can be used with all of the stabilization profiles (the previous triggered only when the bed was in a certain position).  See [this issue](https://github.com/FormerLurker/Octolapse/issues/309) for an extensive report on how this was accomplished by one user.  It will definitely take a while to get that working, but we know it CAN be made to work.  I do not own a gopro, but I did try to borrow an older one to try out those scripts.  I wasn't able to get it working myself, because the model I borrowed was not compatible with the API used in that thread.  To save you time, [here](https://github.com/yet-another-average-joe/OctoLapse-GoPro) is a link to a repository created by the user @another-average-joe containing the necessary scripts and some videos.

I hope that helps!

rickyh777, 2020-05-14:
Wow thanks so much for the great reply! I love the voice trigger but I'm using a gopro hero 3 so it doesn't have voice trigger compatibility unfortunately. Another idea I was looking into was using a HDMI to CSI-2 bridge but those are rather expensive unfortunately. Thanks again for your help!

closed 2020-05-14 by rickyh777