FormerLurker/Octolapse #523: Octolapse: Pre Render Script fails in DSLR with Deferred Image Download mode

3dprint3d, 2020-05-18:
#### Version of Octolapse

Octolapse Version: Octolapse v0.4.0rc2

#### Version of OctoPrint

OctoPrint Version: OctoPrint 1.4.0

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES 

Following the steps in which logs to check, here:

[https://community.octoprint.org/t/octolapse-how-to-turn-off-saving-and-rendering-to-the-pi/4341/36
](url)

#### What were you doing when the problem occurred?

I'm trying to take make a timelapse with the scripts described here
https://github.com/FormerLurker/Octolapse/wiki/DSLR-with-Deferred-Image-Download

With the 0.4.0 upgrade im presenting the error when the print has ended, excatly when the _before render scripts_ starts

The Pi downloads the images from the SD but when it tries to rename them im getting this error 

_**Octolapse Rendering Failed**
There were problems running the before/after rendering script: The before script failed with the following error: before_render_script_error: A script occurred while executing executing the before-render script. Check plugin_octolapse.log for details.
**Octolapse Rendering Failed**
Rendering failed for camera 'New DSRL SD'. insufficient-images: No snapshots were available for the 'New DSRL SD' camera profile._

The rendering in the Pi (the images in the camera stay there, thats good)

#### What should have happened?

It should rename the photos after the download and render them

#### What happened instead?

The render stops

### Camera

Camera: CANON
Model: EOS REBEL T6

#### Operating System running OctoPrint and Octolapse

OS Name: Octopi 
Os Version: 0.17.0

#### Printer model & used firmware incl. version

Printer Model: Anycubic Chiron
Printer Firmware Version: Chiron Firmware_V1.3.0
[https://drive.google.com/open?id=146wb7yevvqFrqO481jTHrYxEr67inj4s
](url)

#### Browser and version of browser, operating system running browser

Browser: Chrome
Browser OS: Versión 81.0.4044.138 (Build oficial) (64 bits)

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: [https://drive.google.com/open?id=1zqorunAOj0e4XmnlHEQ1iBUGtiF-mWxA](url)

#### Link to settings.json

Link to settings.json with all passwords removed: 
[https://drive.google.com/open?id=10tDBwr8iGkD7BguWQgmih1OCeD2fLsBj](url)

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  
[<script src="https://gist.github.com/3dprint3d/0b27c7a9ce8cdd6b2055605d787d77f5.js"></script>](url)

Link to octoprint.log: [ https://drive.google.com/open?id=1-dPMUlrACt-LJFG4Wh9QkR0iLvV73M5H](url)

#### Screenshots and/or videos of the problem:

Screenshot/Video Links:
[https://drive.google.com/open?id=1TPQR2ZG_ngeIWuBktCAbhvvB1pxQST_S](url)

FormerLurker, 2020-05-19:
Nicely completed issue 👍.  I will take a look as soon as I can.  I've been thinking about improving the default copy and rename script, so this will be a good opportunity.  I will add proper error outputs too so the test button in the camera profile will work.

FormerLurker, 2020-05-30:
Update: i have been spending all of my time the last couple weeks getting things ready for the release of v0.4.  I will definitely get to your issue shortly after that is finished.  I am very sorry for the delay.

FormerLurker, 2020-06-09:
I think I fixed this.  Try updating to V0.4.0 and see if that helps.  If not, let me know.  There is also new logging profile you can import (click 'logging' in the tab and then 'add profile', then import the camera logging script) that will assist in creating a log file.  This will be extremely useful for figuring out what is going on.