FormerLurker/Octolapse #524: Error when attempting to enable Custom Preferences with Logitech C920 Webcam

WeakenedSoul, 2020-05-23:
#### Version of Octolapse
 
Octolapse Version: 0.3.4

#### Version of OctoPrint

OctoPrint Version: 1.4.0

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred

1. Setting up Octolapse on fresh octoprint setup
2. changed octopi.txt settings as instructed to enable C920 webcam

camera_usb_options"-r 1920x1080 -f 15"
camera_http_webroot="./www"
camera_http_options=""

3. Configure Octolapse Camera settings, Check "Enable and Apply Preferences Before Print"
4. Recieve Error - "Unable To Enable Custom Preferences
unknown-server-type: An unknown webcam server type 'Werkzeug/0.16.1 Python/2.7.16' is being used for the 'C920' camera profile. Currently only MJPEGStreamer is supported. Unable to apply custom image preferences."

#### What should have happened?
Custom Preferences should have been enabled

#### What happened instead?
Received Error - 
"Unable To Enable Custom Preferences unknown-server-type: An unknown webcam server type 'Werkzeug/0.16.1 Python/2.7.16' is being used for the 'C920' camera profile. Currently only MJPEGStreamer is supported. Unable to apply custom image preferences."

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: 1.4.0

#### Printer model & used firmware incl. version

Printer Model: Ender 3 Pro
Printer Firmware Version:  Creality 3D 1.1.6.2

#### Browser and version of browser, operating system running browser

Browser: Chrome
Browser OS: Version 83.0.4103.61 (Official Build) (64-bit)

FormerLurker, 2020-05-23:
Are you running octoprint anywhere or the spaghetti detective by chance?

WeakenedSoul, 2020-05-23:
The spaghetti detective is installed, yes. I don't find it as useful as I thought so thought about removing it. Does this cause conflicts?

WeakenedSoul, 2020-05-23:
Removing The Spaghetti Detective fixed the issue. Thanks for your prompt assistance! Now to work on the terrible stringing that I get when using Octolapse lol

FormerLurker, 2020-05-23:
No problem!

Regarding your stringing issue, are you running Marlin 2.0 or smoothieware on your printer?  If so, you may want to upgrade to the latest pre-release.  There is a bug involving the G90/G1 influences extruder setting in 0.3.4 that can cause intractable issues if G90/G91 influences extruder is true.  My firmware doesn't operate like that, so it wasn't tested as well as it could have been.

WeakenedSoul, 2020-05-24:
I don't believe so, its just a bone stock Ender 3 Pro running OctoPrint, not sure what Marlin Version runs on these machines. Fairly new to the 3D printing. I had an Anet A8 I played around with awhile back and just got this Ender 3 Pro last week.

closed 2020-06-02 by WeakenedSoul