FormerLurker/Octolapse #525: Python3: has_key in gcode_parser

OllisGit, 2020-05-23:
Hi @FormerLurker,

I used your ```gcode_parser.py``` (V0.3.4 also in master) in my [DryRun-Plugin](https://github.com/OllisGit/OctoPrint-DryRun/).

You use in line 124 the method ```has_key``` but this is not working in py3.
I implemented my own has_key-Methode. The result looks like this:

```
...
            if parameter_cmd is not None:
                parameter_value, parameters_string = parameter_cmd(parameters_string)
                parameters[parameter] = parameter_value
                additional_parameters = self.parse_parameters(parameters_string)
                self._myParameters = parameters
                if any(filter( self.hasKeyReplacement, additional_parameters.keys())):
                    raise ValueError("Either a parameter value was repeated or an unexpected character was found, "
                                     "cannot parse gcode.")
...
    def hasKeyReplacement(self, keyName):
        return keyName in self._myParameters
```
Not pretty, but it works ;-)

FormerLurker, 2020-05-23:
Hey, thanks for letting me know!  I use a totally different parser in the new version, so this should hopefully be a non-issue if I can ever get it out the door :)

OllisGit, 2020-05-25:
Okay, good to know! I release my version.

closed 2020-05-25 by OllisGit