FormerLurker/Octolapse #527: Cannot enter url in octolapse camera configuration

WeaponsHot, 2020-05-24:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: Octolapse 0.3.4

#### Version of OctoPrint

OctoPrint Version: 0.4.0

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: NO

#### What were you doing when the problem occurred

1. clicked on gear for octolapse camera configuration
2. tried to enter url for webcam
3. tried both localhost and ip address
4. Got error message The value is not a url. You may use {camera_address} to refer to the web camera address.
5. tried testing and succeeded
6. cannot save because The value is not a url. You may use {camera_address} to refer to the web camera address is preventing the save button from showing.

#### What should have happened?
I'm on the PC version and it should have allowed me to save camera configuration

#### What happened instead?
Does not allow me to save

#### Operating System running OctoPrint and Octolapse

OS Name: Windows 10 
Os Version: home

#### Printer model & used firmware incl. version

Printer Model:using virtual printer for testing
Printer Firmware Version: PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

WeaponsHot, 2020-05-24:
I figured it out. need to make sure you add www to what ever. ie http://www.localhost:8888

closed 2020-05-24 by WeaponsHot

FormerLurker, 2020-05-24:
I just now got to this, but am surprised the url parsing failed.  I will take a look at this.  I am working on releasing v0.4 now, so any fixes will apply to that version, fyi.

Thanks for posting, and I hope you are enjoying Octolapse!

WeaponsHot, 2020-05-24:
No, thank you. great app, just got it working on the PC version. had it working on the raspberry pi for about a week now, following some awesome vids on you tube. Trying out the best settings for the lapse. PC version was much more complicated.

WeaponsHot, 2020-05-24:
It would be nice if simplify 3d added this to their machine control panel as an option.