FormerLurker/Octolapse #528: "Full PC" error on Reflex Canon

reddai75, 2020-05-25:
Hello!
I would like to report a problem that is not directly related with Octolapse but it may be that Octolapse can help to circumvent the issue...
I'm seeing this problem on my Canon 7D (Mark 1): the camera reports "FULL PC" error when octolapse calls for consecutive gphoto2-calls in a short space of time. 
The message on the 7D-LCD-display is "Full PC". Once it's in that state, nothing works, and the error is usually only cleared by power cycling the camera or disconnecting it from my Raspberry. I couldn't find much about this issue on internet, only few people reporting the same, without a solution.
Calling gphoto in, say, lessthan 3 second intervals, is very likely to cause the problem; with
intervals over 10s I haven't seen any error... 
So basically when there are small layers the issue is highly likely to happen.
Do you have any advice or a way to circumvent the issue ?

Marco

FormerLurker, 2020-05-30:
This is a strange issue, and unfortunately I do not have a Cannon to test with.  Would you consider maybe opening an issue in the gphoto2 github repository?  The creator is quite responsive, and I would bet if you provide a debug log he would be willing to take a look.  In any case, let me know what you find!

reddai75, 2020-05-31:
Will do! I'll keep this thread posted in case I find a solution

FormerLurker, 2020-05-31:
Excellent, thank you!  Be sure to link back to this issue in case the repository owner has questions.  FYI, I've been working to debug some issues with the new enhanced camera scripting functionality in Octolapse V0.4 today while I fix up the guides.  If I find anything is broken I'll let you know.