FormerLurker/Octolapse #530: Feedback for v0.4.0rc3

FormerLurker, 2020-06-01:
I was SO CLOSE to getting V0.4 out the door, but I found and fixed a couple of high impact bugs, and I'd like to get some verification if possible before finally pulling the trigger.  I'd LOVE to hear from anyone who can test the following two things:

1. **Rendering after Cancelling or Disconnecting** - Octolapse should render any captured frames after the print was cancelled or disconnected.  This was the most concerning bug I fixed.  I recommend using test mode to confirm this fix.
2. [Deferred Download with a  DSLR](https://github.com/FormerLurker/Octolapse/wiki/V0.4---Configuring-a-DSLR-With-Deferred-Download) - Rendering was straight up broken for this, and several changes were necessary to get it going again.  This one is concerning because the fix required some substantial alterations to the rendering job.  If you were using this script in the past, please try it out.  Using test mode is fine.

There was also a significant error in the automatic profile updates, but I don't think this can be tested easily by users.  Also, it only affected the *Alternative Snapshot Command*, which isn't as important in V0.4.

Finally, I also rewrote the [Octolapse Wiki](https://github.com/FormerLurker/Octolapse/wiki).  I'm still fixing broken links due to some editor issues, so it may change rapidly.  I would appreciate any feedback about the wiki as well.

Please see the release notes [here](https://github.com/FormerLurker/Octolapse/releases/tag/v0.4.0rc3). This includes a list of bug fixes, and issues known at the time of release.

All feedback is welcome here!  Let me know what you like, what you hate, what works, what is broken, etc. 

You can install this version of Octolapse using this URL in the OctoPrint plugin manager:
```
https://github.com/FormerLurker/Octolapse/archive/v0.4.0rc3.zip
```
Please [read this guide completely](https://github.com/FormerLurker/Octolapse/wiki/V0.4-Getting-Started-with-Octolapse) before installation.  

Thanks!

gege2b, 2020-06-02:
Hi @FormerLurker 
Unfortunately (for me ^_^), I was forced to test the "render on cancel" bug (my part detached mid print) and it seems to work, the timelapse is rendering right now

every other things seems to work as attended

NickRno77, 2020-06-03:
Seems pretty stable so far, 
1. the camera setup detected my Logitech C920 which is great but I think it has defaulted to 1080p, I edited the standard web cam to lower res to 720P which gives me a wider field of view. Is there a way of setting the resolution.
2. You’ve cut down the amount of stabilisation  modes, I used center right a lot. I guess you expect people to create there own.

Thanks

NickRno77, 2020-06-03:
> Seems pretty stable so far,
>
> 1. the camera setup detected my Logitech C920 which is great but I think it has defaulted to 1080p, I edited the standard web cam to lower res to 720P which gives me a [...]

I just checked the video, it is 720p , I must be seeing  things! Forget about that one.

FormerLurker, 2020-06-03:
@NickRno77, yours is the second report about some missing stabilizations/triggers, so I believe it has something to do with the upgrade process.  You can easily add any missing profiles by clicking on the gear icon within the tab, then selecting the type of profile you want to add (stabilization in this case).  Next, click 'Add New Profile' and select the profile you want to import at the top of the settings.

What version of Octolapse did you upgrade from?  Was it v0.3.4?  If so, perhaps you can download the following file from your pi and post it to gist.github.com so I can run the upgrade through my debugger: /home/pi/.octoprint/data/octolapse/settings_backup_0.3.4.json

That would help me prevent this from happening when 0.4 is released.

Thanks for your feedback!

NickRno77, 2020-06-03:
> @NickRno77, yours is the second report about some missing stabilizations/triggers, so I believe it has something to do with the upgrade process. You can easily add any missing profiles by clicking [...]

I was on v0.3.4 but couldn’t figure out something in v0.4rc3 so I uninstalled old version and started from scratch. I figured out how to add in the end so sorry for waisting your time. 

Many Thanks

FormerLurker, 2020-06-03:
@NickRno77,

> so sorry for waisting your time

Don't worry, you did not.  You provided important info, and I appreciate it!

gege2b, 2020-06-03:
Hi
A had to pause a print because something went bad (again) and octolapse has complained about line number after I clicked on "restart" button in octoprint :

[image]

This only happened today, twice (with different Gcodes)

Here is a snippet for the log :

```
2020-06-03 21:46:26,244 - octolapse.timelapse - DEBUG - Sent: G1 X123.676 Y111.603 E8.49046
2020-06-03 21:46:26,261 - octolapse.timelapse - DEBUG - Sent: G1 X115.749 Y119.524 E8.67022
2020-06-03 21:46:26,277 - octolapse.timelapse - DEBUG - Sent: G1 X115.588 Y119.594 E8.67303
2020-06-03 21:46:26,292 - octolapse.timelapse - DEBUG - Sent: G1 X115.424 Y119.568 E8.67569
2020-06-03 21:46:26,312 - octolapse.timelapse - DEBUG - Sent: G1 X115.339 Y119.513 E8.67731
2020-06-03 21:46:26,340 - octolapse.timelapse - DEBUG - Sent: G1 X115.046 Y119.218 E8.68399
2020-06-03 21:46:26,357 - octolapse.timelapse - DEBUG - Sent: G1 X114.962 Y119.103 E8.68627
2020-06-03 21:46:30,622 - octolapse.timelapse - DEBUG - Sent: M105
2020-06-03 21:46:30,646 - octolapse.timelapse - DEBUG - Sent: M400
2020-06-03 21:46:31,134 - octolapse.timelapse - DEBUG - Sent: M114
2020-06-03 21:46:31,246 - octolapse.timelapse - DEBUG - Sent: M85 S240
2020-06-03 21:46:31,383 - octolapse.timelapse - DEBUG - Sent: M105
2020-06-03 21:46:31,392 - octolapse.timelapse - INFO - Print Paused.
2020-06-03 21:46:34,238 - octolapse.timelapse - DEBUG - Sent: M110 N0 <=== Here I clicked Restart
2020-06-03 21:46:34,328 - octolapse.timelapse - ERROR - File line number 944 was expected, but 1 was received!
2020-06-03 21:46:34,603 - octolapse.timelapse - DEBUG - Sent: M107  <== print restarted
2020-06-03 21:46:34,624 - octolapse.timelapse - DEBUG - Sent: G92 E0
2020-06-03 21:46:34,712 - octolapse.timelapse - DEBUG - Sent: M92 X80.402
2020-06-03 21:46:34,729 - octolapse.timelapse - DEBUG - Sent: M92 Y78.798
2020-06-03 21:46:34,744 - octolapse.timelapse - DEBUG - Sent: M92  Z2622.950
2020-06-03 21:46:34,760 - octolapse.timelapse - DEBUG - Sent: M92 E599
2020-06-03 21:46:34,776 - octolapse.timelapse - DEBUG - Sent: G21
2020-06-03 21:46:34,791 - octolapse.timelapse - DEBUG - Sent: G90
2020-06-03 21:46:34,807 - octolapse.timelapse - DEBUG - Sent: M82
```

Just to be completely honest, I'm not sure this is related to RC3... I did pause/restart a lot of print during lockdown, but I had octolapse disabled  (taking hundreds of timelapse of faceshields wasn't very necessarry then :D )

So, feel free to tell if you prefer a proper ticket for this

FormerLurker, 2020-06-03:
@gege2b, yes, this deserves its own ticket.  I just realized I have NEVER used the restart button ever, so that is basically broken.  Octolapse would need to delete any images already acquired and start over at line 1 to get this working, which is no small task.  I think the fix will have to wait until v0.4.1, and I'll be sure to add this to the known issues list when V0.4.0 drops.

Great catch, and thanks for the report!  I'm pretty surprised that nobody else (including myself) has noticed this before, so props to you :)

gege2b, 2020-06-03:
no problem! that's why free software are for :)

FormerLurker, 2020-06-08:
I just released v0.4.0,!  Thanks to everyone who posted!

Closed!

FormerLurker, 2020-06-08:
Also, @NickRno77, i found and fixed the missing profiles.  It was a version number sorting issue that prevented aome migrations from running.  Unfortunately it will only work for new upgrades.

closed 2020-06-08 by FormerLurker