FormerLurker/Octolapse #531: [Request] Create a New profile for Makerbot Replicator 2 or Replicatory 2X printer

chemixt, 2020-06-02:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE
 I am looking for help to make Octolapse work on a Makerbot Replicator 2x.

I normally use the Makerbot desktop slicer and export to .x3g file on SD card and sneaker net it into the printer. Now I am using the GPX pluggin, and the same Makerbot slicer but exporting as Gcode and upload to Octoprint via wifi. 

I would love to have a profile to use with the Makerbot Printer using GPX on Octoprint for Ocotprint Octolapse.   if the profile could inlcude using the Makerbot Desktop slicer output to Gcode, that would be ideal. 

My print volume  is 152mm wide (X)  by  246mm deep (Y) by 155mm high (z)

I am willing to help test any profiles and report results. 

Octoprint version 1.4.0
Octopi  Version 0.17.0, running on Raspberry Pi 3 Model B Plus Rev 1.3
GPX Pluggin  2.6.5

See additional information on my situation at this issue link https://community.octoprint.org/t/octolapse-replicator-2-settings/15472 

#### Version of Octolapse
  Octolapse 0.3.4
Octolapse Version: OCTOLAPSE_VERSION_GOES_HERE

#### Version of OctoPrint
1.4.0
OctoPrint Version: OCTOPRINT_VERSION_GOES_HERE

chemixt, 2020-06-02:
Attached is the json file from my slicer profile for ABS low resolution printing.
miracle.zip (1.9 KB)

chemixt, 2020-06-02:
sorry, I accdentally closed this issue, I do not know how to undo that.

FormerLurker, 2020-06-02:
Excellent, thank you!  

In order to test the profile you will have to update to the latest release candidate of Octolapse (currently v0.4.0rc3).  Follow [this guide](https://github.com/FormerLurker/Octolapse/wiki/V0.4---Getting-Started) for instructions on how to update.  I would recommend that you restore the default settings after you install (this is suggested in the guide).

Once I have created a test profile, I will send you the json, and you can [import the profile](https://github.com/FormerLurker/Octolapse/wiki/V0.4---Profiles#importing-settings-and-individual-profiles) into Octolapse.  Once we've verified that it's working, I will add it to the plugin repository so that it can be imported directly from the printer profile settings.

chemixt, 2020-06-03:
Awesome.  I will update so I am ready when you are.

FormerLurker, 2020-06-09:
Can you try something for me?  In the OctoPrint terminal tab, enter the following commands:

```
G28 X Y; Home X and Y
G1 X0 Y0
```

And tell me where the extruder is.  Is it in the center of the bed?

FormerLurker, 2020-06-09:
Also, send me a gcode file please.  Thx!

chemixt, 2020-06-10:
I received notification that Octolapes had an update for 0.4.0 so I updated
it.

The first command "G28 X Y; Home X and Y" takes the print head to the back
right corner of the travel (red light zone for both x and y, and locks it
there.)
if i issue the second command as a separate send line, the second command G1
X0 Y0 moves it about 55 mm down (-X) on the X axis and about 55 mm to the
left (-Y) axis and locks it there.

Attached is a gcode file for a ABS print.

FormerLurker, 2020-06-10:
Ok, so there are definitely some oddities here.  What we need to fine is the min and max x and y coordinates.  Maybe you can play around and see if you can discover them?

chemixt, 2020-06-10:
OK.  here is a another Gcode that prints a 10 X 10 X 10 mm box in the
center of the build plate.

chemixt, 2020-06-10:
oops forgot to attached the file

chemixt, 2020-06-10:
sorry, (bad day) it prints a 10mm  X10 mm X 5 mm high square.

chemixt, 2020-06-10:
ok, (remember I am a newbie at this)  my build bed size is X= 246mm,
Y=152mm, z=155mm

G0 X-112 takes the extruder all the way to the Right as far as it will go
without hitting the stops
G0 X165 take the extruder all the way to the right as far as it will go,
without hitting the stops
G0 Y0 takes to the front of the bed
G0 Y149 takes to the back of the bed

So those are the furthest travels that it can make, but that takes the head
beyond the edge of the print bed.

chemixt, 2020-06-10:
PS. I should have mentioned that my printer configuration for the
makerbot was wrong when I set up ocotprint.  I had my x and y axis
switched.  But I corrected this before I did the experiments with G0
codes.

FormerLurker, 2020-06-10:
Excellent, that is very helpful, thanks!

MobiuSRIT, 2021-05-28:
I've subscribed to this Request as I have a Makerbot Replicator 2 that I'm looking to set up correctly as well.

chemixt, 2021-05-28:
I love my octoprint with the MakerBot 2x.  Do you have yours running now?

MobiuSRIT, 2021-05-28:
With Octoprint, yes.  With Octolapse, no.
G28 X Y; Home X and Y <-- This pushes the head to the back right of the printer (where it's normal home is)
G1 X0 Y0 <-- this attempts to break my printer by jamming the head all the way forward.  I have to shut it off to prevent damage.
I'm using Width X at 285mm, Depth Y at 153mm, Height Z at 155mm, Origin Center in Octoprint.  This works fine when I use MakerBot Desktop to export the gcode.
Octolapse freaks out when attempting to print the samething that Octoprint+timelapse printed and starts jamming up against limits and the motors are clunking trying to drive the print head into the nether.

MobiuSRIT, 2021-05-28:
Here's the intro gcode on a working print (exported from MakerBot Desktop)...

M136 (enable build)
M73 P0
G162 X Y F2000(home XY axes maximum)
G161 Z F900(home Z axis minimum)
G92 X0 Y0 Z-5 A0 B0 (set Z to -5)
G1 Z0.0 F900(move Z to '0')
G161 Z F100(home Z axis minimum)
M132 X Y Z A B (Recall stored home offsets for XYZAB axis)
G92 X152 Y75 Z0 A0 B0
G1 X-141 Y-74 Z40 F3300.0 (move to waiting position)
G130 X20 Y20 A20 B20 (Lower stepper Vrefs while heating)
M135 T0
M104 S210 T0
M133 T0
G130 X127 Y127 A127 B127 (Set Stepper motor Vref to defaults)

FormerLurker, 2021-06-01:
Going to take another look at this.  The makerbot is.. difficult to deal with, but I'll give it another shot.  Been trying to get it working for a good long while now.  I wish I had one here to test on :)

chemixt, 2021-06-01:
I have the Makerbot 2x.  If there is any way I can help (with my limited
knowledge of Gcode,etc) let me know.

rathkopf, 2021-06-06:
I too am happy to help/test on a 2x.

fridayshoes, 2021-10-04:
Has anyone made any progress on this? Got a replicator 2X here

solo2424, 2022-07-18:
Just figured I would check in on this again. Just got a Replicator 2x and I am running into issues.