FormerLurker/Octolapse #539: v0.4.0 Octolapses Freezes in Loading Octolapse Screen once the print starts

dgcntrk, 2020-06-09:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: ___0.4.0

#### Version of OctoPrint

OctoPrint Version: ___1.4.0

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: ___yes

#### What were you doing when the problem occurred

1. ___Configure the octolapse
2. ___start the printing
3. ___REPLACE_THIS__STEP_...

#### What should have happened?
___Run like octolapse

#### What happened instead?
___stuck in loading screen

#### Operating System running OctoPrint and Octolapse

OS Name: ___mac os 
Os Version: ___Catalina

#### Printer model & used firmware incl. version

Printer Model: ___anycubic mega s
Printer Firmware Version: ___?

#### Browser and version of browser, operating system running browser

Browser: ___safari
Browser OS: ___?

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: ___https://gist.github.com/dgcntrk/2a241fdc191a9c41947074af8ea1b719

#### Link to settings.json

Link to settings.json with all passwords removed: ___https://gist.github.com/dgcntrk/bbe1d7ad9a2d33bf95cbacc17ddda51d

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://gist.github.com/dgcntrk/722c7fb64af56f7c33d0443eed90293f

#### Link to octoprint.log

Link to octoprint.log:  ___https://gist.github.com/dgcntrk/ae7b791aa4b59e3c97864880343fb027

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___https://snipboard.io/cvtaTD.jpg

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2020-06-09:
Please add the console log output for this issue.  Just hit F12, select the console, refresh the OctoPrint page and copy in any errors you see.  It's probably a client side binding issue of some type.

Also, it looks like there are URLs entered into script paths in your camera profile.  Not sure how that happened.  If you upgraded from Octolapse V0.3.4, I'd like to see the settings backup.  You can find it in /home/pi/.octoprint/data/octolapse and it will have 3.4 somewhere in the name (settings_backup_0.3.4.json or something like that).

Also, see if the settings page comes up -> OctoPrint Settings (wrench/spanner) ->Octolapse.  If it does, and after you have given me the logs, you could try restoring the default settings.

dgcntrk, 2020-06-09:
Hi there, 
Thank you so much for reaching out! 
Here are the 2 errors I see on console. Btw I could not get the console on safari with f12 so this is from Chrome. 

getSnapshot:1 Failed to load resource: the server responded with a status of 404 (Not Found)
getSnapshot:1 Failed to load resource: the server responded with a status of 404 (Not Found)

Also because I have python 3 v0.3.4 wouldn't even install. 
The setting page comes up under octoprint and I can tweak the settings but once I start the print it just freezes at the loading screen.

I tried to reset it multiple times, I even installed the whole octoprint to see if that was causing any problems but still no luck.

FormerLurker, 2020-06-09:
Can you try restoring the default settings?  OctoPrint Settings->Octolapse->Restore Default.  It will take me some time to investigate this.

> once I start the print it just freezes at the loading screen.
Can you elaborate?  Are you saying Octolapse correctly displays until after you start a print?  I may not understand the issue 100%

dgcntrk, 2020-06-09:
Sure, 
and thanks a lot. Here is another screenshot after I reset to defaults. There is an error on top right. 

https://snipboard.io/al8OJz.jpg

dgcntrk, 2020-06-12:
Here is the latest log
https://gist.github.com/dgcntrk/ef8b32c69cc6accc1d0f4b1cc17ec0d9

FormerLurker, 2020-06-12:
Excellent 👍.  I have a few ideas.  Will try to have an update tomorrow.

dgcntrk, 2020-06-12:
That's great! Really appreciate all the effort man 👍

FormerLurker, 2020-06-15:
Sorry about the delay, but the weekend ended up being busier than normal.  I have received a similar report to yours (minus the lockup in the GUI), and I strongly suspect they are related.  I've added some of the logging that will be necessary to fix this, but still have some work to do.  I'll keep you updated, but it will probably take a few weeks to figure this out.  BUT..  the other user is running on CentOS, so it's possible for me to create a VM and actually replicate the bug locally, which is excellent.