FormerLurker/Octolapse #540: OctoLapse Uncontrollably extruding when returning from taking a snapshot

Commiccannon, 2020-06-12:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version:0.4.0

#### Version of OctoPrint

OctoPrint Version: 1.4.0

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: NO

#### What were you doing when the problem occurred

1. I had just updated octolaspe first print with new update 

#### What should have happened?
The print should have resumed normally like it always has

#### What happened instead?
as  soon as it came back from taking a snapshot it started extruding a heap i did not leave it running to see how much as it was stripping my filament 

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: 0.17.0

#### Printer model & used firmware incl. version

Printer Model: Ender3
Printer Firmware Version: Marlin 1.1.9

#### Browser and version of browser, operating system running browser

Browser: ___REPLACE_THIS__BROWSER_VERSION_GOES_HERE
Browser OS: ___REPLACE_THIS__BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: https://gist.github.com/Commiccannon/e75342c9efff5bc1cf9cf9c1c100edbe

#### Link to settings.json

Link to settings.json with all passwords removed: https://gist.github.com/Commiccannon/d967d525d8f9af2da142d478b213f067

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://gist.github.com/Commiccannon/43457786e8971fc20f171e27b703ef64

#### Link to octoprint.log

Link to octoprint.log: https://gist.github.com/Commiccannon/8c249180aff3e92643199c88bcdab9ed

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2020-06-12:
Looks like printer profile issues.  This setting should be false for your firmware:

[image]

I recommend importing the pre-configured profile like this:

[image]

Please note that if you ever upgrade to Marlin 2 you will need to switch the G90/G91 influences extruder setting to True.

You should also take a look at the [getting started](https://github.com/FormerLurker/Octolapse/wiki/V0.4---Getting-Started) guide if you haven't already done so.  There is also a [troubleshooting page for printer profiles](https://github.com/FormerLurker/Octolapse/wiki/V0.4---Troubleshooting-Printer-Profile-Issues#there-are-a-ton-of-huge-blobs-andor-other-artifacts-on-my-print) that explains the issue you are seeing and how to fix it.  If you have similar problems in the future, be sure to check out the troubleshooting pages.

I'm trying to figure out a way to auto-detect this setting, but so far I haven't been able to figure out how to do this.  Let me know if that helps, and thanks for posting!

Commiccannon, 2020-06-13:
Thanks for the quick reply i had not seen the profile obviously i was blind the system now seams to be working perfectly. Thanks!

closed 2020-06-13 by Commiccannon