FormerLurker/Octolapse #541: Octolapse - Position Error (octolapse version: Octolapse 0.3.4)

zivdvir, 2020-06-12:
Hello, I tried to print something in octoprint but I discovered some issues with the octolapse.
I want to to make a time lapse with the octolapse plugin, and I did all the things which related to it, I entered the data about printer about the slicer, etc. when I'm start the print I get this massage:
#### Octolapse - Position Error

Position - Coordinates (X:None,Y:None,Z:None,E:60.00000) are out of the printer area! Cannot resume position tracking until the axis is homed, or until absolute coordinates are received.

I searched on the web how to overcome this problem, and I have tried couple things, such as:
changing the printer profile in octolapse, adding g90 and M82 before the starting script, playing with the X/Y/Z Axis Mode with all the options and all the related things to it.
I really dont know why its not working. if you can help me it will be great.

octoprint version: Version 1.4.0
octolapse version: Octolapse 0.3.4
I'm using simplify3d

starting script:
G21 ; set units to Millimeters
G90 ; absolute positioning
M82 ; set extruder to absolute mode
G92 E0 ; zero the E-axis
G1 E60 F300 ; extrude 20mm of filament at 800mm/min
G28 X Y ; home X Y axes
G1 Y20 ; Y axel forword
G28 Z G ; home Z axel

layer changing: 
G4 P1

ending script:
M104 S0 ; turn off extruder
M140 S0 ; turn off bed
G91 ; relative movements
G1 F150 ; slow Z rate
G1 Z15 ; Z up 15mm
M84 ; disable motors

I will be happpy if you can help me with this. thanks!

FormerLurker, 2020-06-12:
First off, it's safe to ignore this error, but Octolapse 0.3.4 will complain if there is any movement  before all 3 axes are homed.  If you send me your entire gcode file and your settings.json I could figure out exactly which line is throwing the error.

Another option would be to upgrade to Octolapse v0.4.  I've pretty much removed that error completely in the vast majority of cases.  If you upgrade I recommend NOT using the G4 P1 triggering method and switching to the *Smart - Compatibility* trigger.  Check out the [getting started](https://github.com/FormerLurker/Octolapse/wiki/V0.4---Getting-Started) guide for a walk through of the new version and how to use it and check out the [release notes](https://github.com/FormerLurker/Octolapse/releases/tag/v0.4.0) to see what has changed.  V0.4 is superior in almost every way to 0.3.4, so I recommend you update.

zivdvir, 2020-06-12:
Thanks for the quick response, I will be more than happy if you can check
my gcode. I think I will upgrade the version later.
The gcode and the fff file attached, please tell me if I need to do
something. (like I said before, I'm using simplify3d)
Thank you, I appreciate it!

‫בתאריך יום ו׳, 12 ביוני 2020 ב-18:12 מאת ‪FormerLurker‬‏ <‪
notifications@github.com‬‏>:‬

FormerLurker, 2020-06-12:
Not a problem.  I will review the gcode tomorrow while working on the open issues.

zivdvir, 2020-06-12:
Thank you very much!

‫בתאריך יום ו׳, 12 ביוני 2020 ב-21:53 מאת ‪FormerLurker‬‏ <‪
notifications@github.com‬‏>:‬

zivdvir, 2020-06-13:
Hello, do you have new info about this thing? thanks

‫בתאריך יום ו׳, 12 ביוני 2020 ב-21:53 מאת ‪FormerLurker‬‏ <‪
notifications@github.com‬‏>:‬

zivdvir, 2020-06-15:
> Not a problem. I will review the gcode tomorrow while working on the open issues.

Hi, do you have new info about it? you said that you will review it. thank you

FormerLurker, 2020-06-15:
Sorry, things came up over the weekend.  I promise I will get to it ASAP.

FormerLurker, 2020-06-15:
I was going to review your gcode, but it doesn't seem to be attached.  Also, there is no settings.json.  If you tried to attach it to the email reply earlier, I don't believe that works.  Please upload your files to gist.github.com and leave a link here.

Thanks!

zivdvir, 2020-06-17:
Hi, I have upgrade the version of octolapse and it seems like everything
working alright now. thanks anyway!

>

FormerLurker, 2020-06-17:
No problem!  I hope you are liking the new version!

zivdvir, 2020-06-30:
The new version is awesome, it works excellent for me except one problem:
it doesn't take a snapshot before the first layer, and also it take 2
snapshot at the first layer. What can I do in order to fix it?

‫בתאריך יום ד׳, 17 ביוני 2020 ב-19:39 מאת ‪FormerLurker‬‏ <‪
notifications@github.com‬‏>:‬