FormerLurker/Octolapse #542: Octolapse causing bed to crash into frame

AaronLee115, 2020-06-12:
#### Version of Octolapse
 
Octolapse Version: 0.4.0

#### Version of OctoPrint

OctoPrint Version: 1.4.0

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred

1. Generate GCode with Verbose Gcode
2. Ran Gcode through Octoprint
3. Noticed that when going to take picture, the bed traveled too far forward and crashed into the frame causing the issue. This interrupts the print and misaligns the nozzle for continued print. (I've replicated this issue 4 times but only logged the last one)

#### What should have happened?
Bed should have moved forward without crashing into frame and resume print as normal

#### What happened instead?
Bed crashed into frame and ruined the print

#### Operating System running OctoPrint and Octolapse

OS Name: Windows
Os Version: 10

#### Printer model & used firmware incl. version

Printer Model: Prusa i3 MK3S
Printer Firmware Version: 3.9.0

#### Browser and version of browser, operating system running browser

Browser: Chrome v83.0.4103.6
Browser OS: Windows 10

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: https://gist.github.com/AaronLee115/5638124e89f3396c95b6c2c9e63e9405

#### Link to settings.json

Link to settings.json with all passwords removed: https://gist.github.com/AaronLee115/5638124e89f3396c95b6c2c9e63e9405

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log: https://gist.github.com/AaronLee115/5638124e89f3396c95b6c2c9e63e9405

#### Link to octoprint.log

Link to octoprint.log:  (not sure how to get)

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: https://drive.google.com/drive/folders/1DU6IMxjLiCB_Dzc2DoYO-zpPw3fedG-f?usp=sharing

*(all logs are in regards to the sanding stick, but the failed spool holder shows the issue more clearly [don't have logs for spool holder])

FormerLurker, 2020-06-12:
Thanks for reporting this.  I will be debugging more issues tomorrow, so hopefully I can figure it out then.

FormerLurker, 2020-06-13:
Apparently I don't have access to the video you provided via google drive.

Also, I noticed you are using a custom MK3 profile.  Go into your printer profile and select your make and model.  See if that makes a difference.  I noticed one setting in there that may be wrong (it's tied to your Octoprint settings, so I can't be sure).

Finally, can you turn your printer off and slide your bed all the way forward and backward.  See if it feels smooth the entire time, or if it seems to catch when you pull your bed all the way forward.  I had a similar problem when I upgraded to the Mk2.5s, and it was due to a misaligned belt.  It was catching, but only at the extremes of the bed, which caused some skipping.  Aligning the belt better fixed my issue.

FormerLurker, 2020-06-13:
OH, also, it looks like you posted your settings.json twice and didn't post the log file.  Please correct this at your convenience, thanks!

AaronLee115, 2020-06-13:
Hey,

Sorry google is a bit clunky when it comes to sharing these videos. See if
these links work:
https://drive.google.com/file/d/1IKSMywMd5JFdQ-4M7qCqv244DxR2mFzF/view?usp=sharing
https://drive.google.com/file/d/1BnlHxZsgEqSejHx3KaewJ2FXAdSLjBgc/view?usp=sharing

Oh it seems you're right about the printer profile. I remember when I was
first setting it up I was watching a youtube video and I don't remember
them making this step. I've been using octolapse for a month or so before
this started happening so I'm not sure why it came out of the blue. I'll
try it again with this updated profile.

For the printer bed, I had an issue with the bed sliding but I fixed it
before doing that last print with the debug mode on.

Lastly, the log should have the correct information now.

Thanks for responding so quickly to this. As I mentioned I'll keep running
with the profile with Prusa MK3S as the make and model to see if I still
come across the issue.

Best,
Aaron

FormerLurker, 2020-06-15:
Thanks for posting those.  The link worked, but I can't really see anything useful.   

> For the printer bed, I had an issue with the bed sliding but I fixed it before doing that last print with the debug mode on.

Just to be clear, you have not had any issues since switching the profile out and fixing a problem with the bed?  If so I suggest we monitor the situation and see if it occurs again, handling problems as they come up.  What do you think?

AaronLee115, 2020-06-15:
I wasn't able to run any prints over the weekend because I was busy and
didn't trust having it run without my supervision. However, I just ran on
this morning (Monday) with the new profile and the same issue occurred.
Unfortunately, I took it off debug mode so I don't have the logs for it,
but it seems switching out the profile didn't fix it for some reason. Let
me know if you want me to try again with debug mode on or if you want me to
send you any more information. Also, I ran a print where it was set to
center-right and didn't have any issues so it definitely seems like it
keeps crashing into the front frame. What I don't understand is how it goes
about 4-15 layers before causing this crash. Like it seems at some point to
tell the bed to go a bit too far.

FormerLurker, 2020-06-15:
@AaronLee115, I'm suspecting more and more this is a belt issue that is only encountered when your printer is moving at that extreme Y end (Y is around 210).  Try this to verify:

Edit your stabilization profile for the back center, click the 'customize' check box and change your settings to look like this:

[image]

This will cause Octolapse to stabilize at Y=189, which is about 20mm (.75") from the back of the bed.  This should be good enough to hide your extruder, but might be far enough away not to snag your belt.  If this fails, try setting the Y relative coordinate to 80.  Test mode should work fine for this, btw, and it would save you some filament (check out the test mode help for details).

Let me know if that works or if it fails and we'll take it from there.

thanks!

AaronLee115, 2020-06-15:
Hey,

So I adjusted the settings to have the relative coordinate of 90% and ran a
test print with debug mode on. It all worked perfectly. It seems like it
wouldn't obstruct the print as long as I move the extruder off to a side.
You mentioned that this might be a belt issue, do you think it's because
the belt has some slack on it? I may want to try to fix this in the future
should something like this pop up again.

Thanks again for all the help, I really appreciate it! Do you want to close
the issue on the octolapse website and should mention the fix?

FormerLurker, 2020-06-15:
Oh, on the website, just mention that it could be mechanical and close it out.  Leave this open for a bit so you can let me know for sure that that your belt is misaligned.

Regarding the slack, it's more likely it's due to a misalignment and not tension.  A tighter belt may even be more prone to causing skipping if it is misaligned.  It's actually quite difficult to get the alignment right (it took me several attempts once I figured out what was up).  Watch your pulley as you move your bed back and forth.  You may see the belt move to the left and right as you move the bed back and forth.  If it drifts too far (a bit is just fine), it may hit the side of the pulley and bind.  In my experience, if it's off by just enough to rub on the side of the pully when it's all the way back/forward, it may SOMETIMES (not always) cause skips.  keep in mind this can happen when you print too, if the filament is placed right at the back or front of the bed.  Most prints don't do this, so it's not usually a problem, but it can result in skipping/layer offsets.  Also, moving slower is less likely to cause skips, and printing speeds are usually much lower than travel speeds.  This makes Octolapse more likely to skip than just laying down filament.

In general, the bed should be as easy to move when it's all the way back as it is when it's all the way forward.  If there is more resistance at any point, it's something that would ideally be dealt with.  However, now that you know this you could simply avoid printing on the back 10-20mm of your bed and you should be fine.  Do keep an eye out for black soot that may develop under your bed.  It might be worn rubber from your belt rubbing against the pulley, which can damage the belt over the long term.

AaronLee115, 2020-06-15:
OK, I'll leave it open for a bit, maybe a week while I print some more
stuff to confirm to issue being resolved. I took a look at the belt and it
has some slack (per the building instructions) and I haven't noticed any
misalignment but I'll keep an eye on it and leave the coordinate at 90%
(might switch to 95%) just for safety sake.

AaronLee115, 2020-06-18:
Hello, I just wanted to update you on the status of my printing. After
having the test print run and successfully finish. I decided to print the
part with the updated settings, but the issue occurred again with the
coordinate set to 90%. I tested it with the back left, back center, and
back right. Each one failed for the same reason. When I deactivate
Octolapse the issue never occurs. I even set the X axis to have coordinates
to have the 10% buffer like the Y axis just in case and the issue
persisted. Lastly I intend to run a print with the trigger being set to
every layer (like it used to) instead of at the Smart Compatibility. I
should have the final results by the end of today or tomorrow. I also saved
the logs for two of the failed prints.

FormerLurker, 2020-06-18:
@AaronLee115, try stabilizing to the center of the bed and see if that help.  I honestly don't see how the classic layer trigger will work any better since they both use the same gcode generations class to generate the snapshot gcode.

AaronLee115, 2020-06-20:
Hey, so I tried it with the center of the bed with both teh smart and layer
snapshot and it detected a crash both times. So it seems that the printer
crashes even when the bed doesn't move. The difference this time is that
after the crash it just got misaligned on the X axis. After the crashes,
the Y axis was a bit stiff, but after manually moving it once, it works
normally. I did another print with octolapse off again to confirm and it
ran successfully. I have the logs for those if you want me to link them.

AaronLee115, 2020-06-26:
Hey, I haven't heard from you in a while. Did you see my last email?

Best,

FormerLurker, 2020-06-29:
Could this be a crash detection problem?  Can you try disabling that via your LCD screen and see what happens.

Also, thanks for the nudge.  Things have been really crazy the last month, and I apologize for my unusually slow responses.

AaronLee115, 2020-06-29:
Hey,

No worries about the response time. It's a pretty hectic time right now.
Actually I thought that maybe something fishy happened when I updated my
octolapse version to the latest (because that's when these issues first
came up), so I decided to completely uninstall it and reinstall it just for
my sanity sake. So far I've been running a print for an hour and a half
with octolapse on without an issue. I'm going to test it out more with a
few small prints. I'm hoping something just went wrong during the update.
I'll keep you updated on the status.

FormerLurker, 2020-06-29:
That is concerning.  Please do keep me posted, and thanks for sticking with this process.

AaronLee115, 2020-07-02:
Hey, so I ran a few prints with different settings:
Center - 1 hour print - success
Center - 3 hour print - success
Center Left - 3 hour print - success
Front Left - 8 hour print - failure (about 6 hours in) (same crash issue as
before)

Before uninstalling Octolapse, these failures and crash detections
occurred about 10-20 mins in or <10 layers. This time the crash happened
after 6 hours. I still plan on printing a few more things to experiment. I
know you recommended turning off crash detection on my printer, but I'm
still curious about why Octolapse itself is causing these crashes when it
doesn't hit anything.

I've downloaded the latest logs if you want them. So far my profile is the
default prusa one, but let me know if you want me to send over any other
info as well.

Best,

FormerLurker, 2020-07-06:
Not sure what could be causing the crash detection.  If you have the time/desire, I'd love to know if you have success/failure with crash detection off.  It would be good to know if a crash really occurred, or if something is off with the detection.  It is possible that M400s mess with crash detection.  If that's the case, it can probably be fixed.

Lots of ppl with MK3s are running Octolapse without this issue, so I am still somewhat confused.  I really wish I had your printer in front of me, but your reports and testing have been very good so far (thank you for that).

AaronLee115, 2020-07-06:
Yeah, it seems strange why it's happening on my printer only. I have a few
friends running octolapse on their MK3S's and they are running it without
issue. What I've done recently is completely wiped Octoprint from my pi and
reinstalled itwith all my addons again. I've been slowly printing longer
prints to test this out. So far I think I ran a 6 hour print without an
issue. Today I plan to run a 10-12 hour print.

AaronLee115, 2020-08-19:
Hey,

I realized that I never followed up on this issue. My bad. I feel kind of
dumb about this because it did end up being a bearing issue. I didn't
notice any sticking when physically moving along the axes, but after
regreasing them I haven't come across the issue again. Apologies for all
the trouble this may have caused you when this was one of the first things
you recommended I try.

FormerLurker, 2020-08-19:
Lol, no worries!  I am very glad you figured it out!