FormerLurker/Octolapse #543: PLUGIN_OCTOLAPSE_MOVIE_DONE not firing event

AJErazzor, 2020-06-12:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.4.0

#### Version of OctoPrint

OctoPrint Version: 1.4.0

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: NO

#### What were you doing when the problem occurred

1. ___REPLACE_THIS__STEP_ONE_GOES_HERE
2. ___REPLACE_THIS__STEP_TWO_GOES_HERE
3. ___REPLACE_THIS__STEP_...

#### What should have happened?
___REPLACE_THIS__PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
___REPLACE_THIS__PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: ___REPLACE_THIS__OS_NAME_GOES_HERE
Os Version: ___REPLACE_THIS__OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: ___REPLACE_THIS__PRINTER_MODEL_GOES_HERE
Printer Firmware Version: ___REPLACE_THIS__PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: ___REPLACE_THIS__BROWSER_VERSION_GOES_HERE
Browser OS: ___REPLACE_THIS__BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: ___REPLACE_THIS__GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: ___REPLACE_THIS__SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  ___REPLACE_THIS__LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

trying to get PLUGIN_OCTOLAPSE_MOVIE_DONE event to fire after an octolapse is complete.  Tried both upper and lower case - is it in this vonal 0.4.0 version?

FormerLurker, 2020-06-13:
Yes.  It is integrated with the dropbox  upload plugin, so I know it cant work.  I haven't used the YouTube upload plugin though.  Does it work with the stock timelapse plug-in if you use the octoprint movie done event?

AJErazzor, 2020-06-13:
Can’t work or can work?

Thanks
Andrew

FormerLurker, 2020-06-13:
I edited the post, sorry am on mobile.  Yes, the event does fire in 0.4, unless there is some undiscovered bug.  I tgink you can log octoprint events to verify, but I will add some logging to Octolapse for further verification.  Will let you know when that is done.

AJErazzor, 2020-06-13:
hi thanks for the tip i thoght everylogging option was on but can see the different types - it is working with plugin_octolapse_movie_done beautifully combines with the youtube-upload script

closed 2020-06-13 by AJErazzor