FormerLurker/Octolapse #545: Stabilization error on all tested stabilization settings on Prusa MK3S

BirkirFreyr, 2020-06-14:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: v0.4.0

#### Version of OctoPrint

OctoPrint Version: v1.4.0

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: yes

#### What were you doing when the problem occurred

1. Turn on octolapse
2. Turn on print
3. Failure to stabilize

#### What should have happened?
No error.

#### What happened instead?
This may be due to not running on OctoPi but off my CentOS7 "Server" - but I don't see why that would be. More likely that this is either a bug or User error, in either case I'd love to know which and fix it.
Stabilization error on all tested settings - this output is from having stabilization set to Disabled - debuglog for Centered included in octolapse.log section:
```
2020-06-14 15:15:10,929 - octolapse.timelapse - VERBOSE - Queuing: G1 X119.152 Y97.520 E0.01073
2020-06-14 15:15:10,929 - octolapse.gcode_parser - VERBOSE - Parsing gcode.
2020-06-14 15:15:10,929 - octolapse.timelapse - DEBUG - Setting job-on-hold lock.
2020-06-14 15:15:10,929 - octolapse.timelapse - VERBOSE - Sending: G1 X118.430 Y98.148 E0.01073
2020-06-14 15:15:10,930 - octolapse.timelapse - INFO - About to take a snapshot.  Triggering Command: G1 X119.152 Y97.520 E0.01073
2020-06-14 15:15:10,930 - octolapse.timelapse - DEBUG - Sent: G1 X118.430 Y98.148 E0.01073
2020-06-14 15:15:10,931 - octolapse.timelapse - ERROR - Failed to take a snapshot for the provided snapshot plan.
Traceback (most recent call last):
  File "/opt/octoprint/op_venv/lib/python3.6/site-packages/octoprint_octolapse/timelapse.py", line 409, in _take_timelapse_snapshot_precalculated
    self._trigger_profile.get_snapshot_plan_options()
  File "/opt/octoprint/op_venv/lib/python3.6/site-packages/octoprint_octolapse/stabilization_gcode.py", line 1144, in create_gcode_for_snapshot_plan
    self.return_to_original_feedrate()
  File "/opt/octoprint/op_venv/lib/python3.6/site-packages/octoprint_octolapse/stabilization_gcode.py", line 912, in return_to_original_feedrate
    self.get_gcode_feedrate(f_return))
  File "/opt/octoprint/op_venv/lib/python3.6/site-packages/octoprint_octolapse/stabilization_gcode.py", line 1238, in get_gcode_feedrate
    return "G1 F{0:.3f}".format(f)
TypeError: unsupported format string passed to NoneType.__format__
2020-06-14 15:15:10,933 - octolapse.timelapse - INFO - The snapshot has completed
2020-06-14 15:15:10,933 - octolapse.timelapse - INFO - Sending on_snapshot_complete payload.
2020-06-14 15:15:10,934 - octolapse.timelapse - VERBOSE - Queuing: G1 X119.934 Y96.968 E0.01074
2020-06-14 15:15:10,935 - octolapse.gcode_parser - VERBOSE - Parsing gcode.
2020-06-14 15:15:10,936 - octolapse.timelapse - DEBUG - Releasing job-on-hold lock.
2020-06-14 15:15:10,973 - octolapse.timelapse - VERBOSE - Received: ok
```

#### Operating System running OctoPrint and Octolapse

OS Name: CentOS
Os Version: 7

#### Printer model & used firmware incl. version

Printer Model: Prusa i3 MK3S
Printer Firmware Version: 3.9.0

#### Browser and version of browser, operating system running browser

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: https://pastebin.com/ewMCUbf4

#### Link to settings.json

Link to settings.json with all passwords removed: https://pastebin.com/4Z0e6f4Z

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://pastebin.com/GXEmUHeG

#### Link to octoprint.log

Link to octoprint.log:  Emptied logs before the print, octoprint.log was empty.

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2020-06-14:
I am debugging a similar issue for a user running macos, so I strongly suspect this is os/compiler specific.  Will message back if I need more info.  Thx for posting!

BirkirFreyr, 2020-06-14:
Just for debug purposes - i reset everything to defaults, only imported my printer and used default settings for all (auto)
Now i get this error ( which is the error that I got at first ) - This is with all default settings - Stabilization set to Centered and trigger is Smart - Compatibility:
More than happy to test with you if needed, just let me know :)
```
2020-06-14 15:55:18,249 - octolapse.timelapse - DEBUG - Sent: G1 X122.556 Y95.826 E0.01216
2020-06-14 15:55:18,261 - octolapse.timelapse - DEBUG - Sent: G1 X123.492 Y95.627 E0.01216
2020-06-14 15:55:18,294 - octolapse.timelapse - DEBUG - Setting job-on-hold lock.
2020-06-14 15:55:18,294 - octolapse.timelapse - DEBUG - Sent: G1 X123.910 Y95.569 E0.00537
2020-06-14 15:55:18,295 - octolapse.timelapse - INFO - About to take a snapshot.  Triggering Command: G1 X124.817 Y95.506 E0.01155
2020-06-14 15:55:18,296 - octolapse.timelapse - ERROR - Failed to take a snapshot for the provided snapshot plan.
Traceback (most recent call last):
  File "/opt/octoprint/op_venv/lib/python3.6/site-packages/octoprint_octolapse/timelapse.py", line 409, in _take_timelapse_snapshot_precalculated
    self._trigger_profile.get_snapshot_plan_options()
  File "/opt/octoprint/op_venv/lib/python3.6/site-packages/octoprint_octolapse/stabilization_gcode.py", line 1107, in create_gcode_for_snapshot_plan
    snapshot_plan, g90_influences_extruder, options
  File "/opt/octoprint/op_venv/lib/python3.6/site-packages/octoprint_octolapse/stabilization_gcode.py", line 363, in initialize_for_snapshot_plan_processing
    self.distance_to_lift = snapshot_plan.initial_position.distance_to_zlift(self.z_lift_height)
  File "/opt/octoprint/op_venv/lib/python3.6/site-packages/octoprint_octolapse/gcode_processor.py", line 515, in distance_to_zlift
    if amount_to_lift < utility.FLOAT_MATH_EQUALITY_RANGE:
TypeError: '<' not supported between instances of 'NoneType' and 'float'
2020-06-14 15:55:18,296 - octolapse.timelapse - INFO - The snapshot has completed
2020-06-14 15:55:18,297 - octolapse.timelapse - INFO - Sending on_snapshot_complete payload.
2020-06-14 15:55:18,298 - octolapse.timelapse - DEBUG - Releasing job-on-hold lock.
2020-06-14 15:55:18,364 - octolapse.timelapse - DEBUG - Sent: G1 X125.596 Y95.524 E0.00991
2020-06-14 15:55:18,409 - octolapse.timelapse - DEBUG - Sent: G1 X126.598 Y95.641 E0.01281
```

FormerLurker, 2020-06-15:
FYI, I'm going to have to add additional logging to figure this out.  Your report is one of two similar reports, and they all seem to be problems sending the snapshot plans from the c++ extension to OctoPrint.  I will send you a link to install a version with improved logging that can verify this problem as soon as it is completed.  Once I am sure that the problem is sending the snapshot plans to python I will take more steps to pinpoint exactly where it is occurring.

FormerLurker, 2020-11-17:
@BirkirFreyr , I know it's been a LONG time, but it's possible your issue has finally been solved.  Can you try installing from the plugin manager using this link in the From Url... box: https://github.com/FormerLurker/Octolapse/archive/88aa71f0fbac306b0d9acf19dff7388a30b29b64.zip

See if that fixes it.

BirkirFreyr, 2020-11-17:
Hi, i'd love to test it, but since my setup has changed a bit since this came up, i can't. I'm now running octoprint through octopi on a RPi 4 :)

FormerLurker, 2020-11-17:
I totally understand.  Well, if you are running Octolapse, definitely let me know if you have any problems.  Going to release 0.4.1 soon, which includes the fix I posted.  Going to close this issue.  Thanks for your post!

closed 2020-11-17 by FormerLurker