FormerLurker/Octolapse #547: Octolapse - Camera Script Failed

ME-Marty, 2020-06-16:
#### If this is a feature request describe it here

Just wanted to say thanks for the help in advance ! 

#### Version of Octolapse
 
Octolapse Version: Octolapse 0.4.0

#### Version of OctoPrint

OctoPrint Version: OctoPrint 1.4.0 running on OctoPi 0.17.0

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred
1. After getting good results with the octolapse plugin with a webcam, I decided to test out my DSLR. I watched your youtube video tutorial exactly. 
2. I set up a new profile for the DSLR in octolapse. Selected "external camera - script for "camera type" and I entered the script itself - "/home/pi/scripts/take-snapshot.sh". Then I went to test the script and I received this error -
[image]

Below is a image of my octolapse settings:

[image]

3. Enabled "test mode" and selected "debug" under logging. Started my print and I received two failed snapshots and I decided to report my logs. 

#### What should have happened?
I believe the DSLR should have taken snapshots of the print successfully. It is a cannon EOS 700D which is supported to my knowledge and It was recognized by this command "gphoto2 --auto-detect" It also actuated and took a picture when given the command - "gphoto2 --capture-image" 

#### What happened instead?
Octolapse gave me an error "2 failed snapshots" 

#### Operating System running OctoPrint and Octolapse

OS Name: Octopi - running on raspberry-pi 
Os Version: OctoPi 0.17.0

#### Printer model & used firmware incl. version

Printer Model: Creality Ender 3 Pro
Printer Firmware Version: Just bought it brand new a month ago - not entirely sure

#### Browser and version of browser, operating system running browser

Browser: Chrome
Browser OS: ___REPLACE_THIS__BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: https://www.dropbox.com/s/js2w7iv6b5vimrh/CE3PRO_7-8_Bearing_ID_Test.gcode?dl=0

#### Link to settings.json

Link to settings.json with all passwords removed: https://www.dropbox.com/s/dh299hjjuefh8wt/settings.json?dl=0

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://www.dropbox.com/s/thcwnpn0gjn3suj/plugin_octolapse%20%281%29.log?dl=0

#### Link to octoprint.log

Link to octoprint.log: https://www.dropbox.com/s/7wzpl83ryzwwkmf/octoprint.log?dl=0

#### Link to contents of Javascript console in the browser

Link to javascript console output: [image]

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: https://www.dropbox.com/s/wcoyvcgzrpxt0r3/Octolapse_screenshot.PNG?dl=0

FormerLurker, 2020-06-16:
I found this in the octolapse log:

```
2020-06-16 03:18:36,924 - octolapse.script - DEBUG - Executing DSLR - Snapshot Camera Script: "/home/pi/scripts/take-snapshot.sh" "0" "0" "/tmp/tmpHYBR4z" "/tmp/tmpHYBR4z/0deb4c61-a26e-47af-a6f2-0a70c5c5c9f2/f840c2c7-2c9d-4d18-acbc-db96f8c3c75e" "test_snapshot000000.jpg" "/tmp/tmpHYBR4z/0deb4c61-a26e-47af-a6f2-0a70c5c5c9f2/f840c2c7-2c9d-4d18-acbc-db96f8c3c75e/test_snapshot000000.jpg" "DSLR"
2020-06-16 03:18:37,104 - octolapse.script - DEBUG - Console output (stdout) for 'DSLR - Snapshot Camera Script':Creating directory: /tmp/tmpHYBR4z/0deb4c61-a26e-47af-a6f2-0a70c5c5c9f2/f840c2c7-2c9d-4d18-acbc-db96f8c3c75e
2020-06-16 03:18:37,106 - octolapse.script - ERROR - Error output (stderr) for 'DSLR - Snapshot Camera Script':
	sudo: no tty present and no askpass program specified
	The snapshot was not found in the expected directory: '/tmp/tmpHYBR4z/0deb4c61-a26e-47af-a6f2-0a70c5c5c9f2/f840c2c7-2c9d-4d18-acbc-db96f8c3c75e/test_snapshot000000.jpg'.
```

Looks like your sudoers file has not been configured correctly.  Most likely, the path to gphoto2 is incorrect.  Reread [this guide](https://github.com/FormerLurker/Octolapse/wiki/V0.4---GPhoto2-Installation#step-3---configure-etcsudoers-file), and pay extra close attention to the bit that talks about using this command: ```whereis gphoto2```

I will probably move that section to make it more clear, but my guess is that your instance of gphoto2 is NOT at the location in the script example.  Let me know if that's the case.

ME-Marty, 2020-06-16:
I used the command: "whereis gphoto2" it returned the following: 
[image]

FormerLurker, 2020-06-16:
Can you post the contents of your sudoers file?

ME-Marty, 2020-06-16:
I believe this is what you are looking for hopefully - 
[image]

FormerLurker, 2020-06-16:
I dont see the line in there for gphoto2.  Please review the guide I linked to above and add the necessary line at the end.  Be sure your directory matches the whois directory for gphoto2: /usr/bin/gphoto2

ME-Marty, 2020-06-16:
Thanks again for the help, its very much appreciated dude. 

I added the necessary line - 
[image]

I used the command - "whereis gphoto2" it returned this - 
[image]

I'm still receiving errors connecting when I test the external camera script - 
[image]

FormerLurker, 2020-06-16:
Did you reboot after modifying sudoers?  If not, do that now.  If it still doesn't work, edit your debug profile, click clear log, retest the camera, and post the resulting log file.

FormerLurker, 2020-06-16:
Correction-> when i said debug profile I meant logging profile.

FormerLurker, 2020-06-16:
Also, a full reboot of the pi is required, not just restarting octoprint..  sorry about sending my half thoughts.  Work has been busy and I am rushing things.

FormerLurker, 2020-06-23:
Any update here?

ME-Marty, 2020-06-23:
Hey, 

thanks again for all your help I havent gotten back due to project overload at work, I got it working late last night! it was user error

Naver32, 2020-09-20:
I have the same problem.
Everything works perfect up to here:

New file is in location /capt0000.jpg on the camera
Saving file as /home/pi/scripts/test.jpg
Deleting file /capt0000.jpg on the camera
If you enter ls, you should see test.jpg in the current directory. You can delete the test image by entering rm test.jpg.

When I give test script it does the same to me

Naver32, 2020-09-20:
Excellent tutorial. I thank you!
I only need the last step; (

Naver32, 2020-09-20:
friend you have a mistake here
https://github.com/FormerLurker/Octolapse/wiki/V0.4---GPhoto2-Installation

the / local / is missing

FormerLurker, 2020-09-20:
@Naver32, local is missing from where?  Can you take a screenshot and mark where it shout be?

Shawnyd, 2020-10-31:
> @Naver32, local is missing from where? Can you take a screenshot and mark where it shout be?

Hey guys, I was running into an issue where my External Camera Setup - Script was getting an error. I followed the instructions to the letter for both the Gphoto2 and Former Lurker Tutorials (great job by the way). 

I noticed an issue with the Gphoto2 installation wiki that was missing a local in the 'Step 3 - Configure /etc/sudoers file' section.
https://github.com/FormerLurker/Octolapse/wiki/V0.4---GPhoto2-Installation#step-3---configure-etcsudoers-file

Scroll down to where it says "add the following lines to the VERY END of the file:"
Inspect the string for the directory. It is missing one critical part /local/

This is what they say to post;
# allow 'sudo gphoto2' to run without supplying a password
pi ALL = (root) NOPASSWD: /usr/bin/gphoto2

This is what I corrected my sudoers file to look like;
# allow 'sudo gphoto2' to run without supplying a password
pi ALL = (root) NOPASSWD: /usr/local/bin/gphoto2

This solved my issue and Octolapse camera profile is now successfully sending the External Camera Script. 

This is the first time I have ever posted on Github so pelase excuse my bad methods of sharing informaiton. 

Thanks again for being such an awesome dude Former Lurker. You seriously are a godsend with all this stuff.

Cheers!

-Shawn

FormerLurker, 2020-10-31:
Hi shawn, thanks!  I guess I need to add a step that prints the location of gphoto2 (the setup is geared more towards octopi users).  I added that in another step, but it looks like I missed it there.  Nice catch, thanks, and enjoy!  Happy Halloween (if you are into that kind if thing)!

Shawnyd, 2020-10-31:
Thanks! I'm a newbie to all of this and surprisingly it is all going well. Entirely thanks to your wonderful work. Seriously man couldn't do this without you and it is such a fun aspect of this whole 3d printing process.

Thanks man! Happy Halloween to you as well.  Halloween is my favorite Holiday.

JardianJohn, 2021-05-25:
Please help me, I have the same error as ME-Marty, I see the gphoto2 file is the same as the ME-Marty file. I edited but the error is still not fixed, I don't understand why the archive file of gphoto2 is different from the file of the author FormerLurker, I have read all the dialogue between Me-Marty and the author but can't see Me- Marty answered about the problem is solved or not?, FormerLurker please make a new video in 2021 to solve the above problem, because Octopi is now 0.18.0 with many changes. Thanks for reading :) [](https://github.com/FormerLurker/Octolapse/issues/547#issuecomment-644960526).

FormerLurker, 2021-06-01:
@JardianJohn, the error message that me-marty received is pretty generic.  Your best bet is to create a log file by setting your Octolapse logging profile to 'Debug'.  Then run a test to generate the error message.  Next, set your logging profile to 'log all errors', download plugin_octolapse.log, post it to gist.github.com, and I'll take a look at it.  Might need to create a new issue depending on what the error turns out to be.

Regarding that missing section about the gphoto2 location, I'll see if I can add that in the next day or two.  I'll take a look at the video, and will update it if necessary (it probably will be), but that will take longer :(

FormerLurker, 2021-06-01:
Reopening this until the instructions are updated.

JardianJohn, 2021-06-01:
> Reopening this until the instructions are updated.
Thank you for your reply. I fixed the error by installing DSLR with [Raspberry,](https://pimylifeup.com/raspberry-pi-dslr-camera-control/) in the article in the tutorial, in the 7th step there is setting “/usr/local/lib”. I followed and succeeded, along with your instructions it worked. You can refer and guide people. I am very happy to receive a reply from the author :))). Currently I am trying Octolaps with no download. It is also giving error but I will try to find out, I appreciate your help. My English is not good, there are many mistakes, I hope you will understand and sympathize.

FormerLurker, 2021-06-01:
@JardianJohn,

Thanks for your update!  Don't worry about grammar, especially because I understand you perfectly.  I suspect your english is much better than you realize :)  If I am unclear about anything, I will ask.

fiveorders, 2021-06-30:
> Thank you for your reply. I fixed the error by installing DSLR with [Raspberry,](https://pimylifeup.com/raspberry-pi-dslr-camera-control/) in the article in the tutorial, in the 7th step there is [...]

Hello, I am getting the same error with Canon R5 and me followed every tutorial and no help. Still getting this error, camera taking photos, and it renders however it gives this error on every snapshot. 

octolapse.script - ERROR - Error output (stderr) for 'Canon R5 - Snapshot Camera Script':
	ERROR: Could not get image.

/usr/local/lib it was already included on my setup. Everything up to date and upgraded. Any ideas? Many thanks

fiveorders, 2021-06-30:
I think I solved the problem by disabling RAW images. I set it to Jpeg only and error gone.

TerrorInc, 2022-10-17:
Sorry to revive an old post, but I for the life of me cannot get this working, this is the error produced in my log, it takes a photo, but then just spits this out

`2022-10-17 14:26:48,637 - octolapse.script - ERROR - The 'DSLR - Snapshot Camera Script' process has timed out before completing.  Attempting to kill the process.
2022-10-17 14:26:48,669 - octolapse.script - ERROR - An error occurred while killing the 'DSLR - Snapshot Camera Script' process.
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/psutil/__init__.py", line 1176, in _send_signal
    os.kill(self.pid, sig)
PermissionError: [Errno 1] Operation not permitted

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_octolapse/script.py", line 103, in kill
    proc.kill()
  File "/home/pi/oprint/lib/python3.7/site-packages/psutil/__init__.py", line 272, in wrapper
    return fun(self, *args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/psutil/__init__.py", line 1239, in kill
    self._send_signal(signal.SIGKILL)
  File "/home/pi/oprint/lib/python3.7/site-packages/psutil/__init__.py", line 1186, in _send_signal
    raise AccessDenied(self.pid, self._name)
psutil.AccessDenied: psutil.AccessDenied (pid=759)
2022-10-17 14:26:53,362 - octolapse.__init__ - ERROR - An unexpected error occurred while executing the snapshot script.
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/psutil/__init__.py", line 1176, in _send_signal
    os.kill(self.pid, sig)
PermissionError: [Errno 1] Operation not permitted

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_octolapse/script.py", line 103, in kill
    proc.kill()
  File "/home/pi/oprint/lib/python3.7/site-packages/psutil/__init__.py", line 272, in wrapper
    return fun(self, *args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/psutil/__init__.py", line 1239, in kill
    self._send_signal(signal.SIGKILL)
  File "/home/pi/oprint/lib/python3.7/site-packages/psutil/__init__.py", line 1186, in _send_signal
    raise AccessDenied(self.pid, self._name)
psutil.AccessDenied: psutil.AccessDenied (pid=759)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_octolapse/script.py", line 109, in kill
    self._kill_exceptions.append(e)
AttributeError: 'NoneType' object has no attribute 'append'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_octolapse/__init__.py", line 1395, in test_camera_script_request
    success, errors, snapshot_created = camera.CameraControl.test_script(camera_profile, script_type, self._basefolder)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_octolapse/camera.py", line 645, in test_script
    success, error, snapshot_created = CameraControl._test_camera_snapshot_script(camera_profile, script_type, base_folder)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_octolapse/camera.py", line 694, in _test_camera_snapshot_script
    cmd.run()
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_octolapse/script.py", line 685, in run
    return super(CameraScriptSnapshot, self).run(self.get_args(), self.timeout_seconds)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_octolapse/script.py", line 218, in run
    self._run(args, timeout_seconds=timeout_seconds)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_octolapse/script.py", line 281, in _run
    self.kill()
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_octolapse/script.py", line 120, in kill
    self.read_output_from_proc()
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_octolapse/script.py", line 156, in read_output_from_proc
    (exc_stdout, exc_stderr) = self.proc.communicate()
  File "/usr/lib/python3.7/subprocess.py", line 939, in communicate
    stdout, stderr = self._communicate(input, endtime, timeout)
  File "/usr/lib/python3.7/subprocess.py", line 1701, in _communicate
    data = os.read(key.fd, 32768)
OSError: [Errno 9] Bad file descriptor`

[image]

Any help would be greatly appreciated

SB3D9, 2024-03-03:
I got my camera (Sony ZV-1) to connect for the Control tab of octoprint using its Movie, streaming mode setting.
But that mode won't be detected w/  -gphoto2 --auto-detect-  It does detect the camera in it's regular picture mode.

I did everything according to your wonderful tutorial and I got to **Step 6 - Configure Octolapse**
When I test the external script I get a generic error "Camera Script Test Failed" I tried redoing steps but it all still ends the same can you help a guy out?

here are some screenshots

[image]
[image]
[image]