FormerLurker/Octolapse #548: Render script appears to ignore configured ffmpeg path

brilthor, 2020-06-16:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: v0.4.0

#### Version of OctoPrint

OctoPrint Version:  1.4.0

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred

1. Run Octoprint/Octolapse with non-standard ffmpeg path
2. Configure FFMPEG path in octoprint
3. Octolapse passes sanity check for path
4. Octolapse fails rendering unable to find ffmpeg path

#### What should have happened?
Rendering should have succeeded with the configured ffmpeg binary

#### What happened instead?
errored out ( UI shows rendering failed, log has more information (below))

Creating a symlink for the path that appears to be hardcoded allowed the rendering to complete

#### Operating System running OctoPrint and Octolapse

OS Name: Linux
Os Version: Ubuntu 18.04 (Docker Container)

#### Link to settings.json

Link to settings.json with all passwords removed: https://gist.github.com/brilthor/46ed36859dc15157eaf9dd52c7f449f8

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://gist.github.com/brilthor/bb9b46287998efb1d2c0acbfdbd18d49

#### Link to octoprint.log

Link to octoprint.log:  https://gist.github.com/brilthor/8a9e413ee9ae11f54ff87e9de3ddb5f8

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2020-06-16:
I thought I fixed this, but apparently not.  Can you see if a reboot fixes this as a workaround?  I will figure out what's up with the setting and will update the ticket as soon as it's fixed.

brilthor, 2020-06-16:
Confirmed rebooting appears to make the issue go away.  Presumably this means Octolapse is holding on to the value from when it starts up or was first asked to do a run?

FormerLurker, 2020-06-16:
Possibly, I've not determined it yet.  Rendering is done on a separate thread, and I detect changes to any OctoPrint settings and pass them to the thread.  There are a few potential things that could be going wrong here.

FormerLurker, 2020-06-23:
OK, I haven't gotten to this yet, but wanted to let you know I haven't forgotten.  I'll keep this open until the bug is found/patched.