FormerLurker/Octolapse #549: After print initial layer. Nozzle goes to outside of bed bit by bit.

venomwine, 2020-06-16:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: v0.4.0

#### Version of OctoPrint

OctoPrint Version: 1.4.0

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred

1. I followed step by step your new "V0.4 Getting Started". (https://github.com/FormerLurker/Octolapse/wiki/V0.4---Getting-Started)
2. When I reached this step. (https://github.com/FormerLurker/Octolapse/wiki/V0.4---Getting-Started#snapshot-plan-preview)
3. The "Snapshot Plan" won't come out.
4. I decided turn the "Snapshot Plan Preview" off. And Print.

#### What should have happened?
This is what I print. (https://youtu.be/5LUPoHThWCI)

#### What happened instead?
After print initial layer, The nozzle goes to outside of bed bit by bit. (https://youtu.be/uONG34c5n0c?t=74)

#### Operating System running OctoPrint and Octolapse

OS Name: Linux raspberrypi
Os Version: Linux raspberrypi 5.4.42-v8+ #1319 SMP PREEMPT Wed May 20 14:18:56 BST 2020 aarch64

#### Printer model & used firmware incl. version

Printer Model: Anycubic Kossel Linear Plus
Printer Firmware Version: Marlin-2.0.5.3

#### Browser and version of browser, operating system running browser

Browser: Firefox 77.0.1 (64-bit)
Browser OS: Windows 10

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: https://drive.google.com/file/d/1f9Di8R0H7Ff1cUPsVwbKzL4o1cIsFyY_/view?usp=sharing

#### Link to settings.json

Link to settings.json with all passwords removed: https://drive.google.com/file/d/1za4OK9gkEDezbHgf5oRjaNjy_vNsb7eX/view?usp=sharing

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://docs.google.com/document/d/1181v5xcZc5mDk6l-FVl1e_cwG_AL0gRQagc-iJuHEZc/edit?usp=sharing

#### Link to octoprint.log

Link to octoprint.log:  https://docs.google.com/document/d/1t8DOCPLsdPwCklJlV561e6Jk9bFvUSvr7NssJtUv5ww/edit?usp=sharing

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: https://youtu.be/uONG34c5n0c?t=74

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2020-06-16:
It looks like your printer is not waiting for moves to finish when M400 is sent.  I would need a debug log to be certain.  I only need a few layers logged (maybe 5) from a small print.  See [this guide](https://github.com/FormerLurker/Octolapse/wiki/V0.4---Reporting-An-Issue#creating-an-octolapse-log-file) that explains how to post a debug log.

Your printer may use a different command for the 'wait for print moves to finish' (G4 P0).  I have not yet included this capability in Octolapse, but plan to do so soon.  If we determine that your printer should be using the alternative command, perhaps you can help me to test this.

venomwine, 2020-06-17:
I uploaded new log files.

[**Link to settings.json**](https://drive.google.com/file/d/1b1OI2blVdX7Yiqb7gWGGLZW8wyH6Ly4C/view?usp=sharing)

[**Link to plugin_octolapse.log**](https://docs.google.com/document/d/1bsMrrs82MB7vI12uAIuFIrlCTTq88pSxH4tCms07ziA/edit?usp=sharing)

[**Link to octoprint.log**](https://docs.google.com/document/d/1OFiJyOjHNOEsH3f6sUiS2E6qPWYt1X7qQYDxzGcFqa0/edit?usp=sharing)

[**New Timelapse Video**](https://youtu.be/8J2fcqGeVcY)

And I used the same gcode file.
I hope it helps you. Always thank you!

FormerLurker, 2020-06-17:
OK, it's almost certain that your printer is not executing M400 as expected by Octolapse.  See these two log entries:

```
2020-06-17 15:25:26,522 - octolapse.timelapse - DEBUG - Waiting for moves to complete before continuing - sent: M114
2020-06-17 15:25:26,530 - octolapse.timelapse - INFO - Octolapse has received a position request response.
```
Only 8MS elapsed between sending the M114 command and receiving a response.  This is not nearly enough time to clear out the print buffer.  What firmware version are you running (a link to source would be great)?  I've seen people use this printer before, and am unsure why yours is misbehaving here.

venomwine, 2020-06-18:
I'm using latest version of Marlin.
[This is what I downloaded. ](https://github.com/MarlinFirmware/Marlin/archive/2.0.x.zip)
And This link comes from [this web page.](https://marlinfw.org/meta/download/)

I downloaded this [configurations file](https://github.com/MarlinFirmware/Configurations/archive/release-2.0.5.zip)
and copy \Configurations-release-2.0.5\config\examples\delta\Anycubic\Kossel\*.h files to \Marlin-2.0.5.3\Marlin\.

and changed some values of header files.
Here are my [Configuration.h](https://docs.google.com/document/d/1C2_LTW8SxR5azektPNgfIN6gbMbYKLVg2G36mSss1_s/edit?usp=sharing) and [Configuration_adv.h](https://docs.google.com/document/d/1EyPQy-CuaUqnD_0T6RcdIeyfAGLRWvYsgdZmh3QJg-U/edit?usp=sharing) files.

Thank you!

venomwine, 2020-06-19:
Dear FormerLurker,

I found a reason. I was using 64-bit Raspberry Pi OS. 
I think that makes problem. I re-installed 32-bit Raspberry Pi OS. And your Octolapse works perfect!!
[https://youtu.be/LAeP5OD8NRw](https://youtu.be/LAeP5OD8NRw)

Thanks and sorry.

FormerLurker, 2020-06-23:
Hmm..  This is a very interesting find.  No reason whatsoever to apologize, as this is a real problem.  Perhaps I can replicate the issue on the 64 bit OS and fix it now.   I may re-title the issue or open a new issue since we know what the problem is.

venomwine, 2020-06-24:
Ok. My device is Raspberry Pi 4 8GB. Any other information are already mentioned. If you need more further information. let me know. I'll do my best. Thanks!!

FormerLurker, 2020-06-24:
Thanks @venomwine!