FormerLurker/Octolapse #550: [Request] Update plugin incompatibilities listed on plugins.octoprint.org

RaymondHimle, 2020-06-17:
The Spaghetti Detective now supports using mjpeg-streamer for our 25FPS streaming (abet with more CPU usage) and as always supports MJPEG-streamer without 25FPS TSD streaming. Could this wording on plugins.octoprint.org be updated to reflect a partial compatibility instead of seeming like a flat out incompatibility as the current wording suggests?

FormerLurker, 2020-06-17:
I sure would!  I love that plug-in :)

RaymondHimle, 2020-06-17:
Thank you very much! We really appreciate this.

FormerLurker, 2020-06-17:
So, I should have some time to do this on Saturday unless something comes up.  I will note that some additional configuration is required for The Spaghetti Detective.  Do you have a link to instructions?  Also, do you know if this applies to Octoprint Anywhere too?  If so, so a link to some instructions for getting mjpgstreamer working would be great.

Also, has anyone tested the mjpgstreamer config to make sure it works with the Octolapse custom image preferences?  I'm pretty sure it should work, but a test would be great.

RaymondHimle, 2020-06-17:
I'll have to test myself once my current print is done. But it is a bubble selection to enable. See this page https://www.thespaghettidetective.com/docs/streaming-compatibility-mode/
CPU usage can be a problem on older Pis (3B and older) due to the reencoding we do to H264 for the web streaming. But it is at least an option available for those who want best of both worlds. For reference I use a Pi4B with TSD and Compatibility mode without issue

FormerLurker, 2020-06-17:
Excellent, i will mention that tgis works best on a pi4 as well.

FormerLurker, 2020-06-23:
@RaymondHimle, I haven't had time to get to this yet, sorry.  Wanted to give you an update though.  I'll try to sneak it in ASAP, and will provide another update.  Pester me if you don't hear back by next week :)