FormerLurker/Octolapse #551: Starting gcode error

xcjacob30, 2020-06-18:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: ___0.4.0

#### Version of OctoPrint

OctoPrint Version: ___1.4.0

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: ___YES

#### What were you doing when the problem occurred

1. ___Started gcode
2. ___Error happened

#### What should have happened?
___Gcode start

#### What happened instead?
___Error

#### Operating System running OctoPrint and Octolapse

OS Name: ___Octopi
Os Version: ___0.17.0

#### Printer model & used firmware incl. version

Printer Model: ___cr-10 v2
Printer Firmware Version: ___Marlin 2.0

#### Browser and version of browser, operating system running browser

Browser: ___REPLACE_THIS__BROWSER_VERSION_GOES_HERE
Browser OS: ___REPLACE_THIS__BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: ___https://gist.github.com/xcjacob30/1aa826b7dbdebd127790b830c6b7c774

#### Link to settings.json

Link to settings.json with all passwords removed: ___https://gist.github.com/xcjacob30/1aa826b7dbdebd127790b830c6b7c774

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  ___https://gist.github.com/xcjacob30/1aa826b7dbdebd127790b830c6b7c774

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2020-06-18:
In the future please provide a better description of the problem than simply saying there was an error.  Knowing what error specifically, seeing a screen shot, or some other description is required, otherwise people looking at the issue would need to dig through the logs and make a guess about what the problem is.  Obviously that is unhelpful for anyone who might be having the same issue as you.

I was able to go through the plugin_octolapse.log file and see several errors.  Some look like they have been corrected (invalid ffmpeg path), but others do not.  If I'm right about the error you are having now, you should have seen an error toast that contained a help link (did you click the help button and see what it said??).

I looked at your gcode file, and it contains the settings script, but the settings tokens were NOT replaced by cura:

```
; Script based on an original created by tjjfvi (https://github.com/tjjfvi)
; An up-to-date version of the tjjfvi's original script can be found
; here:  https://csi.t6.fyi/
; Note - This script will only work in Cura V4.2 and above!
; --- Global Settings
; layer_height = {layer_height}
; smooth_spiralized_contours = {smooth_spiralized_contours}
; magic_mesh_surface_mode = {magic_mesh_surface_mode}
; machine_extruder_count = {machine_extruder_count}
; --- Single Extruder Settings
; speed_z_hop = {speed_z_hop}
; retraction_amount = {retraction_amount}
; retraction_hop = {retraction_hop}
; retraction_hop_enabled = {retraction_hop_enabled}
; retraction_enable = {retraction_enable}
; retraction_speed = {retraction_speed}
; retraction_retract_speed = {retraction_retract_speed}
; retraction_prime_speed = {retraction_prime_speed}
; speed_travel = {speed_travel
```

If you look at the last line above, you are missing a closing curly brace.  It should be :

```
; speed_travel = {speed_travel}
```

Replace that line and it should start working.

xcjacob30, 2020-06-18:
I'm very sorry for the trouble. I will take better care next time. Thank
you and have a great day.

FormerLurker, 2020-06-18:
@xcjacob30, it wasn't too much trouble, and I was glad to help.  please let me know if this fixes your problem.  I will adjust the title and issue description.

FormerLurker, 2020-06-23:
@xcjacob30, is this issue solved?

xcjacob30, 2020-06-23:
Yes sorry I forgot

FormerLurker, 2020-06-23:
No problem!  I'll go ahead and close this.  Enjoy the plugin!

closed 2020-06-23 by FormerLurker