FormerLurker/Octolapse #552: Timelapse video was not created

bitshiftr, 2020-06-20:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.4.0

#### Version of OctoPrint

OctoPrint Version: 1.4.0

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: NO

#### What were you doing when the problem occurred

1. Snapshot files are available, but video was not created

#### What should have happened?
Timelapse video should have been created

#### What happened instead?
All the snapshots are available, but video was not created
Snapshots are in this directory - ~/.octoprint/data/octolapse/tmp/octolapse_snapshots_tmp/a2541918-bc90-44f0-ba97-b35847b167d7/354def78-9eea-409a-ad23-ee966dfff4ba

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: 0.17.0

#### Printer model & used firmware incl. version

Printer Model: ___REPLACE_THIS__PRINTER_MODEL_GOES_HERE
Printer Firmware Version: ___REPLACE_THIS__PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: ___REPLACE_THIS__BROWSER_VERSION_GOES_HERE
Browser OS: ___REPLACE_THIS__BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: ___REPLACE_THIS__GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: ___REPLACE_THIS__SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://pastebin.com/1q3V7V3q

#### Link to octoprint.log

Link to octoprint.log:  https://pastebin.com/jg2svhGf

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

bitshiftr, 2020-06-20:
All of my past prints (both successful and failed ones) have had videos created successfully using this version of Octolapse. This is the first time this has happened. How do I go about using the existing snapshots to create a video?

FormerLurker, 2020-06-23:
It could be that the rendering thread crashed for some reason (out of memory perhaps depending on the codec), but I don't see any exceptions in the log.  If this happens again, try restarting OctoPrint and see if the failed rendering shows up in the Octolapse tab (should be a big red button on the tab).  If you can replicate the issue I think we could log it and figure out what is going on.

sanmadjack, 2020-09-03:
I'm also experiencing this issue, it creates the snapshots but then does not output a video. I don't see any exceptions in my log, and when I restart octoprint it shows the red button saying there are unfinished timelapses. Clicking it and rendering the timelapses then succeeds.

FormerLurker, 2020-09-03:
@sanmadjack, perhaps you could enable logging of the rendering module?  Edit the 'logging' profile, and add this logger:

[image]

Just click 'customize Profile', check 'Additional Logging', select 'octolapse.render' as the module to log, select 'Verbose' as the logging level, and click the blue plus to add the logger.  That way, the next time the logger fails we should at least some details about the most recent completed operation, which should narrow in on the problem.

Since you didn't see any exceptions, something must be failing that isn't caught properly, which is in itself an error.

Thanks!

sanmadjack, 2020-09-04:
Unfortunately (?) the print I did after restarting successfully produced a timelapse. I'll turn on that logging and if it happens again I'll post here.

FormerLurker, 2020-09-04:
Cool.  Will leave this issue open for a while.