FormerLurker/Octolapse #554: Printer "finishing" after first snapshot only when octolapse enabled

drcheap, 2020-06-26:
I recently got Octolapse set up using a Pi 3B+ & the Pi Cam V2.  Setup was not too bad, and it created a nice timelapse in test mode.  However, I am running into a major issue that prevents it from actually being usable :(

Using Octoprint by itself prints okay, but when Octolapse is enabled, the printer mistakenly thinks the print is "finished" as soon as the first snapshot is taken.  The actual print head & extruder keep going though!  However, other things that normally happen at the end of a print job occur, which includes the heater/fans turning off, and eventually back to the idle mode on the LCD.

At this point, with the printer still going & and the hot end becoming _not hot_, the extrusion gets backed up and a catastrophe ensues if not aborted quickly!

Here's the full/official bug report:

#### If this is a feature request describe it here

(not feature request)

#### Version of Octolapse
 
Octolapse Version: 0.4.0

#### Version of OctoPrint

OctoPrint Version: 1.4.0

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred

1. Printing a simple test file of a tall skinny tower.

#### What should have happened?
Print as usual, with a snapshot taken at each layer.
The printer should have kept fans running and status on LCD should indicate printing in progress throughout.

#### What happened instead?
After the first snapshot was taken, the printer indicated the print finished, and shut off system fan.
After a short delay, the LCD returned back to the idle screen.
Despite this, the print itself continued onward, and snapshots were taken at each layer.
Had I not been in test mode, the bed & extruder heaters would be turned off as well, eventually causing extrusion failure.  This happened once, but I caught it and aborted before any damage occurred.

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: 0.17.0

#### Printer model & used firmware incl. version

Printer Model: Monoprice Select Mini Pro
Printer Firmware Version: V51.266.2

#### Browser and version of browser, operating system running browser

Browser: Firefox 77.0.1
Browser OS: Windows 10.0.18362.836

#### Link to the gcode file you were printing when the problem occurred

https://www.drcheap.com/temp/ol1/test.gcode

#### Link to settings.json

https://www.drcheap.com/temp/ol1/settings.json

#### Link to plugin_octolapse.log

https://www.drcheap.com/temp/ol1/plugin_octolapse.log

#### Link to octoprint.log

https://www.drcheap.com/temp/ol1/octoprint.log

#### Link to contents of Javascript console in the browser

(do not have)

#### Screenshots and/or videos of the problem:

(do not have)

FormerLurker, 2020-06-29:
Hmm..  So, according to your settings, test mode is enabled.  I do not see any extrusion commands being sent to your printer in the log file.  If you are trying for a live print, disable test mode.

Next, I do not see any commands to set the hotend temp, at least in the Octolapse log.  This is what I would expect with test mode enabled.  So, the question is where do the commands come from that are turning on the heaters?  Are you running some other warm-up plugin?  

So, I recommend disabling all non-stock plugins, reboothing the pi, then disabling test mode in Octolapse and testing a live print.  Let me know how that goes.

drcheap, 2020-06-30:
Yes, this log was while in test mode -- and test mode does not turn on heaters (according to the help) so that makes sense that those codes got stripped instead of sent.

This is a fairly fresh OctoPi setup that I just recently configured, and Octolapse is the only plugin I have added.

I had also done live prints, but as mentioned it leads to a dangerous situation after it shuts off the heaters for whatever reason.  I think the heaters aspect is just a side effect though...it's more about what is causing the printer to believe the print is finished (whether in test or live mode) upon taking the first snapshot.

So I went for a session of furious round robin banging of "alt+tab, highlight, ctrl+c, alt+tab, paste, enter" to manually send all the gcode to the printer one command at a time through the terminal (okay, I cheated & left out some of the miniscule G0/G1 moves to save my sanity) until the printer did the wrong thing, and I may have a lead now.

Upon the first G28 is when the printer "woke up" and homed the printhead.  Got through all the early stuff (like the purge wipe down the side) and then over to the center of the print area where the print support would normally start.  Did some G1 moves, then got to where the snapshot trigger was going to happen.  It moving to the stabilization point as expected, and then as soon as I gave it the M400 the LCD went to "Finished" mode and fans shut down.

Does that help any?

drcheap, 2020-08-21:
I came back around to this issue again after a total rebuild of my OctoPrint setup.  When I was configuring Octolapse again I happened to open up the _advanced settings_ on my stabilization profile.  There I saw all this stuff about the "wait for moves to finish" and it being "a beta feature, use at your own risk!"

Other research I had done on the M400 is that this is the gcode for the "wait for moves to finish" so I tried disabling this just to see how things would go.  Sure enough, the printer doesn't kill the heater/fans with this disabled, but as warned on that profile it totally destabilized the timelapse, kind of defeating the purpose of the main feature Octolapse offers.

I just can't figure out why M400 causes the printer to shut off heater/fans, maybe it's a bug in the firmware on my specific printer?  Who knows.

FormerLurker, 2020-08-24:
There are some tests we could run to see what's going on.  M400 should just basically prevent the printer from accepting any new commands or responding until all moves are completed.  This is how Octolapse knows when it's safe to take a snapshot (printer must not be moving).  A small minority of printers respond to G4 P0 the same way M400 is supposed to work (and don't respond to M400), and I'm planning to add a switch for that.However, I've also seen printers where M400 doesn't work at all, but I've never seen it stop a print before...  Do you happen to have a link to your firmware?  If they have source I can take a look and see what they are doing with M400.

I'm pretty much unavailable for the next couple of weeks, but should be back in action sometime in September.  Sorry about any delays responding.

drcheap, 2020-08-25:
Yeah I don't know what the deal is with the M400 not working properly, it's definitely a quirk of this firmware.  I was playing around a lot with gcodes in the Terminal and trying various things at various timings.  It's pretty consistent that once it thinks a print has started, any instance of M400 will make it go to the "finished" indicator on the display (and other effects that happen as a result).  Would be interesting to know if others who use this or similar printers have the same experience.

I tried some G4 commands, and it does obey those properly it seems...
A lone G4 P0 while idle comes back with an immediate okay.
A G1 X100 followed quickly by G4 P0 (while still moving) returns the ok as soon as the move is complete.
Longer values (e.g. G4 P500 or even G4 S5) take their respective amount of time to return back with the ok.

So it seems I might be able to get away with the G4 P0 instead of M400 as you mentioned.  Is there a way for me to try swapping that out, or is the command currently baked into the plugin in such a way that I can't easily change it?

FormerLurker, 2020-08-25:
Someone else forked Octolapse and switched M400 to G4 P0.  You could probably find it by searching through the issues, or through the forks (there are quite a few).  I don't believe it was a big change.  

If not, I'm trying very hard to carve out some time for a new release, and this feature is high on the list since it would enable more firmware to work with Octolapse.  I may be able to add this feature in a fast and dirty way in a new branch for you to try.  Keep an eye on this issue, though it may be a few weeks.

drcheap, 2020-09-06:
I will keep an eye out for updates!

Thank you for the time you put into this, it is much appreciated.