FormerLurker/Octolapse #555: [Request] Camera settings: focus steps

rsaikali, 2020-06-26:
#### If this is a feature request describe it here

Current focus settings (slide setting in the UI) goes 5 by 5.
Either I choose `5` or `10` in my case, but I played in the octopi console and found that `7` is perfect to clearly focus on my print:

`sudo v4l2-ctl --set-ctrl=focus_absolute=7`

But I can't choose this `7` from the slide in the Octolapse UI.

Thanks.

#### Version of Octolapse
 
Octolapse Version: 0.4.0

#### Version of OctoPrint

OctoPrint Version: 1.4.0

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: ___REPLACE_THIS__YES_OR_NO

#### What were you doing when the problem occurred

1. ___REPLACE_THIS__STEP_ONE_GOES_HERE
2. ___REPLACE_THIS__STEP_TWO_GOES_HERE
3. ___REPLACE_THIS__STEP_...

#### What should have happened?
___REPLACE_THIS__PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
___REPLACE_THIS__PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: ___REPLACE_THIS__OS_NAME_GOES_HERE
Os Version: ___REPLACE_THIS__OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: ___REPLACE_THIS__PRINTER_MODEL_GOES_HERE
Printer Firmware Version: ___REPLACE_THIS__PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: ___REPLACE_THIS__BROWSER_VERSION_GOES_HERE
Browser OS: ___REPLACE_THIS__BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: ___REPLACE_THIS__GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: ___REPLACE_THIS__SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  ___REPLACE_THIS__LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

rickgode, 2020-08-02:
I'm having a similar challenge except in my case it is the exposure slider.  Mine goes from 0 to 10000 and I need to adjust mine to around 10, but the slider goes from 13 all the way down to 1 (which makes it completely dark)  I'd like the ability to either directly input the desired number or have little arrow or something to go one by one.  

**Version of Octolapse**
Octolapse Version: 0.4.0

**Version of OctoPrint**
OctoPrint Version: 1.4.0

FormerLurker, 2020-08-02:
Can you send me your input.json file?  Http://{Ip_of_pi}:8080/input.json

I am still hesitant to add text boxes since most cameras don't allow all (most actually) values.  Unsupported values are ignored, which is potentially even more confusing.

If your camera's focus operates between 0 and 10000, but the slider isnt operating over those values, there is either a big in octolapse, the camera driver, or mjpgstreamer.  Input.json should eliminate at least one of those possibilities.

rickgode, 2020-08-02:
On mine, it's not the focus, it's the exposure.  I'm using a Microsoft Lifecam HD camera (it's older, but seems to work so it maybe that?)  

{
"controls": [
{
"name": "Brightness",
"id": "9963776",
"type": "1",
"min": "30",
"max": "255",
"step": "1",
"default": "133",
"value": "0",
"dest": "0",
"flags": "0",
"group": "1"
},
{
"name": "Contrast",
"id": "9963777",
"type": "1",
"min": "0",
"max": "10",
"step": "1",
"default": "5",
"value": "1",
"dest": "0",
"flags": "0",
"group": "1"
},
{
"name": "Saturation",
"id": "9963778",
"type": "1",
"min": "0",
"max": "200",
"step": "1",
"default": "103",
"value": "91",
"dest": "0",
"flags": "0",
"group": "1"
},
{
"name": "White Balance Temperature, Auto",
"id": "9963788",
"type": "2",
"min": "0",
"max": "1",
"step": "1",
"default": "1",
"value": "0",
"dest": "0",
"flags": "0",
"group": "1"
},
{
"name": "Power Line Frequency",
"id": "9963800",
"type": "3",
"min": "0",
"max": "2",
"step": "1",
"default": "2",
"value": "2",
"dest": "0",
"flags": "0",
"group": "1",
"menu": {"0": "Disabled", "1": "50 Hz", "2": "60 Hz"}
},
{
"name": "White Balance Temperature",
"id": "9963802",
"type": "1",
"min": "2500",
"max": "10000",
"step": "1",
"default": "4500",
"value": "4338",
"dest": "0",
"flags": "16",
"group": "1"
},
{
"name": "Sharpness",
"id": "9963803",
"type": "1",
"min": "0",
"max": "50",
"step": "1",
"default": "25",
"value": "5",
"dest": "0",
"flags": "0",
"group": "1"
},
{
"name": "Backlight Compensation",
"id": "9963804",
"type": "1",
"min": "0",
"max": "10",
"step": "1",
"default": "0",
"value": "6",
"dest": "0",
"flags": "0",
"group": "1"
},
{
"name": "Exposure, Auto",
"id": "10094849",
"type": "3",
"min": "0",
"max": "3",
"step": "1",
"default": "3",
"value": "3",
"dest": "0",
"flags": "0",
"group": "1",
"menu": {"0": "", "1": "Manual Mode", "2": "", "3": "Aperture Priority Mode"}
},
{
"name": "Exposure (Absolute)",
"id": "10094850",
"type": "1",
"min": "1",
"max": "10000",
"step": "1",
"default": "156",
"value": "156",
"dest": "0",
"flags": "16",
"group": "1"
},
{
"name": "Pan (Absolute)",
"id": "10094856",
"type": "1",
"min": "-529200",
"max": "529200",
"step": "3600",
"default": "0",
"value": "0",
"dest": "0",
"flags": "0",
"group": "1"
},
{
"name": "Tilt (Absolute)",
"id": "10094857",
"type": "1",
"min": "-432000",
"max": "432000",
"step": "3600",
"default": "0",
"value": "0",
"dest": "0",
"flags": "0",
"group": "1"
},
{
"name": "Focus (absolute)",
"id": "10094858",
"type": "1",
"min": "0",
"max": "40",
"step": "1",
"default": "0",
"value": "0",
"dest": "0",
"flags": "16",
"group": "1"
},
{
"name": "Focus, Auto",
"id": "10094860",
"type": "2",
"min": "0",
"max": "1",
"step": "1",
"default": "1",
"value": "1",
"dest": "0",
"flags": "0",
"group": "1"
},
{
"name": "Zoom, Absolute",
"id": "10094861",
"type": "1",
"min": "0",
"max": "317",
"step": "1",
"default": "0",
"value": "0",
"dest": "0",
"flags": "0",
"group": "1"
},
{
"name": "JPEG quality",
"id": "1",
"type": "1",
"min": "0",
"max": "100",
"step": "1",
"default": "50",
"value": "0",
"dest": "0",
"flags": "0",
"group": "3"
}
],
"formats": [
{
"id": "0",
"name": "YUYV 4:2:2",
"compressed": "false",
"emulated": "false",
"current": "true",
"resolutions": {"0": "640x480", "1": "1280x720", "2": "960x544", "3": "800x448", "4": "640x360", "5": "424x240", "6": "352x288", "7": "320x240", "8": "800x600", "9": "176x144", "10": "160x120", "11": "1920x1080"}
,
"currentResolution": "255"
},
{
"id": "1",
"name": "Motion-JPEG",
"compressed": "true",
"emulated": "false",
"current": "true",
"resolutions": {"0": "640x480", "1": "1920x1080", "2": "1280x720", "3": "960x544", "4": "800x448", "5": "640x360", "6": "800x600", "7": "432x240", "8": "352x288", "9": "176x144", "10": "320x240", "11": "160x120"}
,
"currentResolution": "11"
},
{
"id": "2",
"name": "YUV 4:2:0 (M420)",
"compressed": "false",
"emulated": "false",
"current": "true",
"resolutions": {"0": "640x480", "1": "1280x720", "2": "960x544", "3": "800x448", "4": "640x360", "5": "424x240", "6": "352x288", "7": "320x240", "8": "800x600", "9": "176x144", "10": "160x120", "11": "1920x1080"}
,
"currentResolution": "255"
},
{
"id": "3",
"name": "RGB3",
"compressed": "false",
"emulated": "true",
"current": "true",
"resolutions": {"0": "640x480", "1": "1280x720", "2": "960x544", "3": "800x448", "4": "640x360", "5": "424x240", "6": "352x288", "7": "320x240", "8": "800x600", "9": "176x144", "10": "160x120", "11": "1920x1080", "12": "432x240"}
,
"currentResolution": "255"
},
{
"id": "4",
"name": "BGR3",
"compressed": "false",
"emulated": "true",
"current": "true",
"resolutions": {"0": "640x480", "1": "1280x720", "2": "960x544", "3": "800x448", "4": "640x360", "5": "424x240", "6": "352x288", "7": "320x240", "8": "800x600", "9": "176x144", "10": "160x120", "11": "1920x1080", "12": "432x240"}
,
"currentResolution": "255"
},
{
"id": "5",
"name": "YU12",
"compressed": "false",
"emulated": "true",
"current": "true",
"resolutions": {"0": "640x480", "1": "1280x720", "2": "960x544", "3": "800x448", "4": "640x360", "5": "424x240", "6": "352x288", "7": "320x240", "8": "800x600", "9": "176x144", "10": "160x120", "11": "1920x1080", "12": "432x240"}
,
"currentResolution": "255"
},
{
"id": "6",
"name": "YV12",
"compressed": "false",
"emulated": "true",
"current": "true",
"resolutions": {"0": "640x480", "1": "1280x720", "2": "960x544", "3": "800x448", "4": "640x360", "5": "424x240", "6": "352x288", "7": "320x240", "8": "800x600", "9": "176x144", "10": "160x120", "11": "1920x1080", "12": "432x240"}
,
"currentResolution": "255"
}

]
}

FormerLurker, 2020-08-02:
Hmm, this suggests the step value should be 1:

name": "Exposure (Absolute)",
"id": "10094850",
"type": "1",
"min": "1",
"max": "10000",
"step": "1",
"default": "156",
"value": "156",
"dest": "0",
"flags": "16",
"group": "1"

If you know how to use the debugger in your web browser (F12), perhaps you could cut out the html for that slider?

rickgode, 2020-08-02:
Thank you so much for your time.  I "think" I got it going.  I'm linking a video to what I messed with.  

https://youtu.be/GRW67CMDIq8

I'm sure it is my camera, but after playing with it, I was able to get good results.  I think I may have "messed" something up by uploading that settings.json file, because now I get a popup saying

"Octolapse has detected new camera preferences from your streaming server.  Do you want to overwrite your current image preferences?"

But I think that's just my weirdness by not using a "standard" Logitech camera.

Thanks again!