FormerLurker/Octolapse #556: Sorry for the 'issue being created... the only google reference is this...

HeneryH, 2020-06-27:
I'm getting the same error as Damien listed below.  I set my Prusa profile to 3 extruders and connected using that profile but OctoPrint keeps complaining that my printer only has 1 extruder.  Believe it or not, Damien's post was one of the only google results about that problem.  Have you ever seen this before?
----------
I found a bug, or maybe it is not one :) The test mode does heat because Octolapse get confused because I have declared I have 2 extruders while the printer only have one (cf. https://medium.com/@damien.martin.guillerez/multi-filament-print-with-a-single-extruder-using-prusaslicer-2e0746348cdd?source=friends_link&sk=541dcda7cd469b502874ed76e266267e).

I get the following error in the logs:

```
2019-12-10 21:30:20,500 - octolapse.__init__ - ERROR - Your printer profile has fewer extruders (1) defined than your gcode file (2).
2019-12-10 21:38:11,367 - octolapse.__init__ - ERROR - Your printer profile has fewer extruders (1) defined than your gcode file (2).
```

I'll try to reslice the thing.

_Originally posted by @damienmg in https://github.com/FormerLurker/Octolapse/issues/417#issuecomment-564249372_

HeneryH, 2020-06-27:
Never mind, now that I see this comment was in the OctoLapse git, I made me remember that there is a completely separate set of printer profiles for OctoLapse and the core OctoPrint.

At the least, this error message about the number of defined extruders not matching the gcode should indicate that it is the OctoLapse profile that doesn't match.

I spent a lot of time tinkering with the core OctoPrint profiles.

FormerLurker, 2020-06-29:
You make a good point about the number of extruders in the OctoPrint printer profile.  I may need to add an option to use the OctoPrint profile settings (this capability exists for several other options), which would also handle the other issue about not knowing where that error is coming from.  I'll also improve the error message in the meantime to further reduce confusion.