FormerLurker/Octolapse #559: DSLR Camera Script Failed

Koyodee, 2020-07-01:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.4.0

#### Version of OctoPrint

OctoPrint Version: 1.4.0

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred

Went to camera settings
Clicked test script (as per the setting up DSLR documentation guide)
Camera shutter clicked, but returned an error. 

#### What should have happened?
OctoPi should've received the photo my camera took

#### What happened instead?
It returned an error (I'm assuming it didn't receive aforementioned photo)

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: 0.17.0

#### Printer model & used firmware incl. version

Printer Model: ___REPLACE_THIS__PRINTER_MODEL_GOES_HERE
Printer Firmware Version: ___REPLACE_THIS__PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: Chrome
Browser OS: Windows 10

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: ___REPLACE_THIS__GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: https://pastebin.com/Jec0FanK (There was no password on the file to begin with, I'm not sure if that's a factor to this error or not.)

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log: https://pastebin.com/yNr9iSfm

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: https://imgur.com/a/fifwuWt | https://imgur.com/a/tVkDb5i | https://imgur.com/a/EPsUMBe

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

Koyodee, 2020-07-01:
I should also add the script that is supposed to be run: https://imgur.com/a/6kFEYW4
The file mentions including gphoto2 to /etc/sudoers , but I didn't see any reference to that in the documentation so I'm not sure how necessary that is. :)

cmatthews, 2020-07-01:
I have just this evening been configuring Octolapse with a Nikon D7000 and had a similar problem. Testing in Octolapse, I had the trigger firing but an error message, The log file showed an error had occured, but no detail about what it was.

To cut a long story short, the version that I originally installed from the Raspbian was quite old and so I uninstalled it and used the script from the installation instructions. Seems to works OK now. The test button works and gives a success message. 

Did you install gphoto2 from the repo' or with the script? If you got it from the repo using apt-get install gphoto2 then consider going back and installing it with the script provided. 

Here is my log file in case it is of any help to @FormerLurker :

[plugin_octolapse (4).log](https://github.com/FormerLurker/Octolapse/files/4861263/plugin_octolapse.4.log)

Koyodee, 2020-07-01:
I just found the updated guide, and am going through that procedure now. When I googled "Octolapse DSLR" it brought me to the  0.3.x guide, so I'm making the changes and will go from there. Thanks for your experience, if updating doesn't help I will definitely try that!

FormerLurker, 2020-07-01:
@koyodee, the sudoers stuff is extremely important.  Follow those steps and reboot (a full reboot, not just restarting octoprint).  If it fails again, post the logfile (plugin_octolapse.log).

FormerLurker, 2020-07-01:
Ok, great!  The new guide is comprehensive.  There is also a video that goes with the wiki.

FormerLurker, 2020-07-01:
@cmatthews, thanks for chiming in!  I will take a look at the logfile asap.

Koyodee, 2020-07-01:
Is there any changes I need to make updating to 0.4.0 outside of the guide, or will anything already installed just be ignored? I don't wanna mess anything up.

FormerLurker, 2020-07-02:
Read/follow the [getting started](https://github.com/FormerLurker/Octolapse/wiki/V0.4---Getting-Started) guide first, and you should be good.

Koyodee, 2020-07-02:
I am very happy to report success! Thank you so much @FormerLurker for all the work you've put into this. The video tutorials especially are very thorough and easy to understand.

closed 2020-07-02 by Koyodee

FormerLurker, 2020-07-02:
Enjoy!

Koyodee, 2020-07-02:
Just wanted to make a quick note regarding a solution I found when this problem popped up again. At least with the camera I was using (Nikon D3200), the script would not work if the camera was configured to take both a RAW (or .nef) picture along with the usual .jpg. Here is what the captured image and download script returned when I ran it in PuTTY. 
[image]
As soon as I changed my camera to only shoot in .jpg, the problem went away. I believe it has something to do with the way the photos are retrieved that get messed up when this setting is on.

FormerLurker, 2020-07-06:
@Koyodee, interesting.  I am aware that .raw or .nef images can cause problems, but only if multiple images are downloaded at once (.jpg + raw).  I will try to add this to the guides.