FormerLurker/Octolapse #560: Octolapse fails to initialize the print after parsing, G21 and other mysteries included

sdertien, 2020-07-02:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.4.0

#### Version of OctoPrint

OctoPrint Version: 1.4.0

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: Yes

#### What were you doing when the problem occurred

1. Fresh install of Octoprint, mjpg-streamer and Octolapse on a NVIDIA Jetson Nano (not from OctoPi)
2. Testing the configuration of Octolapse in Test Mode
3. Checked all the default property settings and the test for the image works
4. When starting the print Octolapse parses the gcode but no further activity happens on the printer
4b. When testing with test mode disabled using any smart trigger nothing happens and no command goes to the printer
4c.  When testing with a classic GCode trigger G4 P1 the following error will appear and the print will start even when in Test Mode: Octolapse Timelapse Stopped
The printer profile requires an explicit G21 command before any position altering/setting commands, including any home commands. Stopping timelapse, but continuing the print.

#### What should have happened?
Award winning timelapse photos filling up disk space!

#### What happened instead?
The logs all seem to indicate that it properly parsed for where the camera snapshots should be taken.  There's no GCode commands being sent to start the print jobs in any way though.  I've searched forums high and low and don't see anything quite like this other than a G21/G90 posting.

I did look at prior issues about the G21 error and you'll see that I did move G21 and G90 to the top of the GCode.  I first tried in Prusa Slicer by putting it to the top but there still was a move command being inserted into the GCode above it.  I ultimately just tweaked it by hand to make sure that both were set.  It seems to make no difference.  The logging also seems to indicate that the smart configurations parse fine.  It simply fails to start warming up the printer and sending the commands after it does the evaluation of the GCode.  Nothing stood out in either log file as an error that coincided a failure or anything along those lines.  It just silently continues, the printer in Octoprint will be stuck in printing but there is nothing appearing on the Prusa screen like you would normally see during that initial transfer to the printer.

#### Operating System running OctoPrint and Octolapse

OS Name: Ubuntu 18.04 LTS on Nvidia Jetson Nano
Os Version: Linux <hostname_changed> 4.9.140-tegra #1 SMP PREEMPT Wed Apr 8 18:10:49 PDT 2020 aarch64 aarch64 aarch64 GNU/Linux

#### Printer model & used firmware incl. version

Printer Model: Prusa Mk3s
Printer Firmware Version: ___REPLACE_THIS__PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: Chromium/Chrome
Browser OS: Ubuntu locally and Windows remotely

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: https://gist.github.com/sdertien/dc9456cd26e7c1531e6bd765e01d6b6e

#### Link to settings.json

Link to settings.json with all passwords removed: ___REPLACE_THIS__SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://gist.github.com/sdertien/dc9456cd26e7c1531e6bd765e01d6b6e

#### Link to octoprint.log

Link to octoprint.log:  https://gist.github.com/sdertien/dc9456cd26e7c1531e6bd765e01d6b6e

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

sdertien, 2020-07-02:
Debugging a little more in FireFox and it's likely an issue with the Snapshot Preview dialog not popping out.  This is the output from the Firefox console:

Error calling onDataUpdaterPluginMessage on view model OctolapseViewModel : text@http://192.168.1.78:5000/static/webassets/packed_libs.js?3339d949 line 11374 > Function:3:57
update@http://192.168.1.78:5000/static/webassets/packed_libs.js?3339d949:11404:237
m/</<@http://192.168.1.78:5000/static/webassets/packed_libs.js?3339d949:11378:287
Pc@http://192.168.1.78:5000/static/webassets/packed_libs.js?3339d949:11357:355
Qc@http://192.168.1.78:5000/static/webassets/packed_libs.js?3339d949:11357:133
aa@http://192.168.1.78:5000/static/webassets/packed_libs.js?3339d949:11356:482
a.B@http://192.168.1.78:5000/static/webassets/packed_libs.js?3339d949:11355:39
m/<@http://192.168.1.78:5000/static/webassets/packed_libs.js?3339d949:11378:273
q@http://192.168.1.78:5000/static/webassets/packed_libs.js?3339d949:11317:245
m@http://192.168.1.78:5000/static/webassets/packed_libs.js?3339d949:11377:318
k@http://192.168.1.78:5000/static/webassets/packed_libs.js?3339d949:11375:387
a.Rb@http://192.168.1.78:5000/static/webassets/packed_libs.js?3339d949:11381:414
c/<@http://192.168.1.78:5000/static/webassets/packed_libs.js?3339d949:11416:349
b@http://192.168.1.78:5000/static/webassets/packed_libs.js?3339d949:11416:1
c@http://192.168.1.78:5000/static/webassets/packed_libs.js?3339d949:11416:298
p@http://192.168.1.78:5000/static/webassets/packed_libs.js?3339d949:11419:176
a.a.Bb@http://192.168.1.78:5000/static/webassets/packed_libs.js?3339d949:11426:174
w@http://192.168.1.78:5000/static/webassets/packed_libs.js?3339d949:11345:392
a.kd/<@http://192.168.1.78:5000/static/webassets/packed_libs.js?3339d949:11420:5
Pc@http://192.168.1.78:5000/static/webassets/packed_libs.js?3339d949:11357:355
Qc@http://192.168.1.78:5000/static/webassets/packed_libs.js?3339d949:11357:133
aa@http://192.168.1.78:5000/static/webassets/packed_libs.js?3339d949:11356:482
ac@http://192.168.1.78:5000/static/webassets/packed_libs.js?3339d949:11356:335
notifySubscribers@http://192.168.1.78:5000/static/webassets/packed_libs.js?3339d949:11343:108
fa@http://192.168.1.78:5000/static/webassets/packed_libs.js?3339d949:11346:499
c@http://192.168.1.78:5000/static/webassets/packed_libs.js?3339d949:11346:277
Octolapse.snapshotPlanStateViewModel/self.update_current_plan@http://192.168.1.78:5000/static/webassets/packed_plugins.js?8913daea:9550:30
Octolapse.snapshotPlanStateViewModel/self.update@http://192.168.1.78:5000/static/webassets/packed_plugins.js?8913daea:9480:30
Octolapse.StatusViewModel/self.previewSnapshotPlans@http://192.168.1.78:5000/static/webassets/packed_plugins.js?8913daea:8352:48
OctolapseViewModel/self.updateState@http://192.168.1.78:5000/static/webassets/packed_plugins.js?8913daea:1546:42
OctolapseViewModel/self.onDataUpdaterPluginMessage@http://192.168.1.78:5000/static/webassets/packed_plugins.js?8913daea:1646:34
callViewModelIf@http://192.168.1.78:5000/static/webassets/packed_core.js?f52b1299:16437:35
callViewModelsIf/<@http://192.168.1.78:5000/static/webassets/packed_core.js?f52b1299:16386:28
Pn@http://192.168.1.78:5000/static/webassets/packed_libs.js?3339d949:11208:530
ur/<@http://192.168.1.78:5000/static/webassets/packed_libs.js?3339d949:11229:66
callViewModelsIf@http://192.168.1.78:5000/static/webassets/packed_core.js?f52b1299:16384:7
callViewModels@http://192.168.1.78:5000/static/webassets/packed_core.js?f52b1299:16378:21
DataUpdater/self._onPluginMessage/<@http://192.168.1.78:5000/static/webassets/packed_core.js?f52b1299:15072:27
DataUpdater/self._ifInitialized@http://192.168.1.78:5000/static/webassets/packed_core.js?f52b1299:15096:13
DataUpdater/self._onPluginMessage@http://192.168.1.78:5000/static/webassets/packed_core.js?f52b1299:15071:14
OctoPrintSocketClient.prototype.propagateMessage/<@http://192.168.1.78:5000/static/webassets/packed_client.js?09c75251:421:24
Pn@http://192.168.1.78:5000/static/webassets/packed_libs.js?3339d949:11208:530
ur/<@http://192.168.1.78:5000/static/webassets/packed_libs.js?3339d949:11229:66
OctoPrintSocketClient.prototype.propagateMessage@http://192.168.1.78:5000/static/webassets/packed_client.js?09c75251:420:15
onMessage/<@http://192.168.1.78:5000/static/webassets/packed_client.js?09c75251:515:22
Zt/<@http://192.168.1.78:5000/static/webassets/packed_libs.js?3339d949:11225:23
_t@http://192.168.1.78:5000/static/webassets/packed_libs.js?3339d949:11213:392
Vt/<@http://192.168.1.78:5000/static/webassets/packed_libs.js?3339d949:11224:369
ur/<@http://192.168.1.78:5000/static/webassets/packed_libs.js?3339d949:11229:73
onMessage@http://192.168.1.78:5000/static/webassets/packed_client.js?09c75251:514:15
[5]</EventTarget.prototype.dispatchEvent@http://192.168.1.78:5000/static/webassets/packed_libs.js?3339d949:22600:20
[14]</</SockJS.prototype._transportMessage/<@http://192.168.1.78:5000/static/webassets/packed_libs.js?3339d949:23313:16
[14]</</SockJS.prototype._transportMessage@http://192.168.1.78:5000/static/webassets/packed_libs.js?3339d949:23311:17
[3]</EventEmitter.prototype.emit@http://192.168.1.78:5000/static/webassets/packed_libs.js?3339d949:22516:18
WebSocketTransport/this.ws.onmessage@http://192.168.1.78:5000/static/webassets/packed_libs.js?3339d949:25387:10
packed_core.js:16450:17
    callViewModelIf http://192.168.1.78:5000/static/webassets/packed_core.js?f52b1299:16450
    callViewModelsIf http://192.168.1.78:5000/static/webassets/packed_core.js?f52b1299:16386
[... cut ...]
    onMessage http://192.168.1.78:5000/static/webassets/packed_client.js?09c75251:514
    dispatchEvent http://192.168.1.78:5000/static/webassets/packed_libs.js?3339d949:22600
    _transportMessage http://192.168.1.78:5000/static/webassets/packed_libs.js?3339d949:23313
    _transportMessage http://192.168.1.78:5000/static/webassets/packed_libs.js?3339d949:23311
    emit http://192.168.1.78:5000/static/webassets/packed_libs.js?3339d949:22516
    onmessage http://192.168.1.78:5000/static/webassets/packed_libs.js?3339d949:25387

FormerLurker, 2020-07-06:
So, the error you are seeing should not be happening since it looks like your printer profile is setup correctly (using the MK3 profile).  This could be some compilation/OS issue.  I need to get ubuntu up and running to test this.  Sorry for the delay, btw.

sdertien, 2020-07-08:
No worries.  I'm trying to figure out another way to get Ubuntu up and running in a VM that I can pass my camera through on just to see if it's the same experience there.

FormerLurker, 2020-11-17:
@sdertien, I know it's been a LONG time, but it's possible your issue has finally been solved.  Can you try installing from the plugin manager using this link in the From Url... box: https://github.com/FormerLurker/Octolapse/archive/88aa71f0fbac306b0d9acf19dff7388a30b29b64.zip

See if that fixes it.

FormerLurker, 2020-12-05:
This should be fixed in V0.4.1, which just dropped. Might take a while to show up in the plugin manager. Reopen a new issue if you continue to have problems. Thanks!

closed 2020-12-05 by FormerLurker