FormerLurker/Octolapse #561: [Request] Save time the nozzle is in the air by allowing the start and return positions to be different

Menace-Dennis, 2020-07-02:
#### If this is a feature request describe it here

Currently, when z-hop is enabled, Cura will lift the head and move to the starting x and y location of the next layer before the layer change. Octolapse in addition then takes the nozzle to the snapshot location, takes the snapshot and returns to that position. Seeing as you now perform pre-processing, I was wondering if you could provide an option for when z hop is enabled that the pre-next layer travel moves will be ignored and then will go to the starting point afterwards. I.e. the nozzle lifts from the end of the last layer goes to the snapshot location, takes the snapshot and then goes to the starting point location. Currently I have the stabilisation on disabled and run a post-processing script on Cura to do this for me. It involves tracking the final location and z height before the end of the layer and putting this after the trigger to make sure it returns to this location, the movements from ";MESH:NONMESH" to ";TIME_ELAPSED:" are then removed. You then have GCode like below which works with the Disabled stabilisation:

For when already lifted:

G1 F300 Z6.2
;MESH:NONMESH
;TIME_ELAPSED:2719.947034
;TIMELAPSE BEGIN
G0 F300 Z6.2 ;Making sure height is correct
; Already Retracted: True
; Already Lifted: True
G0 F9000.0 X140.0 Y140.0 ;Park print head
@OCTOLAPSE TAKE-SNAPSHOT ;Octolapse Trigger
G0 F9000.0 X111.435 Y99.349 ;Move to the next point
;TIMELAPSE END
;LAYER:53
G0 X111.435 Y99.349 Z6.3

For when not already lifted:

G0 F9000 X106.126 Y129.481
;MESH:NONMESH
;TIME_ELAPSED:3757.061674
;TIMELAPSE BEGIN
G0 F300 Z8.4 ;Making sure height is correct
; Already Retracted: False
G0 F1800.0 E1052.38779 ;Retract Filament
; Already Lifted: False
G0 F300.0 Z9.2 ;Z hop
G0 F9000.0 X140.0 Y140.0 ;Park print head
@OCTOLAPSE TAKE-SNAPSHOT ;Octolapse Trigger
G0 F9000.0 X116.48 Y131.922 ;Move to the next point
G0 F300.0 Z8.4 ;Undo Z hop
G0 F1800.0 E1057.38779 ;Undo Retract Filament
;TIMELAPSE END
;LAYER:82

#### Version of Octolapse
 
Octolapse Version: 0.4.0

#### Version of OctoPrint

OctoPrint Version: 1.4.0

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: ___REPLACE_THIS__YES_OR_NO

#### What were you doing when the problem occurred

1. ___REPLACE_THIS__STEP_ONE_GOES_HERE
2. ___REPLACE_THIS__STEP_TWO_GOES_HERE
3. ___REPLACE_THIS__STEP_...

#### What should have happened?
___REPLACE_THIS__PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
___REPLACE_THIS__PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: ___REPLACE_THIS__OS_NAME_GOES_HERE
Os Version: ___REPLACE_THIS__OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: ___REPLACE_THIS__PRINTER_MODEL_GOES_HERE
Printer Firmware Version: ___REPLACE_THIS__PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: ___REPLACE_THIS__BROWSER_VERSION_GOES_HERE
Browser OS: ___REPLACE_THIS__BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: ___REPLACE_THIS__GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: ___REPLACE_THIS__SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  ___REPLACE_THIS__LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2020-07-06:
I will look into this.  Octolapse will already do what you are asking if there is only a single travel move in the case that the triggering command is a travel (G0/G1 with no E parameter), but it is not currently capable of storing multiple travel commands (lift/travel/delift, for example).  Anyway, neat idea, and I'll leave this open.

FormerLurker, 2020-07-06:
Whoops, didn't mean to close :)

Menace-Dennis, 2020-07-06:
Cheers! I was just watching my 3D printer and realised, depending on the filament, the oozing from that long z-hop can cause "zits" so I think it would definitely help. My post-processing script for Cura has definitely helped the issue, it would just be nice if it was inbuilt into Octoprint, especially now as you're doing pre-printing analysis of the Gcode. Anyway, thanks for all your work with Octolapse, it's a great plugin.