FormerLurker/Octolapse #563: Configuring a Raspberry Pi camera help

neobuzzsaw, 2020-07-04:
All goes well until I carry out the reboot following the 'sudo apt-get dist-upgrade', following this I can no longer log back into the Pi, the only way of accessing again is to refresh the SD card and start again. I have noticed when it is carrying out dust-upgrade there are a couple of errors showing up in the list at the end.

Not sure what else to do as tried twice now with same results and I which to use the custom preferences on my Pi Cam to sort out a colour imbalance issue.

Many thanks.

neobuzzsaw, 2020-07-04:
[image]

Just looked on the SD Card and it appears there are a lot of files missing?

FormerLurker, 2020-07-06:
Not sure what is going on with the update.  Perhaps you had an older version of rasbian?  I would recommend a re-flash, but I'm no expert regarding fixing rasbian update issues unfortunately.  I wish I knew some simple steps to fix this for you :(

neobuzzsaw, 2020-07-06:
Hi, I have sort of found a way round it. I followed your instructions but just missed out the update section and it seems to be working fine!

I now have access to all camera options. Just need to figure out a way of sorting the colours as I seem to be only able to get a blue/purple haze or red haze. If I put it into greyworld it just goes B&W. Any suggestions?

BTW - Great work!

FormerLurker, 2020-07-06:
Are you using a raspberry pi camera perhaps?  If so, see this setting:

[image]

It is pretty hard to get those two values correct.  You can start with my settings and go from there.  FYI, some cameras have a sticker (IR camera I think) that fixes the colors.  If you had that sticker or have removed it your settings here would be very different.

neobuzzsaw, 2020-07-06:
[image]

neobuzzsaw, 2020-07-06:
[image]
This it what it looks like now

FormerLurker, 2020-07-06:
Definitely looks like a red/blue balance issue.  As I said before it's difficult to get this setting right, and you seem to have a different model than I do.  I recommend that you continue to play around with it.

Also, FYI, there is a bug that prevents you from changing from manual to automatic balance without a full reboot (not just an octoprint restart).  Unfortunately there are several bugs like that for the RPI cams.

neobuzzsaw, 2020-07-06:
ok thanks, I will continue to mess about. Camera Im using is a Makerhawk Raspberry Pi Camhttps://www.amazon.co.uk/MakerHawk-Raspberry-Camera-Vision-Adjustable/dp/B071718FDK

FormerLurker, 2020-07-06:
FYI, there is a slight chance that something else is going on since this isn't an official RPi camera (unless I'm misreading).  However, since your control page looks right it seems like it's at least mostly compatible.  Keep me posted, and good luck!

neobuzzsaw, 2020-07-06:
Will do, thanks for help

neobuzzsaw, 2020-07-06:
Out of interest what do you have your colour effects set to?

FormerLurker, 2020-07-06:
Set to none.  If you changed it, try a full reboot (one of those bugs I mentioned affects this setting)

neobuzzsaw, 2020-07-06:
Will try reboot, just finishing a print. Don't suppose you have a screenshot of the rest of your settings (stops me asking for one at a time :) )Some off them don't mean a thing to me ha ha !

FormerLurker, 2020-07-06:
Be sure to click on the blue question marks, they explain each setting.  Here is the image you requested:  

[image]