FormerLurker/Octolapse #564: Octolapse photo folder impossible to find?

Suri63, 2020-07-04:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: Octolapse v0.4.0

#### Version of OctoPrint

OctoPrint Version: OctoPrint 1.4.0

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: No..

#### What were you doing when the problem occurred

1. Snapshot files are available, but video was not created
2. ```
Octolapse Rendering Failed
Rendering failed for camera '500D'. render-error: Unknown render error. Please check plugin_octolapse.log for more details.. Inner Exception: [Errno 28] No space left on device: '/home/pi/.octoprint/data/octolapse/tmp/octolapse_rendering_tmp'
```

#### What should have happened?
Timelapse video should have been created

#### What happened instead?
The photo folder is ready but a folder of 900 photos for 4GB (22 hours of printing) but obviously the folder is too big and the software does not want to mount it for lack of available memory. However it is the only photo folder on Octoprint while all the other timelapse and photos are deleted! There should be room
The file cannot be compressed and downloaded because it is too big ... well Octoprint gives me the default "The zip archive is too large to import." ...
The file cannot be deleted because I have the default:
in short I am blocked

#### Operating System running OctoPrint and Octolapse
<!--
OctoPi, Linux, Windows, MacOS, something else?
Include the version number if possible.
OctoPi's version can be found in /etc/octopi_version or in the lower left
corner of the web interface.
DO NOT OMIT
-->
OS Name: Windows 10
Os Version: OctoPrint 1.4.0 running on OctoPi 0.17.0

#### Printer model & used firmware incl. version
<!--
OMIT ONLY IF IT IS DEFINITELY NOT NEEDED.
-->
Printer Model: Ender 3
Printer Firmware Version: Using Firmware homemade... But on Octolapse profile, i'm using Anet A8 beta And it works very well for small timelapse!

#### Browser and version of browser, operating system running browser
<!--
OMIT ONLY IF IT IS DEFINITELY NOT NEEDED
-->
Browser: ___REPLACE_THIS__BROWSER_VERSION_GOES_HERE
Browser OS: ___REPLACE_THIS__BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred
<!--
Many problems I've seen are due to specific gcode files.
Sometimes it's a bug, sometimes it can be slicer settings,
other times it can be Octolapse or OctoPrint setting related.

In any case, the gcode file is often very useful.
On gist.github.com or pastebin.com.
OMIT ONLY IF IT IS DEFINITELY NOT NEEDED
!-->
Link to Gcode File: ___REPLACE_THIS__GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json
<!--
Lots of problems are settings related, and having them available
makes debugging a lot easier.  You can export the settings.json
file from within the Octoprint settings.

BE AWARE - IF YOU HAVE ANY PASSWORDS IN YOUR CAMERA PROFILE,
THEY WILL BE EXPORTED ALONG WITH THE REST OF YOUR SETTINGS!
IF YOU ARE USING CAMERA AUTHENTICATION IN ANY OF YOUR PROFILES,
EITHER REMOVE THE PASSWORDS BEFORE EXPORTING, DELETE THEM FROM THE FILE
AFTER EXPORTING, OR DON'T POST SETTINGS.JSON!  You can search for

"password":

to find all of the entries containing passwords.
OMIT ONLY IF YOU HAVE ENTERED A PASSWORD TO ACCESS YOUR CAMERA
WITHIN OCTOLAPSE
!-->
Link to settings.json with all passwords removed: ___REPLACE_THIS__SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log
<!--
On gist.github.com or pastebin.com.

DO NOT OMIT IF YOU ARE USING OCTOLAPSE V0.2.2 OR LATER.

IF YOU ARE USING V0.2.1 OR EARLIER DO NOT POST YOUR LOGFILE
SINCE IT MAY CONTAIN CAMERA AUTHENTICATION PASSWORDS IF THEY WERE
ENTERED.
-->
Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log
<!--
On gist.github.com or pastebin.com.
DO NOT OMIT
-->
Link to octoprint.log:  ___REPLACE_THIS__LINK_GOES_HERE

#### Link to contents of Javascript console in the browser
<!--
If you are having UI problems, like buttons not working, or problems saving
settings, please inlude the javascript console output
on gist.github.com or pastebin.com.
OMIT ONLY IF IT IS DEFINITELY NOT NEEDED
-->
Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:
<!--
This is really nice to include, especially if you're having trouble
putting your issue into words.  A picture speaks 1000 of them, afterall!
-->
Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

Suri63, 2020-07-04:
I can't understand where I can find the folder full of "timelapse" photos which visibly full and which consumes all the space on the SD card
Do I have to use a 32/64 GB card to manage Octoprint and Octolapse videos?

FormerLurker, 2020-07-06:
Look in ```/home/pi/.octoprint/data/octolapse/tmp/octolapse_rendering_tmp```.  If you don't have enough space to store and render the videos, you will need a bigger SD card.  It looks like you are using a DSLR, so each photo could be multiple megabytes.

Suri63, 2020-07-07:
Hi FormerLurker
Sorry to ask this question, but how do you access this file? with the web browser page? with ssh?
Thank's

FormerLurker, 2020-07-07:
I use an SFTP program (filezilla), though there are lots of ways to do it.  I find SFTP to be the most convenient.   You can use the same username (PI) and password you use to connect via SSH.

Normally you can just download these images straight from the Octolapse tab, but since you have no free disk space it won't work.  Octolapse zips up the images before the download starts, so it needs enough space to do the zipping.

Suri63, 2020-07-12:
Thanks for the solution !! I was able to recover and save my file !! I took the opportunity to clean up!