FormerLurker/Octolapse #565: [Request] Additional overlays

cengelbracht, 2020-07-07:
#### If this is a feature request describe it here

Please add Infill, layer height, hot end temperature, and bed temp overlay options. Maybe using the parsed custom gcode script thing or something. Thanks :)

It would be useful to see these settings in the video when prints fail and when they come out great.

#### Version of Octolapse
 
Octolapse Version: 0.4.0

#### Version of OctoPrint

OctoPrint Version:  1.4.0

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: ___REPLACE_THIS__YES_OR_NO

#### What were you doing when the problem occurred

1. ___REPLACE_THIS__STEP_ONE_GOES_HERE
2. ___REPLACE_THIS__STEP_TWO_GOES_HERE
3. ___REPLACE_THIS__STEP_...

#### What should have happened?
___REPLACE_THIS__PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
___REPLACE_THIS__PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: ___REPLACE_THIS__OS_NAME_GOES_HERE
Os Version: ___REPLACE_THIS__OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: ___REPLACE_THIS__PRINTER_MODEL_GOES_HERE
Printer Firmware Version: ___REPLACE_THIS__PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: ___REPLACE_THIS__BROWSER_VERSION_GOES_HERE
Browser OS: ___REPLACE_THIS__BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: ___REPLACE_THIS__GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: ___REPLACE_THIS__SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  ___REPLACE_THIS__LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2020-07-07:
Thank you for the suggestion.  I will leave this open and will update the ticket if/when this gets added.

FYI, these values are somewhat harder to implement than the existing replacement tokens, but I think it is possible.

cengelbracht, 2020-07-08:
Thanks, fantastic tool, nice work. Could you have the user have it added to the gcode via that script (example below), have the slicer populate it, and octolapse parse it? Maybe allow any replacement token that is called out in the "Global Settings" and skip it if it does not exist in the present gcode. (I'm really only familiar with Cura so far so forgive my ignorance.)

[; Script based on an original created by tjjfvi (https://github.com/tjjfvi)
; An up-to-date version of the tjjfvi's original script can be found
; here:  https://csi.t6.fyi/
; Note - This script will only work in Cura V4.2 and above!
; --- Global Settings
**; layer_height = {layer_height}**
; smooth_spiralized_contours = {smooth_spiralized_contours}
; magic_mesh_surface_mode = {magic_mesh_surface_mode}
; machine_extruder_count = {machine_extruder_count}
; --- Single Extruder Settings
; speed_z_hop = {speed_z_hop}
; retraction_amount = {retraction_amount}
; retraction_hop = {retraction_hop}
; retraction_hop_enabled = {retraction_hop_enabled}
; retraction_enable = {retraction_enable}
; retraction_speed = {retraction_speed}
; retraction_retract_speed = {retraction_retract_speed}
; retraction_prime_speed = {retraction_prime_speed}
; speed_travel = {speed_travel}](url)

FormerLurker, 2020-07-08:
I've got a few ideas on how it could be done.  Layer height can be extrapolated or extracted as you have suggested.  Current temps could be queried from OctoPrint.  Regarding infill, not sure what you are wanting there (percentage or current feature type).

cengelbracht, 2020-09-05:
Never mind I found a workaround that provides the same information. I modified the end gcode script, with the "print settings", as follows:

; Script based on an original created by tjjfvi (https://github.com/tjjfvi)
; An up-to-date version of the tjjfvi's original script can be found
; here:  https://csi.t6.fyi/
; Note - This script will only work in Cura V4.2 and above!
; --- Global Settings
; layer_height = {layer_height}
; smooth_spiralized_contours = {smooth_spiralized_contours}
; magic_mesh_surface_mode = {magic_mesh_surface_mode}
; machine_extruder_count = {machine_extruder_count}
; --- Single Extruder Settings
; speed_z_hop = {speed_z_hop}
; retraction_amount = {retraction_amount}
; retraction_hop = {retraction_hop}
; retraction_hop_enabled = {retraction_hop_enabled}
; retraction_enable = {retraction_enable}
; retraction_speed = {retraction_speed}
; retraction_retract_speed = {retraction_retract_speed}
; retraction_prime_speed = {retraction_prime_speed}
; speed_travel = {speed_travel}
; ---Print Settings
; printing_temperature = {print_temperature}
; build_plate_temperature = {print_bed_temperature}
; infill_density = {infill_sparse_density}
; infill_pattern = {infill_pattern}
; wall_thickness = {wall_thickness}
; support = {support}
; adhesion type = {adhesion_type}