FormerLurker/Octolapse #567: Octolapse causes SimpleEmergencyStop to misbehave in FireFox 78.0.2 on OctoPrint 1.4.1rc2

b-morgan, 2020-07-09:
[See https://github.com/Sebclem/OctoPrint-SimpleEmergencyStop/issues/14#](https://github.com/Sebclem/OctoPrint-SimpleEmergencyStop/issues/14#). I am opening this issue because disabling Octolapse fixes SimpleEmergencyStop.

This problem did (does not) occur with OctoPrint 1.4.0.

#### Version of Octolapse
 Octolapse Version: 0.4.0

#### Version of OctoPrint
OctoPrint Version: 1.4.1rc2

#### When you ran into the problem, did you have diagnostic logging enabled?
Diagnostic Logging was Enabled: No

#### What were you doing when the problem occurred

1. Install SimpleEmergencyStop plugin
2. Install Octolapse plugin
3. Open OctoPrint in FireFox

#### What should have happened?
NavBar should display like it does in Chrome

#### What happened instead?
NavBar contains both versions of SimpleEmergencyStop buttons. Neither one is functional. Octolapse NavBar button is missing.

#### Operating System running OctoPrint and Octolapse
OS Name: OctoPi
Os Version: 0.17.0

#### Browser and version of browser, operating system running browser
Browser: FireFox 78.0.2
Browser OS: Windows 10 (64-bit)

FormerLurker, 2020-07-09:
@b-morgan, I will see what I can find.  I wonder if other navbar buttons work with SimpleEmergencyStop.

b-morgan, 2020-07-09:
The answer to that question may be "No". Below are screenshots of the NavBar from FireFox and Chrome. As you can see, I've got two other NavBar plugins that disappear in FireFox. List of my plugins is also included.

FireFox:
[image]
Chrome:
[image]

[plugins.txt](https://github.com/FormerLurker/Octolapse/files/4899431/plugins.txt)

FormerLurker, 2020-07-09:
OK, it's looking like a shared CSS issue, which  might be OctoPrint related.  If I find this is the case I'll post either an issue or a pull request on the OctoPrint github page.

foosel, 2020-07-13:
Just for the record, that is an issue with SimpleEmergencyStop (it causes a JS SyntaxError when minified, nuking all of `bundled_plugins.js`), not Octolapse.